gioblu/PJON #101: PJON Standard: Opinions / Suggestion

gioblu, 2016-11-16:
What do you think about it? What you would change? Have you any suggestions?

muratangi, 2016-11-19:
I hope this is notice is for collecting suggestions from the community.
Would it be nice to set a connection from linux arduino over serial?
This sort of connection will open many doors to 
My appologies if it is already available.

gioblu, 2016-11-19:
ciao @muratangi, thank you for your feedback! Yes this is the right place for suggestion from the community related to the PJON Standard. Speaking about linux - arduino connection see PJON-Python of @girgitt user, and for C PJON-C of @Matheus-Garbelini. Both implementations needs to be updated to be compatible with 6.0 and following but are a good start.

gioblu, 2017-09-01:
Closing for now, feel free to re-open if needed.

closed 2017-09-01 by gioblu