gioblu/PJON #107: Oversamplig for ESP8266

Radiusxe, 2016-12-31:
Hi,
will be oversampling method implemented for ESP8266?

Thanks.

With best regards
Radius

gioblu, 2016-12-31:
Ciao @Radiusxe, it should work also on ESP8266, give it a try.

gioblu, 2016-12-31:
I am sorry, but I am not able today to test this configuration, I can follow with practical tests in the next days. Happy tinkering, and new year :)

Radiusxe, 2017-01-02:
Hi,
for me it is not working. This sketch receives nothing and on this page is ESP
 not in supported list.
https://github.com/gioblu/PJON/tree/master/strategies/OverSampling

I have try all combinations of settings (with, without receiver, with without simplex; another pins, not ESP like names of pins, but arduino names of pins etc D2 = 4)
#include <PJON.h>
//#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

// <Strategy name> bus(selected device id)
PJON<OverSampling> bus(44);

void setup() {
  bus.set_communication_mode(SIMPLEX);
  //bus.strategy.set_pins(D2, D7);
  bus.strategy.set_pins(D2, NOT_ASSIGNED);
  //bus.strategy.set_pin(D2);
  bus.set_receiver(receiver_function);
  bus.set_error(error_handler);
  bus.begin();

  

  Serial.begin(115200);
  Serial.println("Booot");
  Serial.println("Booot");
};

void error_handler(uint8_t code, uint8_t data) {
  if(code == CONNECTION_LOST) {
    Serial.println(F("Connection with device ID "));
    Serial.print(data);
    Serial.println(F(" is lost."));
  }
  if(code == PACKETS_BUFFER_FULL) {
    Serial.print(F("Packet buffer is full, has now a length of "));
    Serial.println(data);
    Serial.println(F("Possible wrong bus configuration!"));
    Serial.println(F("higher MAX_PACKETS in PJON.h if necessary."));
  }
  if(code == CONTENT_TOO_LONG) {
    Serial.print(F("Content is too long, length: "));
    Serial.println(data);
  }
}
void receiver_function(uint8_t *payload, uint16_t length, const PacketInfo &packet_info) {
  int i = 0;
String prislo;

  while (i< length)
  {
    prislo += char(payload[i]);
    i++;
  }

  
Serial.println(prislo);
}

void loop() {

   bus.receive();
   //bus.update();
};

With best regards Radius

gioblu, 2017-01-03:
Ciao @Radiusxe thank you for assessing that it actually doesnt work out of the box.
I am working on it, a potential issue could be that is necessary to power both transmitter and receiver modules with the same voltage (so no tx 5v, rx 3v3 or the contrary because, power supply voltage slightly shifts transmission frequency) Thank you for your support.

Radiusxe, 2017-01-03:
Hi,
doesn't know why, but on the fifth line should be (and I have) PJON<OverSampling> bus(44);

Tested nodemcu with 5V receiver (PJON 6,2) and still nothing :/. I have test Arduino Uno with 3,3V receiver and works like with 5V receiver (PJON 6,2).

With best regards
Radius

gioblu, 2017-01-04:
Ciao @Radiusxe thank you for your feedback. I am trying to port it, will give you a response late tomorrow.

Radiusxe, 2017-01-04:
Ah, I need to add the code as "code"
`#include <PJON.h>
//#include <ESP8266WiFi.h> //https://github.com/esp8266/Arduino

// bus(selected device id)
PJON<OverSampling> bus(44);

void setup() {
bus.set_communication_mode(SIMPLEX);
//bus.strategy.set_pins(D2, D7);
bus.strategy.set_pins(D2, NOT_ASSIGNED);
//bus.strategy.set_pin(D2);
bus.set_receiver(receiver_function);
bus.set_error(error_handler);
bus.begin();

Serial.begin(115200);
Serial.println("Booot");
Serial.println("Booot");
};

void error_handler(uint8_t code, uint8_t data) {
if(code == CONNECTION_LOST) {
Serial.println(F("Connection with device ID "));
Serial.print(data);
Serial.println(F(" is lost."));
}
if(code == PACKETS_BUFFER_FULL) {
Serial.print(F("Packet buffer is full, has now a length of "));
Serial.println(data);
Serial.println(F("Possible wrong bus configuration!"));
Serial.println(F("higher MAX_PACKETS in PJON.h if necessary."));
}
if(code == CONTENT_TOO_LONG) {
Serial.print(F("Content is too long, length: "));
Serial.println(data);
}
}
void receiver_function(uint8_t *payload, uint16_t length, const PacketInfo &packet_info) {
int i = 0;
String prislo;

while (i< length)
{
prislo += char(payload[i]);
i++;
}

Serial.println(prislo);
}

void loop() {

bus.receive();
//bus.update();
};`

Radiusxe, 2017-01-04:
> PJON<OverSampling> radio_bus(44);

gioblu, 2017-01-05:
Ciao @Radiusxe, I have accepted your pull and now working on porting, needed to test, retest, organize and push the changes I had locally here. Sorry for the delay and thank you again.

Radiusxe, 2017-01-06:
Hi,
I'm glad to hear these good news.

Radius

gioblu, 2017-01-15:
Ciao @Radiusxe now ESP8266 is perfectly receiving in SIMPLEX. Added dedicated timing: https://github.com/gioblu/PJON/commit/9507cfc1f05bb5b45c2704445cad7d416baa7542

they provide now nominal functionality on from Arudino Mega, Leonardo, Uno, Duemilanove, Nano.

I still detect a problem in transmission on ESP, but time slot for this ended now.
Will dig further as soon as possible

Radiusxe, 2017-01-15:
Hi gioblu,
very thanks for your time. I will try it when it will be available, but still wait for full support (I use acknowledge).

With best regards
Radius

divya0056, 2017-01-16:
Hello, Gioblu, Tested with Nano as Tx and Esp8266 as RX in simplex mode , works great. 
used gain as 20 on tx side , with higher gain more packets were lost. Also noticed if there is a delay after bus.update(), causes significant packets  lost.

//Code used for Arduino Nano
#define OS_GAIN_REFRESH_DELAY 20
#include <PJON.h>
PJON<OverSampling> bus(45);
int t;

void setup() {
  Serial.begin(9600);
  bus.set_communication_mode(SIMPLEX);
  bus.strategy.set_pins(NOT_ASSIGNED,12);
  bus.begin();
  t = 0;
}

void loop() {
t = t+1;
Serial.println(t);
char TEMP [3] = {'T', t >> 8, t & 0xFF};
bus.send(44, TEMP, 3); 
bus.update();

}

//Code for Esp8266

#include <Wire.h> 
#include <LiquidCrystal_I2C.h>
LiquidCrystal_I2C lcd(0x3F, 16, 2);

#define OS_GAIN_REFRESH_DELAY 0
/* Gain refresh time of SRX882 module is around 100 milliseconds.
   If only one pair of SRX and STX are used to connect 2 devices in SIMPLEX
   mode, there is no need to refresh receiver's gain, being communication
   mono-directional. */

#include <PJON.h>

PJON<OverSampling> bus(44);

void setup() {
  bus.set_communication_mode(SIMPLEX);
  bus.strategy.set_pins(16, NOT_ASSIGNED);
  bus.begin();
  lcd.begin();
  bus.set_receiver(receiver_function);

  Serial.begin(9600);
};

void receiver_function(uint8_t *payload, uint16_t length, const PacketInfo &packet_info) {

  if((char)payload[0] == 'T') 
  {
    Serial.print("TEMPERATURE received: ");
    // Get back the original integer from the 2 separate bytes
    
    int temp = (payload[1] << 8 | payload[2] & 0xFF);
    lcd.setCursor(0,0);
    lcd.print(temp);  
    Serial.println(temp);
 }
 
}

void loop() {
  
 bus.receive(); 
}

Thank you .

gioblu, 2017-01-25:
CIao @divya0056 thank you for your report. Working to get it right in both ways as soon as possible

gioblu, 2017-03-31:
Ciao @divya0056 sorry for the delayed response, it seems could be a problem of power supply, trying supplying rf modules with 3v3, it should work