gioblu/PJON #110: Should 7 be camelized?

gioblu, 2017-01-08:
Please feel free to express you opinion on codestyle. Many users privately asked me to switch to camelized to support keyword highlight feature in the Arduino IDE (that seems broken if `_` is used) and to be compliant with the camelized style, suggested by the Arduino Team. This is a big change that could bother many users forced to edit their sketches, to support the new versions.

nomeme, 2017-01-08:
I could see this offending other c/c++ developers that are used to the `_` style in their daily work.
Also changing an interface in so many places is quite a hassle for the developer and user.

On the other hand being conform to platform style guides is also quite important and makes usage of several libraries easier and increases code readability.

As many other libraries also ignore the style guide I would not see it as a high priority change, as the styling will be mixed anyways.

gioblu, 2017-01-08:
Ciao @nomeme thank you for your opinion. I totally agree with you.

gioblu, 2017-01-19:
I will leave out this issue for another 48 hours, if you have something to say about, please do.
We are feeling not to change codestyle for now.

closed 2017-01-21 by gioblu