gioblu/PJON #111: Link error if PJON is used in several cpp files

nomeme, 2017-01-08:
Hi, I discovered a link error when using PJON in several cpp files:

See the example code for a small library

```
#ifndef Test_h
#define Test_h

#include <PJON.h>

class Test
{
    public:
        void send(int i);
    private:
        PJON<OverSampling>* _bus;
};
#endif
```

```
#include "Test.h"

void Test::send(int i)
{
    int ii=i+i;
}
```

And the respective sketch code.

```
#include "Test.h"

PJON<OverSampling> bus(44);

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

using this i jet following link errors when compiling the sketch.

```
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `EthernetLink::start_listening(unsigned int)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `roll_crc_8(char, unsigned char)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `compute_crc_8(unsigned char const*, unsigned int)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `compute_crc_32(unsigned char const*, unsigned int, unsigned long)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `bus_id_equality(unsigned char const*, unsigned char const*)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `copy_bus_id(unsigned char*, unsigned char const*)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `EthernetLink::init()'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `EthernetLink::read_bytes(EthernetClient&, unsigned char*, unsigned int)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `EthernetLink::read_until_header(EthernetClient&, unsigned long)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `EthernetLink::receive(EthernetClient&)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `EthernetLink::disconnect_out_if_needed(int)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `EthernetLink::find_remote_node(unsigned char)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `EthernetLink::add_node(unsigned char, unsigned char const*, unsigned int)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `EthernetLink::accept()'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `EthernetLink::disconnect_in_if_needed()'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `EthernetLink::connect(unsigned char)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `EthernetLink::single_socket_transfer(EthernetClient&, int, bool, char const*, unsigned int)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `EthernetLink::receive()'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `EthernetLink::receive(unsigned long)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `EthernetLink::send(EthernetClient&, unsigned char, char const*, unsigned int)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `EthernetLink::poll_receive(unsigned char)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `EthernetLink::send(unsigned char, char const*, unsigned int, unsigned long)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
libraries/Test/Test.cpp.o (symbol from plugin): In function `EthernetLink::start_listening(unsigned int)':
(.text+0x0): multiple definition of `EthernetLink::send_with_duration(unsigned char, char const*, unsigned int, unsigned long)'
sketch/sketch_jan08a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
exit status 1
```

nomeme, 2017-01-08:
I can take a look into this and try fixing it within the next week or so, creating a pull request.

gioblu, 2017-01-08:
I have to dedicate some time to the ESP porting for OverSampling as promised :) so if you could would be amazing. I in any case reported to @fredilarsen that is the creator of EthernetTCP.

nomeme, 2017-01-08:
It has something to do with how the includes are handled in the strategies. After fixing ethernet i ran into other issues, but don't have the time to do it today so I'll take a look next week.

gioblu, 2017-01-12:
Ciao @nomeme now it should work! see https://github.com/gioblu/PJON/commit/e039f746819d0ec1f3e36a2bccc5f7a0238b0d60

nomeme, 2017-01-12:
oh wow i just wanted to start fixing it.

gioblu, 2017-01-12:
Thanks @fredilarsen for fixing, closing.

closed 2017-01-12 by gioblu

keyeh, 2017-01-13:
I'm still getting a few linker warnings:

```
In file included from .piolibdeps\PJON/strategies/EthernetTCP/EthernetTCP.h:23:0,
from .piolibdeps\PJON/PJON.h:78,
from src/ComBus.hpp:10,
from src\ComBus.cpp:1:
.piolibdeps\PJON/strategies/EthernetTCP/EthernetLink.h:139:0: warning: "FAIL" redefined [enabled by default]
#define FAIL          0x100
^
In file included from .piolibdeps\PJON/PJON.h:72:0,
from src/ComBus.hpp:10,
from src\ComBus.cpp:1:
.piolibdeps\PJON/PJONDefines.h:82:0: note: this is the location of the previous definition
#define FAIL         65535
^
```

gioblu, 2017-01-14:
Ciao @keyeh thank you for your feedback, fixing.

gioblu, 2017-01-15:
This: https://github.com/gioblu/PJON/commit/a4d5b090ec4a709d706e9177382460749283f0f8
should avoid the redefinition. Thanks again

keyeh, 2017-01-17:
@gioblu Thanks for the fix!

I'm not sure if this is normal, but PJON.h almost always needs to be included before other files.

I'm working with ESP8266 chips on the latest a4d5b090ec4a709d706e9177382460749283f0f8 with PlatformIO IDE

This code:
```
#include <Arduino.h>
#include <SPI.h>
#include <ESP8266WiFi.h>    // notice this is above PJON.h
#include <PJON.h>

PJON<SoftwareBitBang> bus(45);

void setup() {
}

void loop() {
};

```

Does not compile and causes this error: https://gist.github.com/keyeh/8f6e21309b551d0f1889f3e6c3c7c832

But changing the order of the includes compiles ok:
```
#include <Arduino.h>
#include <SPI.h>
#include <PJON.h>
#include <ESP8266WiFi.h>   // Below PJON.h
```

Is this intentional? This makes it difficult to use PJON with object oriented style in a complex project.

gioblu, 2017-01-17:
Ciao @keyeh thank you for your support. No, this is not intentional at all and will be fixed as soon as possible.

gioblu, 2017-01-20:
Ciao @keyeh this pull of @fredilarsen should fix the bug you reported: https://github.com/gioblu/PJON/pull/114

Thank you for your support.