gioblu/PJON #113: RS485: 1 master with a lot of "speaking" slaves

ibantxo, 2017-01-15:
Hi @gioblu !

Congratulations for PJON. I have been searching for something like it over internet, and I found it!

I am developing a RS485 physical bus (<100m). For now (testing) it is some centimeters long  :-)
In my project I want to mount about 20 "slaves" and 1 master. The slaves will speak (when a sensor is activated) to master, and master rarelly will send commands to slaves. So I am searching a way for slaves to start the conversation. Master is a gateway to "ethernet world" for this slaves. It is a "pub/sub client" of a larger system (mosquitto). 

Master is an Arduino Uno, and all slaves are arduino nano. May be too much for an Arduino? We'll see!!! lol

I was reading the "modbus like" implementations but they are based on "master pooling", so I decided to search more...

I have been reading the info of the wiki, issues and examples and I am confussed: I do not know what strategy I have to implement: ThroughSerial or SoftwareBitBang. Why? Because in the code of [this link](https://github.com/gioblu/PJON/tree/master/strategies/ThroughSerial) appears a reference to SoftwareBitBang.

I have not undestood what TS_FREE_TIME_BEFORE_START value I have to use: 500 or 110000?

For now I have mountd only 1 master and 3 slaves. I have got 1 slave TXing the data to the master. I have not got 2 slaves working. Second is not working for now... And there is something wrong because when I interchange master position and slave position the communication does not work.

Well... I do not know if "speaking" slaves is possible using a half-douplex communication in a same bus with a lot of "speaking" slaves. 

My idea is to continue developing with PJON and if not, try again modbus.

Many thanks again!!!

Iban

gioblu, 2017-01-16:
ciao @ibantxo! I would should use `ThroughSerial` + some serial to rs485 modules using asynchronous acknowledgment and disabling the synchronous!
`ThroughSerial` avoid collisions checking for incoming data + random delay (collision is generally handled by the data link, as in `SoftwareBitBang` and `OverSampling` with direct channel state analysis).

Speaking about the `TS_FREE_TIME_BEFORE_START` value, stricly depends on the modules you use, the traffic and so quantity of devices present on your network. If is a master-slave network, the free time can be a minimal timeframe, higher timeframe in case of mult-master.

ibantxo, 2017-01-16:
Hi @gioblu, pleased to meet you!!

I have some TTL to rs485 modules "working" in the same network. Interesting [link](http://www.instructables.com/id/RS485-Serial-Communication-Between-Arduino-Mega-an/)

Only 2 nodes are working for now OK. Not working with 3... I think there is a problem with reflections in the bus, so I have to test putting the "terminal resistance loads" at the beginning and at the end of the bus. I'll try it...

**asynchronous acknowledgement**
 I think asynchronous is the default, so I have not to do nothing:
>   #define ACK_MODE_BIT      B00001000 // 1 - Asynchronous acknowledge | 0 - Synchronous acknowledge

**ThroughSerial**
I am already using this strategy. So, no problem!

**TS_FREE_TIME_BEFORE_START**
I'll try with different values.

Thank you very much!!!

gioblu, 2017-01-16:
CIao @ibantxo, the default acknowledgment system used is the synchronous. If you want to use the async, look at this example: https://github.com/gioblu/PJON/tree/master/examples/Network/AsyncAck
It is necessary to predefine the `INCLUDE_ASYNC_ACK` before PJON.h inclusion to keep thin the memory footprint if the mechanism is not used (see example).

It could be a matter of configuration:
- Time dedicated to reception i.e. receive(time in microseconds)
- `TS_FREE_TIME_BEFORE_START` setup
- Correct packet configuration

ibantxo, 2017-01-20:
Thanks @gioblu.

I have got 2 devices working with no problem: MASTER + SLAVE via RS485 (with TTL to RS485 converters)

I have 5 nodes in the same bus. First and last are the unique that have a "load" of 120ohm. The pull-up and pull-down resistors are only in the first node ("master" or "controller"). I  think this is very important to avoid reflections in the network.

Ground "line" is very important to be in parallel to the bus, using the same path that the bus uses. It is a bad ide to use an alternative path to for the ground...

For the moment, I have an "physical" issue to solve. I want to have 5 devices working in the same bus. I think if I solve it the system will work...

I'll tell you my progress... Many thanks!!!

Girgitt, 2017-01-20:
@ibantxo what kind of cabling are you using for the RS-485 bus? It's differential bus and I hope you use a twisted pair. What is your baud rate and resonator frequency? Can you also post a gist with the code you use for testing?

ibantxo, 2017-01-20:
I @Girgitt!!! Pleased to meet you!!

Well... I am using a twisted pair of a FTP. But in my testings I have an "alone" twisted pair (i.e. I have remove one "twisted pair" from the FTP for laboratory tests...)

I have to make a better test, using the whole FTP: 1 twisted pair for RS485 differential bus and 1 twisted pair for GND and +24V. It is very important to "transport" the ground in parallel to signal pair. Same "trip", same distance, close one of each other...

In each node I use a "step-down" converter [Pololu 5V, 500mA Step-Down Voltage Regulator D24V5F5] to converting from 24V DC to 5V DC. So I am using a 4wire of FTP but in a "DIY way"... a PoE like "powered RS485" lol

I do not know the resonator frequency, but in my tests I am using a very low baud rate of 9600. I am sure I have a "physical" issue I have to resolve first.

I could attach the code that I am using but I am using ThroughSerial strategy with a INCLUDE_ASYNC_ACK before PJON.h. I have used/adapted some examples inside the PJON "example" folder.

I preffer to inprove my "laboratory environment" first because I think I have an issue in my hardware.

Any way, many thanks @Girgitt. We'll be in contact!!!

Iban

ibantxo, 2017-01-20:
Well... I have make a better test, using a normal FTP: about 2m long with MASTER at the beggining, SLAVE1 in the middle and SLAVE2 at the end.

- 1 twisted pair for +5 and GND 
- 1 twisted pair for signal (differential signal) bus

When both slaves are working at the same time, only the info of SLAVE1 is OK at MASTER.
Only if I switch manually the node in the middle into a "forced listening" status then info of SLAVE2 is OK at MASTER.

I have been interchanging hardware in order to know if all nodes are OK. Everythin seems OK at physical layer...

So, I think I have something wrong in the code, or somethin missing...

[SLAVE code](https://gist.github.com/ibantxo/9025720824cb0ff4d8b1120a5b01d781)
[MASTER code](https://gist.github.com/ibantxo/d916fc09f340a54d652b701260573cea)

ibantxo, 2017-01-20:
I have inserted in both slaves and master:
`  #define TS_MAX_BYTE_TIME           1000000
  #define TS_FREE_TIME_BEFORE_START  1100000`

That is: 1s and 1.1s.

I have seen that depending on what slave "starts" the communication first, then it get's the communication "online" with the master all the time, not allowing to other slave to "insert" its info...
I have seen this because depending on what slave go out from reset status first that is happening.

Girgitt, 2017-01-20:
@ibantxo try setting timeouts at approx. 2 x byte transmit time for your baud rate. Your values are an order of magnitude too long - should be couple ms max. BTW for bus length measured in meters you should be able to use much higher baud rate like 156000 (example optimized for 16MHz). You could also try other PJON tags - e.g. v5 where CRC is calculated incrementally by receiving party (to eliminate posible timeout issues with CRC calculated after packed reception; this should not matter for short payloads from your example code but who knows) . Synchronous ACK should work just fine.

Girgitt, 2017-01-20:
Other thing is your slaves are not really "slaves" as they initiate connection (send packets) to "master" on their own. 

I assume addresses of each node are unique (?). 

PJON, in contrast to Modbus, is a multi-master protocol and "slave" is just a possible role of a node but in your example "slave" does not send packet in response to any command. This should work thanks to collision avoidance, outgoing packets queue, acknowledgements and repetitive sending on delivery failure (feautures of a multi-master protocol for a half-duplex bus the RS-485 is).

ibantxo, 2017-01-20:
You are right. The are not "slaves" because they initiate connection. That is the main purpouse of the project for "trap" certain changes in the status and not to wait a "pooling" like modbus (for example)

Yes, of course, I use 11 and 22 for address for each ""slave"" node.

I am checking now with values of
`#define INCLUDE_ASYNC_ACK true`
`#include <PJON.h>`
`#define TS_MAX_BYTE_TIME 2000`
`#define TS_FREE_TIME_BEFORE_START 2500`

No success for the moment. lol

I'll try now with SYNC "ACK" before changing baud rate. I think 9600 is quite good for the purpose of the project but I'll test with other higher values.

ibantxo, 2017-01-20:
Umm... there is something wrong...
Step 0. Master + 2 "slaves" connected.
Step 1. Master only "reads" info from "slave2"
Step 2. I disconnect "slave2" and set that node in "listening mode" 
Step 3. "slave1" does not send anything (TX led is not flashing)
Step 4. If I reset "slave1" then starts sending (TX and RX leds are flashing)

Maybe the problem is not those values. Maybe overflow in serial buffer?

ibantxo, 2017-01-21:
I have made a test with modbusrtu (pooling sequence) @ 115200 baudrate and everything works very well. 

But I do not like "modbusrtu" solution because in my project every "slave" will be in "silence" at 99.99% of time. But it is very important to minimize delay when the "slave" have to notify a change to the "master". It should be as fast as possible.

gioblu, 2017-01-21:
CIao @ibantxo, could you explain more in particulars your hypothesis of serial buffer overflow?

ibantxo, 2017-01-21:
Well... I have configured my _slaves_ in this way: there is a counter (each loop, counter=counter+1). Only when the counter==100000 then _slave_ sends info to the _master_ via bus.send(44,"B",1)
But, if not, _slave_ do nothing. So... almost all the time _slave_ is not sending info... only waiting...

But someway it is like the bus is "granted" for one _slave_ and not for the other. The info from other _slave_ does not arrive to _master_. So, I have 2 hypothesis:
1) the bus is "blocked" by the slave all the time
2) the bus is "blocked" by the master for that slave all the time
3) the "waiting _slave_" has "blocked" its serial port TX buffer or something like that.

If I follow the sequence explained previously, then:
case 1) it cannot be, because I remove that micro from the bus, setting its node in "listening mode". Other slave is not starting by its own...
case 2) it cannot be, because If after setting the node in "listening mode" I reset the master, the other node continues with same problem. 
case 3) if I reset the micro of that "waiting _slave_" and previously I have configured other node in "listening mode" (so it is alone in the bus) then everything starts working without reseting the master.

I'll think about more test to obtain more data...

Many thanks

gioblu, 2017-01-21:
Ciao @ibantxo, thank you for your support to the project. 
It would be nice to isolate the problem, is this happening also simply through serial without rs485 involved?

ibantxo, 2017-01-21:
I tested without rs485 last day: one to one. Master <--> Slave, everything was OK.
I tried Master <--> 2xSlave (slaves in parallel) via Serial(PJON), but it was not OK.

I have just make another test. Now I have configured a "input pin #3" in each slave. By default its value is '0' (using a resistor to gnd). Only if pin3=+5V then --> bus.send(44, "B", 1);
So... no bus.send if not +5 present.

Steps:
1) switch on everything. By default... no bus.send at all.
2) I "switch on and off" pin3 @ slave1 --> incredible fast flashing and unflashing led13 @ master. Perfect! Pin3 @ slave1 is now OFF.
3) I "switch on and off" pin3 @ slave2 --> nothing happens at led13 @ master.
4) I reset slave2
5) I "switch on and off" pin3 @ slave2 --> incredible fast flashing and unflashing led13 @ master. Perfect! Pin3 @ slave2 is now OFF.
6) I "switch on and off" pin3 @ slave1 --> nothing happens at led13 @ master.
7) I reset slave1
8) I "switch on and off" pin3 @ slave1 --> incredible fast flashing and unflashing led13 @ master. Perfect! Pin3 @ slave1 is now OFF. 

Some PJON config issue at slaves or master. I have tried a lot of values for TS_MAX_BYTE_TIME and TS_FREE_TIME_BEFORE_START.

Could it be a "max retry overflow" at slave or something like that?

Many thanks!!!

ibantxo, 2017-01-24:
Hi,

I have received new TTL to RS-485 conversors, so I can do now more tests. For now, I have 4 nodes:
node 1 (at one end): MASTER
node 2: SLAVE1
node 3: SLAVE2
node 4 (at the other end): SLAVE3

I have inserted in the same bus 4 nodes, but 4th node in "listening mode" without SLAVE3 arduino. So 4 nodes = 1 MASTER + 2 SLAVES + 1 NODE "idle" (listening mode). SLAVE3 is not connected to NODE (only for power) for now.

If one of the nodes "speaks" to master, everything goes OK. If the same node respeaks, everything OK. But after that, if I put the other slave speaking to master, master does not receive that info or the slave is not really transmitting. I think is not transmitting because TX led is not flashing.

After that, if I connect SLAVE3 to NODE3, and if SLAVE3 starts speaking... MASTER receives the info.

So, in my opinion the bus is "busy" for slaves. So other connected slaves cannot speak after one slave has send some information to master.

Could it be? Could it be the reason?

Many thanks!!

gioblu, 2017-01-25:
Ciao @ibantxo I acknowledge your report, and looking for a reason :) hope to have an answer soon. Please post the code if you can.

ibantxo, 2017-01-25:
Hi @gioblu !!!

I have just captured some digital information on second slave:
- scan000: it is the scan just after booting the system (no slave started sending info yet)
- scan001: it is just when first slave is speaking

[image]

[image]

Maybe RX pin the problem is that "RX" pin is 1 always (when nobody is speaking)?
When "nobody" is speaking, the RS485 "diferential" bus status is "High Z" and at the slave the output of the RS485 to TTL (i.e. the RX of the arduino) is "1", according to the specifications of MAX485:

[image]

I am posting the code in another post now. 

Many thanks!!

gioblu, 2017-01-30:
CIao @ibantxo thank you for your valuable support, I am sorry for the delay, busy weeks.
@Girgitt is who originally tested and ported RS485 in "auto-tx" mode, and I never had still the time to test ThroughSerial over RS485, but I convey with you that having as a base state 1 there could avoid multi-master communication as a whole, but I still didnt get completely the auto-tx arrangement @Girgitt used also because is not so much documented. Other users got it working fine, but maybe they inadvertently have not implemented a real multi-master setup, more digging and testing necessary on my side to have a satisfactory answer..

ibantxo, 2017-01-30:
Hi @gioblu!

I have found the problem, and I solved it. :-)

The problem was in the code. "slave" code cannot send info if there is something ready in the "RX buffer". "can_start is false when serial_available()", so "send_packet" is not possible if RX buffer is not emptied previously.

So... "slave" code must have a "bus.receive()" call and a receiver_function.

Now, every slave can send info to "master".

Many Thanks!!!

gioblu, 2017-01-30:
Cool @ibantxo !! Thank you so much! Could you please pull request that?
Many thanks to you for your support and passion!

ibantxo, 2017-01-30:
Well... Sorry @gioblu , but I do not understand what I have to do exactly (pull request)... I think that is not a bug of PJON...

gioblu, 2017-01-30:
So you are not speaking about PJONMaster and PJONSlave, but of separate third-party classes?
If so I agree, not PJON related, sorry for misunderstanding.

closed 2017-01-30 by gioblu

dukelec, 2018-06-12:
Hi guys, @ibantxo @gioblu 

Please take a look at: https://github.com/dukelec/cdbus_ip

It can avoid RS485 collision as the same way as CAN bus, I hope PJON could support transmission through the CDBUS frame.

More info: https://github.com/dukelec/cdbus_doc