gioblu/PJON #115: Sending a response

Ardumania, 2017-01-20:
Hello.
I'm writing code for my project and there are some things I can't understand and can't find an explanation in the wiki.
In some examples I have seen inside receiver_function a reply just as:
  `bus.reply(var, length_in_bytes);`
In the example local>SendAndReceive there is a global variable int packet and in the receiver_function I can see:
      if(!bus.packets[packet].state)
      packet = bus.reply(var,  length_in_bytes); // Avoid duplicate sending checking old packet state
And now I don't know who of both use, can you explain me the difference?

gioblu, 2017-01-20:
If you need simply to reply within the receiver function use: `bus.reply(var,  length_in_bytes);`
the example you quoted is made like this to avoid to send a reply twice (for example a device could receive, using synchronous acknowledgment, the packet 2 times, because its acknowledge is not heard by who transmitted the packet) and to avoid to reply multiple times / accumulate many replies (if already is present a dispatched but still not delivered reply).

This is an old example, simply using asynchronous acknowledgment thanks to the packet id added in this case, makes impossible for the receiver function to be called twice on a packet with the same packet id. The reply state handling is in any case sane to be handled to avoid multiples.

gioblu, 2017-01-20:
Ciao @Ardumania, forgot to mention that the global variable, is simply used to statically handle the state of the reply packet.
if you do: `uint16_t packet = bus.reply("X", 1);` `packet` will contain the index of the packet in the packets buffer so `bus.packets[packet]` will point to your packet in the memory so `bus.packets[packet].state` contains the state of your packet. if there is no state `if(!bus.packets[packet].state)` it means that the packet was delivered and deleted from memory. This was considered by many users as snooping and insane heheh, also because we are trying to solve the problem on the other way round, catching the errors. Still in some cases you could be forced for necessity to act like this, so is still documented as an option, also if not suggested (because you could create yourself strange bugs, not handling coherently the packet dispatch, for example, before checking if the reply packet was delivered, you could dispatch another packet that could take the place of the delivered reply, so should be used consciously)..

Ardumania, 2017-01-20:
Thank you very much @gioblu, you are the best!!!

gioblu, 2017-01-21:
Don't exagerate @Ardumania heeh.
Always happy to help :)

closed 2017-01-21 by gioblu