gioblu/PJON #119: RS485 support

ekarlso, 2017-02-19:
Hi, does this support rs485 as well ?

gioblu, 2017-02-19:
Ciao @ekarlso, using ThroughSerial strategy it is possible to adopt serial to RS485 modules and so run PJON ThroughSerial over RS485

ekarlso, 2017-02-19:
Ya but how does this handle collissions etc ?

gioblu, 2017-02-20:
Ciao @ekarlso In SoftwareBitBang, thanks to channel analysis, random delays, padding bits and to keep the channel busy between a packet and a synchronous acknowledgment the collision problem is solved safely on strategy or data link level: https://github.com/gioblu/PJON/blob/master/strategies/SoftwareBitBang/specification/PJDL-specification-v1.0.md

ThroughSerial has the feature of collision avoidance based on random delay and channel analysis before transmission; to avoid a collision between a packet transmission and a sync ack transmission the solution proposed for now is simply to wait more time the maximum time imposed for the transaction and so avoid to start to transmit in the middle: https://github.com/gioblu/PJON/blob/master/strategies/ThroughSerial/Timing.h#L37

If the timing constraints (totally related to the hardware performance and baudrate used) are not acceptable you are free to simply to avoid sync ack and switch to the asynchronous acknowledgment instead.

In both cases, PJON on top of the strategy, or data link, does packet handling, back-off, CRC and so on.

Looking again the codebase I see here:
https://github.com/gioblu/PJON/blob/master/strategies/ThroughSerial/ThroughSerial.h#L60

before returning it may be more safe to do as well as in SoftwareBitBang:
https://github.com/gioblu/PJON/blob/master/strategies/SoftwareBitBang/SoftwareBitBang.h#L88
add the dedicated random delay!

gioblu, 2017-03-15:
CIao @ekarlso closing here for now, feel free to reopen if needed!

closed 2017-03-15 by gioblu