gioblu/PJON #120: compile failure on mega2560

soccou, 2017-02-22:
Hello,
I am not sure if I am missing something here, but when I try to compile for the arduino mega 2560 on linux using the arduino 1.0.6 64bit IDE I get many ISO C++ errors :+1: 
`Script started on Wed 22 Feb 2017 04:25:05 PM EST

In file included from lib/PJON-master/PJON.h:76,
                 from .build/mega2560/src/ENCJ2860_Datalogger_House_furnaceroom.cpp:15:
lib/PJON-master/PJONDefines.h:175: error: ISO C++ forbids initialization of member ‘header’
lib/PJON-master/PJONDefines.h:175: error: making ‘header’ static
lib/PJON-master/PJONDefines.h:175: error: ISO C++ forbids in-class initialization of non-const static member ‘header’
lib/PJON-master/PJONDefines.h:176: error: ISO C++ forbids initialization of member ‘id’
lib/PJON-master/PJONDefines.h:176: error: making ‘id’ static
lib/PJON-master/PJONDefines.h:176: error: ISO C++ forbids in-class initialization of non-const static member ‘id’
lib/PJON-master/PJONDefines.h:177: error: ISO C++ forbids initialization of member ‘receiver_id’
lib/PJON-master/PJONDefines.h:177: error: making ‘receiver_id’ static
lib/PJON-master/PJONDefines.h:177: error: ISO C++ forbids in-class initialization of non-const static member ‘receiver_id’
lib/PJON-master/PJONDefines.h:179: error: ISO C++ forbids initialization of member ‘sender_id’
lib/PJON-master/PJONDefines.h:179: error: making ‘sender_id’ static
lib/PJON-master/PJONDefines.h:179: error: ISO C++ forbids in-class initialization of non-const static member ‘sender_id’
In file included from lib/PJON-master/PJON.h:103,
                 from .build/mega2560/src/ENCJ2860_Datalogger_House_furnaceroom.cpp:15:
lib/PJON-master/strategies/AnalogSampling/AnalogSampling.h:289: error: ISO C++ forbids initialization of member ‘threshold’
lib/PJON-master/strategies/AnalogSampling/AnalogSampling.h:289: error: making ‘threshold’ static
lib/PJON-master/strategies/AnalogSampling/AnalogSampling.h:289: error: ISO C++ forbids in-class initialization of non-const static member ‘threshold’
In file included from lib/PJON-master/PJON.h:106,
                 from .build/mega2560/src/ENCJ2860_Datalogger_House_furnaceroom.cpp:15:
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h:33: error: ISO C++ forbids initialization of member ‘serial’
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h:33: error: making ‘serial’ static
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h:33: error: invalid in-class initialization of static data member of non-integral type ‘Stream*’
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h:150: error: ISO C++ forbids initialization of member ‘_enable_RS485_pin’
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h:150: error: making ‘_enable_RS485_pin’ static
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h:150: error: ISO C++ forbids in-class initialization of non-const static member ‘_enable_RS485_pin’
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h: In member function ‘boolean ThroughSerial::can_start()’:
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h:57: error: ‘serial’ was not declared in this scope
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h:60: error: ‘serial’ was not declared in this scope
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h: In member function ‘uint16_t ThroughSerial::receive_byte()’:
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h:83: error: ‘serial’ was not declared in this scope
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h: In member function ‘void ThroughSerial::send_byte(uint8_t)’:
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h:101: error: ‘serial’ was not declared in this scope
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h: In member function ‘void ThroughSerial::send_response(uint8_t)’:
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h:112: error: ‘serial’ was not declared in this scope
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h: In member function ‘void ThroughSerial::send_string(uint8_t*, uint8_t)’:
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h:127: error: ‘serial’ was not declared in this scope
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h: In member function ‘void ThroughSerial::set_serial(Stream*)’:
lib/PJON-master/strategies/ThroughSerial/ThroughSerial.h:137: error: ‘serial’ was not declared in this scope
In file included from lib/PJON-master/strategies/EthernetTCP/EthernetLink.h:131,
                 from lib/PJON-master/strategies/EthernetTCP/EthernetTCP.h:23,
                 from lib/PJON-master/PJON.h:109,
                 from .build/mega2560/src/ENCJ2860_Datalogger_House_furnaceroom.cpp:15:
lib/Ethernet/Ethernet.h: At global scope:
lib/Ethernet/Ethernet.h:39: error: ‘EthernetClass Ethernet’ redeclared as different kind of symbol
lib/ethercard-master/enc28j60.h:187: error: previous declaration of ‘typedef class ENC28J60 Ethernet’
In file included from lib/PJON-master/strategies/EthernetTCP/EthernetTCP.h:23,
                 from lib/PJON-master/PJON.h:109,
                 from .build/mega2560/src/ENCJ2860_Datalogger_House_furnaceroom.cpp:15:
lib/PJON-master/strategies/EthernetTCP/EthernetLink.h:179: error: ISO C++ forbids initialization of member ‘_receiver’
lib/PJON-master/strategies/EthernetTCP/EthernetLink.h:179: error: making ‘_receiver’ static
lib/PJON-master/strategies/EthernetTCP/EthernetLink.h:179: error: invalid in-class initialization of static data member of non-integral type ‘void (*)(uint8_t, const uint8_t*, uint16_t, void*)’
lib/PJON-master/strategies/EthernetTCP/EthernetLink.h:180: error: ISO C++ forbids initialization of member ‘_error’
lib/PJON-master/strategies/EthernetTCP/EthernetLink.h:180: error: making ‘_error’ static
lib/PJON-master/strategies/EthernetTCP/EthernetLink.h:180: error: invalid in-class initialization of static data member of non-integral type ‘void (*)(uint8_t, uint8_t)’
lib/PJON-master/strategies/EthernetTCP/EthernetLink.h:181: error: ISO C++ forbids initialization of member ‘_callback_object’
lib/PJON-master/strategies/EthernetTCP/EthernetLink.h:181: error: making ‘_callback_object’ static
lib/PJON-master/strategies/EthernetTCP/EthernetLink.h:181: error: invalid in-class initialization of static data member of non-integral type ‘void*’
lib/PJON-master/strategies/EthernetTCP/EthernetLink.h:184: error: ISO C++ forbids initialization of member ‘_local_id’
lib/PJON-master/strategies/EthernetTCP/EthernetLink.h:184: error: making ‘_local_id’ static
lib/PJON-master/strategies/EthernetTCP/EthernetLink.h:184: error: ISO C++ forbids in-class initialization of non-const static member ‘_local_id’
lib/PJON-master/strategies/EthernetTCP/EthernetLink.h:186: error: ISO C++ forbids initialization of member ‘_local_port’
lib/PJON-master/strategies/EthernetTCP/EthernetLink.h:186: error: making ‘_local_port’ static
lib/PJON-master/strategies/EthernetTCP/EthernetLink.h:186: error: ISO C++ forbids in-class initialization of non-const static member ‘_local_port’
lib/PJON-master/strategies/EthernetTCP/EthernetLink.h:189: error: ISO C++ forbids initialization of member ‘_remote_node_count’
[... 113 lines cut ...]
lib/PJON-master/PJON.h:798: error: ‘localhost’ was not declared in this scope
.build/mega2560/src/ENCJ2860_Datalogger_House_furnaceroom.cpp: At global scope:
.build/mega2560/src/ENCJ2860_Datalogger_House_furnaceroom.cpp:91: error: expected ‘,’ or ‘...’ before ‘&’ token
.build/mega2560/src/ENCJ2860_Datalogger_House_furnaceroom.cpp:91: error: ISO C++ forbids declaration of ‘PacketInfo’ with no type
lib/PJON-master/PJON.h: In instantiation of ‘PJON<SoftwareBitBang>’:
src/ENCJ2860_Datalogger_House_furnaceroom.ino:84:   instantiated from here
lib/PJON-master/PJON.h:120: error: ISO C++ forbids in-class initialization of non-const static member ‘PJON<SoftwareBitBang>::config’
lib/PJON-master/PJON.h:1041: error: ISO C++ forbids in-class initialization of non-const static member ‘PJON<SoftwareBitBang>::_auto_delete’
lib/PJON-master/PJON.h:1044: error: ISO C++ forbids in-class initialization of non-const static member ‘PJON<SoftwareBitBang>::_packet_id_seed’
lib/PJON-master/PJON.h:1045: error: ISO C++ forbids in-class initialization of non-const static member ‘PJON<SoftwareBitBang>::_random_seed’
lib/PJON-master/PJON.h:1047: error: ISO C++ forbids in-class initialization of non-const static member ‘PJON<SoftwareBitBang>::_router’
src/ENCJ2860_Datalogger_House_furnaceroom.ino: In function ‘void setup()’:
[... cut ...]
src/ENCJ2860_Datalogger_House_furnaceroom.ino:4450: error: ‘PACKETS_BUFFER_FULL’ was not declared in this scope
src/ENCJ2860_Datalogger_House_furnaceroom.ino:4456: error: ‘CONTENT_TOO_LONG’ was not declared in this scope
src/ENCJ2860_Datalogger_House_furnaceroom.ino: At global scope:
src/ENCJ2860_Datalogger_House_furnaceroom.ino:4466: error: expected ‘,’ or ‘...’ before ‘&’ token
src/ENCJ2860_Datalogger_House_furnaceroom.ino:4466: error: ISO C++ forbids declaration of ‘PacketInfo’ with no type
make: *** [.build/mega2560/src/ENCJ2860_Datalogger_House_furnaceroom.o] Error 1
Make failed with code 2
`

gioblu, 2017-02-23:
Ciao @soccou which PJON version you are using? 
Which sketch gives you this compilation result?
Thank you for your report.

soccou, 2017-02-23:
@gioblu , thanks for the quick response. After a lot of additional testing my conclusion is that since I am using a older compiler due to another library dependency that is the issue.

I can compile your lib + example with a newer version of avr-g++ OK.

gioblu, 2017-02-23:
Ciao @soccou, yes reading back the error, now I see!
Feel free to ask! (closing here)
Happy tinkering 👍

closed 2017-02-23 by gioblu

gioblu, 2017-02-24:
Ciao @soccou another user contacted me with the same issue, how you fixed that?

soccou, 2017-02-25:
I could only get it to compile with a current compiler. I think the ISO 
C++ standard changed.