gioblu/PJON #121: titles in wiki and doc lack space after ##

zcattacz, 2017-03-05:
Hi I tried to build offline document for PJON on ReadTheDocs.

Sphinx failed badly. Nearly all md pages either don't have a title or with sub titles lacking the proper space.
Also all in-line links are with absolute path pointing to this repo, or 7.0 release. 
It makes the generated doc unusable. If compiling docs on RTD will be ever considered, please considering fix these glitches. 
The generated doc locates https://readthedocs.org/projects/pjon-doc/downloads/
also the https://readthedocs.org/projects/pjon-wiki/
I wonder why they have to be separated and cross referring to each other. So neither is complete.

Thanks.

gioblu, 2017-03-05:
Ciao @zcattacz I see the problem, thank you very much for reporting. I will fix soon.
What you mean with "I wonder why they have to be separated and cross referring to each other while neither is complete." ?

You mean that a part of the wiki is inside MD files in the repo and another part is in the wiki? 
This mishap came out of two conflicting requirements, 1) documentation in MD files in the repo 2) do not duplicate information to enhance mantainability.

I would be happy to restructure it to get a better result, so if you have any suggestion feel free to propose.
Thank you again.

gioblu, 2017-03-05:
@zcattacz do you think specification dir should be included too?

gioblu, 2017-03-07:
Ciao @zcattacz thank you for reporting this and also reporting the possibility to use those tools I was ignoring so far. I have minified the graphs as you can see in the related commit to avoid the line break occurs in the pdf version. It would be nice to host on www.pjon.org the documentation and the wiki in this form: http://pjon-doc.readthedocs.io/en/latest/  it can be done? is this opensource?

gioblu, 2017-03-20:
Closing, thank you again.
Will evaluate the use of readthedocs on the website.

closed 2017-03-20 by gioblu