gioblu/PJON #123: Duplicate packets

soccou, 2017-03-23:
Every time I send a packet, the receiver gets four copies. Any idea what could cause a situation like that?

char b[sizeof(bytePacket)];

packet = bus.send_packet_blocking(PJON_RECIEVER_ID, b, sizeof(bytePacket));

gioblu, 2017-03-24:
ciao @soccou, which hardware and strategy you are using? Could you paste here the sketch?
In any case, thank you for reporting.

soccou, 2017-03-24:
the hardware is two arduino UNO devices, and the strategy is 
  PJON<SoftwareBitBang> bus(BUS_ID);

the code is at least 400 lines so is a little long.

I did try a test that maybe could shed some light. i setup two arduino on the bench and used just a short wire ~1inch with the same send/recv code and it works. only one packet as expected.

in the actual application that code with the wire about 80ft, get 4 packets for each one sent. is there some timeout that may be expiring? or a reply that needs to be sent as ACK?

gioblu, 2017-03-24:
Ciao @soccou the resistor 10k is too low, the one suggested is in the mega ohm order of magnitude, give a try to some and see which is the optimal, consider that theoretically should work also without the resistor. Yes the re-send occurs if acknowledge is not received and it could happen because the pull down with too low resistance is driving the voltage down the threshold. 25 meters should not be a range problem.

soccou, 2017-03-25:
well this is quite strange, i replaced the resistor with a 1M ohm and get the same result. just as a test i also removed a resistor entirely and get the same. receive four packets for every one sent. each is complete with no error but a duplicate.
i was afraid before i opened the issue that i had done something stupid, and it appears that is the case. 
after triple checking everything again i found that when i moved from the bench breadboard to the application location i had the wrong pin on the arduino! i had defined the bus as pin 8 in code but had it plugged into pin 9.

now i am curious how it worked at all?

gioblu, 2017-03-25:
@soccou  Yes this is quite strange, are you sure you are not shorting the soldering under the arduino with a conductive standing base?

gioblu, 2017-03-31:
Ciao @soccou have you cleared out? Can I close here?
Happy tinkering!

soccou, 2017-04-03:
Hi @gioblu, sorry for the late response. Everything is working great right now.
re: your question above, the mcu is on an old breadboard. while i don't think anything is shorted it may be possible.
Thank you for the help

gioblu, 2017-04-03:
Its a pleasure @soccou!
Happy tinkering!

closed 2017-04-03 by gioblu