gioblu/PJON #130: why put class declarations and definitions in header file?

skcks, 2017-06-02:
HI, PJON is excellent project。but I don't understand your code style after learned your all code, why put class declarations and definitions in header file? the common way is and always has been to put code in .cpp files  and declarations in headers. so what is you purpose about putting all class declarations and definitions in header file？

gioblu, 2017-06-02:
Ciao @skcks, in the case of strategies, implementation is in the header file to have those functions inlined. PJON classes are implemented in the header files because, being templates, compilation and linking problems can arise if following the standard .h / .cpp, keeping al in the .h ease compilation on different platforms with no complains. I also consider this scheme less dispersive and less prone to bugs, exposing less places where a human error can be inserted, is this choice creating technical issues to your project?

skcks, 2017-06-02:
thanks, I got you point. I had a plan to port it to stm32f0 series mcu,  so I wanner know more details as many as possible.

closed 2017-06-02 by skcks

gioblu, 2017-06-02:
CIao @skcks would be cool to have ported also stm32f0 thank you for your support to the project!

mjasny, 2019-05-09:
Sorry to dig this old issue up, but I had a similar problem when I'm using PJON is a more bigger Linux-based project.
I got the multiple definition compiler error because my header-file which included PJON.h was included my multiple other files. I suggest the following paradigm to still have the benefits of a header-only library, but also the optional possibility to have separate .cpp and .hpp files.

See this here for example.
https://github.com/mjasny/PJON/tree/master/src/interfaces/LINUX

Now in my bigger application I only have to do `#define PJON_LINUX_SEPARATE_DEFINITION` and include the .cpp files in my Makefile.

(Better would it be to have a global define for all interfaces and maybe rename the .cpp file to .inl to avoid confusion)

fredilarsen, 2019-05-10:
It is standard to have templates fully contained in header files, and the compiler/linker will make sure only one copy of each compiled template instantiation will be linked in (often template instantiation and compilation is done in the link phase).

Do you get duplicate symbols for functions in the PJON class? Or is it the non-template PJONTools class defined in PJONDefines.h that gives the problems?

mjasny, 2019-05-11:
Hi @fredilarsen ,
I did not get the error for the PJON-internal templated classes, only for the implementations of the micros(), millis() etc.. functions which are located inside the Linux-interface implementation. They are not part of any template, just global functions and therefore caused the multiple-definitions error.

gioblu, 2019-06-11:
Ciao @mjasny sorry for the late response, would you create a pull request (just the work you have done already)? Obviously I can do myself, although it could be nice to record your contribution! This fix to the LINUX interface may for sure be useful for many applying PJON in more complex linux projects.

gioblu, 2019-06-11:
Thanks for your fix @mjasny