gioblu/PJON #134: Receiver-example for RaspberryPi? (noob question)

PhinPhin, 2017-08-24:
Hi,

I'm working on a project that needs to connect 2 Arduinos to a Raspberry Pi 3. The example only features the Pi transmitting data, how can I actually send periodically requests to the Arduinos to send their data after another? Or can I use the Arduino-examples on the Pi?

gioblu, 2017-08-24:
Ciao @PhinPhin I will be back operational tomorrow, and add the example you report missing, in the meantime you can edit examples dedicated to arduino and test them on RPI adding the necessary constants toy can see in the Rpi transmitter example

PhinPhin, 2017-08-24:
That's cool, thanks :) I'm sure that will help other people who're not so much into programming, too

gioblu, 2017-09-04:
Ciao @PhinPhin here you find the example you have requested: https://github.com/gioblu/PJON/tree/master/examples/RPI/Local/ThroughSerial/BlinkTestReceiver

I hope will be useful for you and other users, sorry for being a little late :)

closed 2017-09-04 by gioblu