gioblu/PJON #138: ESP8266 Wi-Fi

viktor1970, 2017-09-23:
Hallo, I tried to make communicate a NodeMCU V1 (Wi-Fi) with an Arduino (LAN), but nothing is sent or received.
Is that communication supported?

Of course, both micros are on the same LAN and I can ping both.

Regards

gioblu, 2017-09-23:
Ciao @viktor1970 thank you for your pull request. The performance issues you have experienced have been probably solved in another branch by @fredilarsen already. We are evaluating how to adopt your changes in a portable way, to let ETCP handle also ESP trasparently without adding a new strategy related to ESP, probably simply using ifdefs (as it is done in ThroughSerial). Soon will follow up changes. If you have any suggestion, feel free to share.

gioblu, 2017-10-31:
Ciao @viktor1970 ESP is now running fine with the GlobalUDP strategy developed by @fredilarsen .
Here the example: https://github.com/gioblu/PJON/tree/master/examples/ESP8266/Local/GlobalUDP/PingPong
I close for now, if you have any question feel free to open back.

closed 2017-10-31 by gioblu