gioblu/PJON #140: Routing protocol support?

Matheus-Garbelini, 2017-09-24:
Hello, I'm seeing that pjon is growing more and more. And with IOT getting very hot nowadays, are there plannings in implementing a routing protocol to use pjon in LoRa or similar radio modules? (Right now making some tests with a LoRa strategy). I've seen that you have an over sampling strategy to use in simple radio transceiver, but would it be nice if woul be possible to create a mesh or tree based routing strategy?

If so, i would recommend implementing a RPL like routing protocol (from 6lowpan). Thanks

gioblu, 2017-09-27:
Ciao @Matheus-Garbelini yes, after the release of v9 we can start reasoning about this feature.
I think this routing / mesh networking entity should be built on top of the PJON class, having its own specification and should not be a strategy, generally strategies are used to implement a certain data link, so layer 2, PJON acts as layer 3 (or network layer). The newcoming layer should be transport, containing the routing and segmentation features. So I would suggest you, if you are testing LoRa, to implement a basic data link strategy supporting framing. With that it will be possible through PJON to build a mesh network implementing and specifying layer 3.

gioblu, 2017-11-15:
Ciao @Matheus-Garbelini how its going? Would you share your experiments with the LoRa strategy? I bought some LoRa modules, it would be nice to implement it in the next version.

Speaking instead about routing, I was considering an additional abstraction for the routing process, to support different options, because a single routing strategy would in any case not cover all the use cases. What do you think about it?

Matheus-Garbelini, 2017-11-22:
@gioblu Hello, gioblu, thanks for tour interest, I actually used a third party arduino library when wrapping PJON, is it okay to joing another library to pjon or it needs to be a clean implementation?

gioblu, 2017-11-22:
The only problem I can see could be about the license compatibility. A third party library could be included within the strategy, no problem. Another thing to consider is the library compatibility.

Matheus-Garbelini, 2017-11-22:
As I didnt create a repository yet, here's the initial files that reside in strategies folder (2 files and 2 examples)

ThroughLora.h: https://pastebin.com/nbTGzHv1
Timing.h: https://pastebin.com/hp42Ubds
Client Example: https://pastebin.com/TDpWbagA
Server Example: https://pastebin.com/3GDJmUjf
Third Party Lora Library that i used in startegy (LoRa.h): https://github.com/sandeepmistry/arduino-LoRa

My code that wraps the reading method is not optmized yet, maybe you will find sumething weird there.
Also, I didn't made a static method to access special funcitons of lora, of wich i should put inside strategy.

Matheus-Garbelini, 2017-11-22:
@gioblu Next week i will return to focus in this research of mine so i can work all day with this.
The generic routing layer is grat idea, i'd love that. None of the current iot OS implementations allow that, so it would be a great breakthrough in this.

gioblu, 2017-11-22:
The code seems quite ok! It would be nice to receive a pr from you to record your contribution.
I am now not able to test it practically so I would make some cleanup from your pr but not go much into it being not able to verify its state after my changes are done.

About the routing, I would develop this abstraction level as originally planned within [OSPREY](https://github.com/gioblu/OSPREY) where there are already some early experiments. 

It would be probably sane and nice to create a tree based routing (that is really simple) as a first step, on top of a generalized prototype that can be feeded in OSPREY like strategies are feeded to PJON. 

Then this OSPREY entity will receive PJON instances passed by user and route packets within this instances (so transport a packet from a medium to the other) and also outwards to other users (transport a packet through the network from one device to another).

Matheus-Garbelini, 2017-11-25:
@gioblu a tree based routing is perfect for LoRa application. I had looked into OSPREY and it looks very solid even at this stage.
This weekend i will submit the PR so you can review it better.
About the third party library, it uses MIT license, so I think it will can be included alongside a LoRa strategy with no problem.
I will get back to you soon, and thank you so much for your work, i've been following your work since the very first versions.

gioblu, 2017-11-25:
Thank you @Matheus-Garbelini for your passion and support to the project :).
OSPREY is actually not working it is a place where I was sketching experimentally, it should be already done but I didnt have the physical time for that too :)

gioblu, 2017-11-25:
@Matheus-Garbelini speaking about third-party libraries, they should not be included in the repository but only linked in as we did in the other strategies. Thank you again!

gioblu, 2017-12-27:
CIao @Matheus-Garbelini finally port identification arrived. Now as it is done in TCP different encapsulated protocols can be identified dynamically. This is extremely useful for encapsulating the routing protocol we were speaking about.

Matheus-Garbelini, 2018-01-01:
@gioblu Hello again in this new year :). I made an PR so you can review better the changes.
The documentation of the strategy is not yet made.
So LoRa RAW communication with PJON is made with this changes but the next steps will be applied directly to OSPREY?

gioblu, 2018-01-02:
Ciao @Matheus-Garbelini I have studied your proposal, it seems really well done (see related pr).
Yes, for routing the next step is to make OSPREY work basically supporting the PJON instances, define a port identification for it, and start the routing implementation : )

Matheus-Garbelini, 2018-01-09:
@gioblu Hello, sorry for the delay. I tested the last changes and everything seems OK, next weekend I can prepare the network examples and strategy documentation if you'd like.
Also, I finished the Sensor/Gateway node project (For Lora and/or XBEE). And as an option of cost, the MCU of choice will be STM32f103C8T6 (1 dollar each in lcsc.com). The total cost of the project including GPS/BDS is around 20 dollars per board. I'm waiting for the approval of my coordinator to buy the components.

Here's the initial rendering of the board and the mounting in raspberry pi (for gateway nodes):
[image]
[image]
[image]

gioblu, 2018-01-09:
Ciao @Matheus-Garbelini! That is a masterpiece! Really nicely done.
Is there a way to have a chat with you to define the routing protocol you would need for this?

Matheus-Garbelini, 2018-01-09:
@gioblu Thanks for appreciating the board. I didn't use Gitter before but I just created an account there and sent you a message 👍

ristomatti, 2018-01-16:
This looks awesome! @gioblu / @Matheus-Garbelini have you looked at the worked done for NRF24L01+ transceivers by the RF24/RFNetwork/RF24Mesh/RF24Ethernet devs? It seems to have a lot of similarities with the ideas you've been discussing.

gioblu, 2018-02-05:
Ciao @ristomatti @Matheus-Garbelini we are discussing here: https://github.com/gioblu/PJON/issues/167 about basic routing functionality. The change proposed by @fredilarsen of adding a constructor to PJON to also support strategies virtual inheritance seems really cool. This change could help to reduce the amount of complication needed by the routing protocol layer, supporting arrays of PJON instances with different strategies that can be used dynamically.
If you are interested to help defining this part feel free to share your point of view!
Thank you for your support.

ristomatti, 2018-02-11:
@gioblu Thanks but I doubt I'd have anything to contribute on that. Just wanted to point out the RF24 project in case some ideas could come up from the work done there. :)

gioblu, 2018-06-15:
Ciao, I will close here for now because we finally have added an initial transparent routing implementation.

closed 2018-06-15 by gioblu