gioblu/PJON #142: Support for ATtiny44/84

willie68, 2017-10-12:
Hi,
i just try to get PJON runnable on a attiny84 with arduino ide 1.8.5. 
I take the attiny85 parts and simply added sections for the attiny 84.
The following changes i have made. 
PJON_IO.h
Line 187: 
```
/* AVR ATtiny44/84 ------------------------------------ */

#if defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny84__)
  #define PJON_IO_PIN_TO_PORT_REG(P) \
    ((P >= 0 && P <= 7) ? &PORTA : &PORTB ))
  #define PJON_IO_PIN_TO_DDR_REG(P) \
    ((P >= 0 && P <= 7) ? &DDRA : &DDRB ))
  #define PJON_IO_PIN_TO_PIN_REG(P) \
    ((P >= 0 && P <= 7) ? &PINA : &PINB))
  #ifndef PJON_IO_PIN_TO_BIT
    #define PJON_IO_PIN_TO_BIT(P) \
      ((P >= 0 && P <= 7) ? P : P - 8 ))
  #endif
#endif

```
in PJON_Strategies.h modified Line 52
`#if !defined(__AVR_ATtiny45__) && !defined(__AVR_ATtiny85__) && !defined(__AVR_ATtiny44__) && !defined(__AVR_ATtiny84__)
`

in Timing.h.h modified Line 81
`#if defined(__AVR_ATtiny45__) || defined(__AVR_ATtiny85__) || defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny84__)
`
After that i get compile errors for some macros, Message: error: macro "bitWrite" requires 3 arguments, but only 1 given
`PJON_IO_PULL_DOWN(_input_pin); PJON_IO_MODE(_input_pin, INPUT);PJON_IO_WRITE(_output_pin, LOW);`
It seems that the definition from PJON_ARDUINO_Interface.h are no longer present if i switch to ATTiny 84.
If i switch to ATTiny85 the sketch compiles...
Any ideas what i have missed?

Files in here: [PJON.zip](https://github.com/gioblu/PJON/files/1378646/PJON.zip)

gioblu, 2017-10-12:
Ciao @willie68, Are you sure about the exposed constants? __AVR_ATtiny84__ ?? Is that defined when flashing it? Thank you for sharing your work on 44/84 it will be for sure useful for other users :)

willie68, 2017-10-12:
Yes, first because for the TIny85 it's the same pattern. And if i add the required macros to the PJON_IO.h ATtiny84 part, the sketch can be compiled. (At the moment i can't say, if it's working)
Now the section for the attiny44/84 looks like this:
'''/* AVR ATtiny44/84 ------------------------------------ */
#if defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny84__)
  #define PJON_IO_PIN_TO_PORT_REG(P) \
    ((P >= 0 && P <= 7) ? &PORTA : &PORTB ))shdgfh
  #define PJON_IO_PIN_TO_DDR_REG(P) \
    ((P >= 0 && P <= 7) ? &DDRA : &DDRB ))
  #define PJON_IO_PIN_TO_PIN_REG(P) \
    ((P >= 0 && P <= 7) ? &PINA : &PINB))
  #ifndef PJON_IO_PIN_TO_BIT
    #define PJON_IO_PIN_TO_BIT(P) \
      ((P >= 0 && P <= 7) ? P : P - 8 ))
  #endif
  
  #if !defined(PJON_IO_PULL_DOWN)
    #define PJON_IO_PULL_DOWN(P) \
      do { \
        digitalWrite(P, LOW); \
        pinMode(P, INPUT); \
      } while(0)
  #endif
  #if !defined(PJON_IO_MODE)
    #define PJON_IO_MODE pinMode
  #endif
  #if !defined(PJON_IO_WRITE)
    #define PJON_IO_WRITE digitalWrite
  #endif
  #if !defined(PJON_IO_READ)
    #define PJON_IO_READ digitalRead
  #endif
#endif
'''
With this changes the code can be compiled... but i'm still looking, why there is an difference between the attiny#5 and attiny#4 version.

gioblu, 2017-10-12:
Ok, it may be necessary to tweak timing constants depending on ATtiny84 performance, although as in ATtiny85 you may be lucky and not need a different timing configuration at all :). 

The use of an external 16MHz oscillator is suggested to obtain high performance. 

If you have no luck with standard timing (Duemilanove, Uno, ATtiny85) and you have an oscilloscope you can measure the bit duration and set timing constants to obtain the same duration observed in Duemilanove, Uno, Attiny85. 

Execution time overhead can be present and it is avoided with tweaking SWBB_READ_DELAY (doing so you are moving the position of the bit reading on receiver side, that should obviously be in the middle of them), the higher is the delay the higher execution overhead  delay is expected.

willie68, 2017-10-13:
The bug can be closed. I found out that the first version of my ATtiny44/84 Implementation in PJON_IO.h was buggy. Here is the right section for the PJON.h
```
/* AVR ATtiny44/84 ------------------------------------ */
#if defined(__AVR_ATtiny44__) || defined(__AVR_ATtiny84__)
  #define PJON_IO_PIN_TO_PORT_REG(P) \
    ((P >= 0 && P <= 7) ? &PORTA : &PORTB )
  #define PJON_IO_PIN_TO_DDR_REG(P) \
    ((P >= 0 && P <= 7) ? &DDRA : &DDRB )
  #define PJON_IO_PIN_TO_PIN_REG(P) \
    ((P >= 0 && P <= 7) ? &PINA : &PINB)
  #ifndef PJON_IO_PIN_TO_BIT
    #define PJON_IO_PIN_TO_BIT(P) \
      ((P >= 0 && P <= 7) ? P : P - 8 )
  #endif
#endif
```
Now i go into the timing things.

gioblu, 2017-10-13:
Ciao @willie68 I'm happy to hear that, please feel free to write me directly on gitter https://gitter.im/gioblu/PJON for direct support if needed. Good luck :)

gioblu, 2017-11-13:
CIao @willie68 I have pushed in master few changes to support 84A, although I still had no chance to get an 84 to test. Reading about it, seems that 84A works fine 84 should work fine too.
Could you give it a try? https://github.com/gioblu/PJON/commit/92663414bdc25a0c4578b50d9d7ec30a2c3fb525

willie68, 2017-11-13:
OK, thats fine. Lasdt week the pcb's for my project were delivered. So i can test it.
Yes the 84A is mainly identical to the 84, only lesser power consumption.

gioblu, 2017-11-13:
CIao @willie68, thank you in advance.

gioblu, 2017-11-21:
CIao @willie68 tests here show that all works fine with both ATtinty84 and 84A will close the issue here.
Thank you for your support :)

closed 2017-11-21 by gioblu