gioblu/PJON #148: Node.js bindings?

vshymanskyy, 2017-11-13:
I'm working a lot with different IoT systems, and for me it looks like PJON would really benefit from Node.js bindings. Are there any projects that try to achieve this, or is it planned?

gioblu, 2017-11-14:
Ciao @vshymanskyy, thank you for your support. I agree with you, absolutely. 
For now there are no projects using node.js, as far as I know.

I would say that it is planned, but, being horrified by the npm dependency system, and being a vanilla extremist, I have used it rarely, was waiting for node.js more adavanced users feedback, so help obviously would be highly appreciated if you have any will and experience with that :)

Girgitt, 2017-11-14:
@vshymanskyy I'm doing "sort of" node.js development - it's actually IBM's node-red based but the concept applies to pure node.js as well. 

This is a kind of hack but it works fine and gives some flexibility regarding the application layer. 
I'm using: 
1. Redis server as a pub/sub communication broker
2. proxy program utulizing PJON-python to interface with Redis and PJON-enabled micro over serial (e.g. Arduino connected over USB). code example is here: [https://gist.github.com/Girgitt/e490ff6d0e2fbcb4a9cf01d9bfb34373](https://gist.github.com/Girgitt/e490ff6d0e2fbcb4a9cf01d9bfb34373)

## async receive from PJON bus:
- Proxy gets PJON message and passes it to Redis over a pub channel 'pjon-python-redis'
- node.js redis client picks up the message which is a dictionary {'sender_id','receiver_id','payload_length','payload'..}

## async send to PJON bus
- Node.js client sends JSON message to pub channel 'pjon-python-redis' wiith sender_id set to the same ID as used for the pjon_redis_mock client in the proxy conde. 
- When to PJON-python redis client gets a message sent with its own sender id it forwards it to PJON-python serial client which passes it to microcontroller over USB/serial. 

You can run multiple proxy programs connected to the same Redis server. Depending on the sender_id from the message sent to Redis a proper proxy client will pass the message to PJON-python serial client. Lets suppose you have two micros you want to communicate with:

Arduino(id=101) ---<USB>--- proxy[PJON-python serial(id=100) <-> PJON-python redis(id=100)] --<Redis>-- Node,js

Arduino(id=201) ---<USB>--- proxy[PJON-python serial(id=200) <-> PJON-python redis(id=200)] --<Redis>-- Node,js
 
to send payload to node 101 the JSON dict passed to redis should have sender_id set to 100 and receiver id set to 101

to send payload to node 201 the JSON dict passed to redis should have sender_id set to 200 and receiver id set to 201

I'll make a better description how to use PJON-python for such integration when time allows.

gioblu, 2017-11-14:
Ciao @Girgitt, thank you for your feedback, I still have to try your setup!

It seems that bindings would simplify the integration process, potentially enabling PJON to run over serial or TCP/UDP directly in node!

Girgitt, 2017-11-14:
@gioblu sure they would simplify the stack but adds additional module to keep updated. Using Redis for communication opens many programming languages and environments and seems a good solution for now as only PJON-piper needs to be kept compatible with PJON (and preferably should be extended with TCP and UDP strategies).

gioblu, 2017-11-14:
Ciao @Girgitt absolutely, I will run some tests of the redis server, thank you for your support!
I agree with you that the setup you have designed could cover a lot of use cases.

I agree also with @vshymanskyy considering handy to have a direct interface from node to the c++ implementation. I will study bindings.

vshymanskyy, 2017-11-14:
@gioblu 
https://medium.com/the-node-js-collection/n-api-next-generation-node-js-apis-for-native-modules-169af5235b06

Girgitt, 2017-11-14:
@vshymanskyy that looks very promising. If such bindings could be automatically compiled from PJON sources - I'm all for it.

vshymanskyy, 2017-11-14:
@Girgitt @gioblu i don't think it can auto-generate something, but it should be rather simple to wrap your library with N-API.
the hard part may be to provide proper Node objects (that act the Node way). But usually node community helps with that...

gioblu, 2017-11-14:
Ciao @vshymanskyy @Girgitt probably without changing method names or parameters a version update should go through transparently? This seems really cool, thank you @vshymanskyy

vshymanskyy, 2017-11-14:
@gioblu exactly. That's what they promise, at least ;)
I'm sorry I can't help you with coding currently - too much stuff is going on ;(

gioblu, 2017-11-15:
Ciao @vshymanskyy dont be preoccupied :), compliments for your work and thank you for your feedback.

ristomatti, 2018-01-16:
Just found this thread and I've been searching for a clean solution for Arduino <-> Node.js (running on a Raspberry Pi) bi-directional communication. Has anyone taken the challenge yet? The idea is so good I might try to hack together some prototype if I only just find the time. I'm familiar with the NPM ecosystem, have been planning on trying N-API and PJON (originally for ESP8266 <-> AVR communication) for some time now. I've only got half a year of C/C++ experience through hobby projects though so I'm hoping no one is holding their breath waiting for this to happen. :roll_eyes:

I'll sure to post here in case I manage to pull something off.

gioblu, 2018-01-16:
Ciao @ristomatti I still have no physical time to work on this issue, any support would be highly appreciated :) Thank you for your support to the project.