gioblu/PJON #160: SoftwareBitBang  short circuit risk

filogranaf, 2017-12-29:
When using only one cable, can short circuits occur when two devices try to write different bits at the same time? Should not some transistors be used or do the pull down resistor be taken to bring the voltage to zero without setting the output pin directly to zero?
Thanks in advance

gioblu, 2017-12-29:
Ciao @filogranaf the pull-down resistor is there to ground potential induced interference bringing the channel high (it must be low for other devices to start transmission). Before the transmission of a packet there is a polling timeframe where the bus state is checked and transmission starts only if bus state was always low. After polling the transmitter sends 3 synchronization bits where it also checks 3 times for a potential collision. Furthermore for each transmitted byte a collision check timeframe is present to abort transmission if a collision, or interference occurred. Collision and so short circuits can always occur but the chances of their occurrence and their duration is minimized by the scheme described above. Extended tests demonstrate that this procedure can be applied in production for long periods of time with high reliability.

filogranaf, 2017-12-30:
Thanks for reply.
To avoid any risk of damage it would not be enough to avoid setting the output pin to 0, allowing the pull down resistor to act? in the code it would be sufficient not to use digitalWrite (pin, 0) and set the pin in read mode to make the pull down rest act. This would avoid the condition 2 described in https: //www.rugged-circuits.com/10-ways-to-destroy-an-arduino/

gioblu, 2017-12-30:
CIao @filogranaf I have never seen a single pin failure in 7 years of testing and I am not aware of any user have experienced this issue although really complex setups are present worldwide. I know about the article you have linked and the points it describes.

It could be possible to switch to input low, instead of output low but that would influence binary timing, some experiments should be made. Thank you for your reply.

filogranaf, 2017-12-30:
One last note ... the reading mode for 0 could be used to detect conflicts immediately (it must remain at 0 for the duration of the bit) and immediately stop the transmission without waiting for the ack.

gioblu, 2017-12-30:
ciao @filogranaf I have explored the possibility of switching to input while transmitting, in my tests I have noted that this change adds an architecture related timeframe that brings much more difficult to let all supported mcus to effectively communicate with acceptable accuracy. But I will try again in the next iteration :).

Obviously, without changing procedure, it could be added a current limiting resistor to each pin feed to reduce the maximum current drain to 30ma.

gioblu, 2017-12-31:
Ciao @filogranaf thank you for pointing out, added the missing documentation about the optional use of current limiting resistors. I have tested practically over a 20m bus, 180Ω resistors on each pin end are not affecting transmission reliability. I have tested also with 1KΩ and 15KΩ on 20m range and it also works fine as well. So I expect the practical maximum range to be well beyond 50 meters using 180Ω current limiting resistors,

gioblu, 2017-12-31:
Closing, feel free to reopen if you consider that necessary.
Happy new year 👍 and thank you for your support.

closed 2017-12-31 by gioblu

filogranaf, 2017-12-31:
Very good.
Thanks and happy new year