gioblu/PJON #161: Raspberry SoftwareBitBang

gingo, 2017-12-31:
Hi!

PJON is really great library. I build my own simple protocol based on it for my smart-home solution. It works great for Arduino <-> Arduiono communication, but actually I need to connect all my Arduino devices to central Raspberry. I thought the best way to do this could be to connect every Arduino to one digital pin on Raspberry and talk to them via SoftwareBitBang. 

Is there any significant problem with SBB on Raspberry? Or is only about playing with Timing.h constants?

Any other idea how to connect all my Arduino devices to central Raspberry Pi?

gioblu, 2017-12-31:
Ciao @gingo for now SWBB is not compatible with RPI, but yes it is a matter of testing and timing definition / tweak. Another way to accomplish it is to use ThroughSerial and connect a SWBB compatible device to RPI using a TS connection and to other devices using a SWBB connection (a sort of gateway). Here the router example: https://github.com/gioblu/PJON/blob/master/examples/ARDUINO/Network/SoftwareBitBang/RecursiveAcknowledge/Router/Router.ino 

Thank you for your compliments and happy new year :)

gingo, 2017-12-31:
Is there any guide how to tweak timing for new device? Is better to test RPi <-> RPi, or RPi <-> Arduino to test with already compatible device?

gioblu, 2017-12-31:
Ciao @gingo, start using the standard timing and test using `SpeedTest` example using the Arduino Uno on receiver side (able to print connection quality via serial monitor), see if RPI is able to transmit data to Uno. If does not work an oscilloscope could be useful to measure binary duration discrepancies of the two devices. For example if RPI bits are slightly longer, its related timing constant should be slightly reduced to match the desired duration in real life. If does work try the `NetworkAnalysis` example that outputs communication state on transmitter side, using the transmitter sketch on RPI side. If does work you are done, if does not work `SWBB_READ_DELAY` may need to be tweaked or `SWBB_ACCEPTANCE` reduced.

Matheus-Garbelini, 2018-01-02:
@gingo as @gingo is everything a matter of timing with SoftwareBitbang. Unfortunattely, if you are running raspbian on raspberry pi is almost impossible to get an accurate timing (unless you do busy waiting). If you really need to run PJON on raspberry pi for reasons like reducing the cost of the project you should consider a baremetal approach like circle: [https://github.com/rsta2/circle/tree/eclipse-support](url)

Circle currently does not have a PJON port but support ethernet,hdmi output,i2c,spi for your smarthome project. As you have control over each core of raspberry pi and also it's timer, it would be not difficult to port PJON over a framework like that. 

In short, if your project does not need wifi (for now) and are not running a very complex webserver that can't be ported to C++, circle is a good alternative to programming raspberry pi as you have an instant boot (less then 2 seconds to boot to yout application) and makes the raspberry pi pratically a microcontroller, so you don't have user and kernel space bottering you, just acces peripherals directly like arduino (Currently the framework uses it's own library, but in future it may have an arduino support so you can use the arduino ide).

gingo, 2018-01-02:
@Matheus-Garbelini I need Raspbian on my Pi :)

So recommended approach is to use another Arduino as router? Can I connect them via USB serial line?

gioblu, 2018-01-02:
Ciao @gingo, I am not sure about the feasibility of RPI strict timing bit banging. I have tested before the last two iterations to generate a squarewave and that was pretty stable and near to the requested form. I agree with @Matheus-Garbelini that there are some chances that the real time operative system interruptions could make SWBB on RPI less reliable or effective, but it may be worth a try.

Yes RPI can speak with an Arduino using ThroughSerial over a usb connection!