gioblu/PJON #167: Router functionality

gingo, 2018-02-04:
I really like idea behind buses, because it allows us to do efficient routing. Following [this example](https://github.com/gioblu/PJON/blob/master/examples/ARDUINO/Network/SoftwareBitBang/RecursiveAcknowledge/Router/Router.ino) man can fast build simple routing functionality. 

Is there any reason why this is not part of library? I would be much more happier to write only something like: 

```c
void setup() {
  ...
  busA.routeTo(busB);
  busB.routeTo(busA);
  ...
}
```

With all possible bugs, network detection etc. solved.

gioblu, 2018-02-05:
Ciao @gingo I agree with you that an easy way to handle routing would be a nice to have.
PJON architecture has been chosen to limit program size, for that reason has been chosen to use templates instead of virtual inheritance, One thing proposed by @fredilarsen that seems a solution is to provide an additional constructor like:
```cpp
StrategyLink<SoftwareBitBang> strategy;
PJON bus(strategy, id);
```
could support something ike:
```cpp
StrategyLink<SoftwareBitBang> strategyA;
PJON busA(strategyA, idA, bus_id_A);

StrategyLink<OverSampling> strategyB;
PJON busB(strategyB, idB, bus_id_A);

StrategyLink<GlobalUDP> strategyC;
PJON busC(strategyC, idC, bus_id_C);

busA.bidirectional_routing(busB);
busA.bidirectional_routing(busC);
busB.bidirectional_routing(busC);

// or 

busA.route_to(BusB); // Mono directional routing
```
This would be possible because all PJON objects here implement PJON<StrategyLinkBase>.
So a common interface class, enabling to handle dynamically PJON instances, for example have an array of PJON instances using different strategies and use them agnostically.

At the same time the normal procedure of PJON could be still untouched where static declaration is needed: 
```cpp
PJON<SoftwareBitBang> busA(idA, bus_id_A);
```
Probably this is a change that will need some testing and experimentation, so I think it cannot be fitted in 10.1 that I would release as soon as possible.

gingo, 2018-02-05:
I have another idea, how to deal with routing. It's not crucial feature for normal users and even advanced users need it only on particular nodes. Lets build it as independent extension. First we need some `Routable` interface:

```c
class Routable {
  // PJON methods needed for routing
}
```

Lets create `RoutablePJON`:

```c
class RoutablePJON<Strategy>: public PJON<Strategy>, public Routable {

}
```

Now we can create `Router` with all the routing functionality and pack it all in PjonRouter.h to link it only when needed. We can start with manual routing functionality and maybe later create also switch functionality, where all the devices are on the same network and switch knows where they are based on some PJON_HELLO message send when device is up.

gioblu, 2018-02-05:
Ciao @gingo it seems a reasonable suggestion, absolutely, firstly because like this you correctly avoid mixing functionalities, and also in terms of memory footpring, although I should still make some practical test.

I would release tonight 10.1 including the last fixes.

If you have a working example feel free to send a pull request, we can start experimenting from there.
About starting with manual routing I absolutely agree with you.

gingo, 2018-02-06:
Unfortunately I don't have any example. I can start working on it maybe this evening. I'll let you know about eventual progress.

gioblu, 2018-02-06:
Ciao @gingo thank you for your support.
I have released only now 10.1, because of a last round of tests.

I look forward to work on your proposal as soon as possible, it is for sure a must-have and also how you proposed to implement it seems really flexible. Thank you again.

gioblu, 2018-02-10:
Ciao @gingo here is the virtual inheritance support: https://github.com/gioblu/PJON/commit/dd7ad14de8d03fa589199c5537c4a2b75b477e8d
and here a couple of working examples:
https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Local/Any

Feel free to share your point of view over this. 
Seems not affecting performance, it also adds a negligible program memory overhead, so seems not bad, though still experimental.

gioblu, 2018-03-29:
Ciao @gingo, I will close here, feel free to reopen if needed.
Thank you for your support :)

closed 2018-03-29 by gioblu