gioblu/PJON #170: PJON on additional serial port

paulnd, 2018-02-19:
Hello,  apologies if I have missed this in the documentation but how do I use PJON only on an additional serial port on a Mega?

I plan to connect the Mega to a PC via the USB (serial 1, non-PJON) and then to other Arduinos using PJON/RS485 via serial 2, I hope that this is possible?

Hope that I've explained it clearly and thanks for developing this protocol. 

Paul

gioblu, 2018-02-20:
Ciao @paulnd 
If I undestood correct:

```cpp

PJON<ThroughSerial> bus1;
PJON<ThroughSerial> bus2;

void setup() {
  Serial.begin(9600);
  bus1.strategy.set_serial(&Serial);
  Serial1.begin(9600);
  bus2.strategy.set_serial(&Serial1);
}

```
On the Mega side you can instantiate more than one PJON instance using the data link you need.
Consider that PJON runs also on a computer, so maybe the serial non-PJON side, could also use PJON :)

Happy tinkering :)

paulnd, 2018-02-20:
Perfect, thank you.

I just wasn't sure how to instantiate PJON on serial1.

Unfortunately the PC application is 3rd party and uses a very basic data stream to communicate.

Paul

closed 2018-02-20 by paulnd