gioblu/PJON #173: MultiByte char instead unicode in Visual Studio community 2017

sigmaeo, 2018-02-22:
I just started with pjon and compiled the transmitter example on windows 7 with VS C++ community 2017.
First I have to change project settings to "multibyte" charset instead "unicode". Maybe this can add to comments somewhere?
Second I have to change the line 38 in Serial.cpp from CreateFile() to CreateFileW().
With this it compiles and work fine.

gioblu, 2018-02-23:
Ciao @sigmaeo thank you for reporting.
Will test and fix the issue as soon as possible.

gioblu, 2018-03-25:
Ciao @sigmaeo thank you for reporting the broken example, I have fixed as you suggested.
I have also added the vcxproj files and updated the readme to reduce the amount of steps required to have the example running. Thank you again for your support to the project.

closed 2018-03-25 by gioblu