gioblu/PJON #176: Many Fail (no answer from receiver) ~400 per sec

Tintin4000, 2018-02-23:
I am using PJON for the first time in a real application.  Transmitter, Pro Mini 8MHz connected via Serial Hardware to HC-12, on the receiver side, ESP8266 with Serial Hardware to the other HC-12.  While working with the Through Serial Strategy, I noticed a huge amount of Fail (no answer from receiver), around 400 per sec.  I have been trying to play with a few parameters but nothing seems to affect that value.  Any idea?

gioblu, 2018-02-23:
CIao @Tintin4000, thank you for reporting.
Generally in the examples SoftwareSerial is used instead, have you tried with it?
Could you please share the code?

Are packets received by ESP? 
What happens disabling the acknowledge?
Have you tried to higher `TS_RESPONSE_TIME_OUT` on transmitter side?

Thank you again for your support.

gioblu, 2018-02-23:
@Tintin4000 thank you for editing adding the test results.
Depending on which side is printing this output it may not be packet transmission error.
If that is amount of failures in receiving a message in the receiver sketch (so the amount of calls to bus.receive() that returned PJON_FAIL), that may be related to delays in the transmitter sketch (if transmitter never transmits the number of fails is equal to the number of reception attempts).

Tintin4000, 2018-02-23:
I am using a one way communication for this test, meaning that the receiver, the ESP8266 is the one printing the output.  The setup for the HC-12 is the default one, which if I am correct is 9600 Bd.  I have modified the exemple to store the value in an array instead of printing during reception, this has not make any difference so I assumed that there is no interference.  Somehow expected since the RX is used for the test and the TX for the printing.  Note as well that I have configure both unit in SIMPLEX mode.
I have tried to implement the error callback on the transmitter side but that do not work at all, loop is never executed?  How can I verify on the transmitter side that the packet is delivered correctly?

gioblu, 2018-02-23:
Ciao @Tintin4000, the error on transmitter side is not called because no error occurred I suspect. If a bus.receive() call ends up in PJON_FAIL that does not mean a transmission failed, but simply that the reception attempt failed (it may happen that no information has been transmitted in the first place in that moment).

For transmitter side see:

https://github.com/gioblu/PJON/blob/master/examples/ARDUINO/Local/SoftwareBitBang/BlinkTest/Receiver/Receiver.ino

https://github.com/gioblu/PJON/blob/master/examples/ARDUINO/Local/SoftwareBitBang/NetworkAnalysis/Transmitter/Transmitter.ino

Tintin4000, 2018-02-23:
How does it differs from the ThroughSerial strategy?

gioblu, 2018-02-24:
Ciao @Tintin4000 Im sorry, probably is late here, heeh
You mean you get this result without having a receiver device connected (transmitter detects an acknowledge when should not)?

Running the second SWBB example linked without any receiver connected here leads to:
Packet Overhead: 7B - Total: 0B
Maximum Bandwidth: 0B/s
Data throughput: 0B/s
Packets sent: 0
Mistakes (error found with CRC): 0
Fail (no answer from receiver): 55
Busy (Channel is busy or affected by interference): 0
Accuracy: nan %
---------------------

gioblu, 2018-02-25:
Ciao @Tintin4000 I get your point. I would try to discover the baudrate at which you are transmitting, so to then be able to estimate if there is a bandwidth loss. The fails could be the number of tries of receive call between every packet, which distance in time depends on the baudrate. If this is happening the expected behaviour is: the higher the baudrate the lower the numbers of fails

closed 2018-03-23 by gioblu