gioblu/PJON #180: esp8266 error compiling

andrepalma, 2018-02-24:
Hi,
I'm trying this library and i'm not getting to compile even the example showed for ESP8266..
it gives me this error:

.platformio\lib\PJON_ID245/strategies/EthernetTCP/EthernetLink.h: In member function 'int16_t EthernetLink::read_bytes(TCPHelperClient&, uint8_t*, uint16_t, uint16_t)':
.platformio\lib\PJON_ID245/strategies/EthernetTCP/EthernetLink.h:316:63: error: no matching function for call to 'min(int, uint16_t&)'
(uint32_t)(millis() - start_ms) < min(1000, timeout_ms)

Anyone has any idea?

Thank you

gioblu, 2018-02-24:
Ciao @andrepalma I have checked it seems there was no general constant check (ESP8266), this may fix the problem. Will push and relate the commit to this issue. Thank you for reporting.

andrepalma, 2018-02-25:
Thank you for your very quick response... :)

Now it compiles and allow to upload.

Another thing that I'm facing is I was testing the communication between 2 ESP (so 3.3V) with one wire and I was using the ESP8266\Local\SoftwareBitBang\SpeedTest example for transmistter and receiver and I'm not getting communication successful.. This example should work for this type of interfacing?

Thank you

gioblu, 2018-02-25:
Ciao @andrepalma sorry I fell asleep, could you link here the code you are using?
Thank you again for reporting :)

andrepalma, 2018-02-25:
Hello gioblu,

Here is the code, are the code from examples for speedtest that you have on github examples, the only thing i change is the pin connected on receiver as the original is pin is for an arduino (I believe)

Thank you.

transmitter example:
`/* ESP8266 - Node MCU side
   Devices can be wired directly being ESP8266 5v tolerant */

#include <PJON.h>

// <Strategy name> bus(selected device id)
PJON<SoftwareBitBang> bus(45);

char content[] = "01234567890123456789";

void error_handler(uint8_t code, uint8_t data) {
  if(code == PJON_CONNECTION_LOST) {
    Serial.print("Connection lost with device id ");
    Serial.println(bus.packets[data].content[0], DEC);
  }
};

void setup() {
  bus.strategy.set_pin(5); // ESP8266 GPIO 5 or pin D1
  bus.set_error(error_handler);
  bus.begin();

  Serial.begin(115200);
}

void loop() {
  if(!bus.update())
    bus.send(44, content, 20);
};`

receiver code:
`/* Arduino compatible device
   Devices can be wired directly being ESP8266 5v tolerant */

#include <PJON.h>

float test;
float mistakes;
int busy;
int fail;

// <Strategy name> bus(selected device id)
PJON<SoftwareBitBang> bus(44);

void receiver_function(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info) {
 // Do nothing to avoid affecting speed analysis
}

void setup() {
  bus.strategy.set_pin(5);
  bus.begin();

  bus.set_receiver(receiver_function);

  Serial.begin(115200);
};

void loop() {
  Serial.println("Starting 1 second communication speed test...");
  long time = millis();
  unsigned int response = 0;
  while(millis() - time < 1000) {
    response = bus.receive();
    if(response == PJON_ACK)
      test++;
    if(response == PJON_NAK)
      mistakes++;
    if(response == PJON_BUSY)
      busy++;
    if(response == PJON_FAIL)
      fail++;
  }

  Serial.print("Packet Overhead: ");
  Serial.print(bus.packet_overhead(bus.last_packet_info.header) + 1);
  Serial.print("B - Total: ");
  Serial.print((unsigned int)((bus.packet_overhead(bus.last_packet_info.header) + 1) * test));
  Serial.println("B");
  Serial.print("Absolute com speed: ");
  Serial.print(test * (20 + bus.packet_overhead(bus.last_packet_info.header) + 1));
  Serial.println("B/s");
  Serial.print("Practical bandwidth: ");
  Serial.print(test * 20);
  Serial.println("B/s");
  Serial.print("Packets sent: ");
  Serial.println(test);
  Serial.print("Mistakes (error found with CRC) ");
  Serial.println(mistakes);
  Serial.print("Fail (no answer from receiver) ");
  Serial.println(fail);
  Serial.print("Busy (Channel is busy or affected by interference) ");
  Serial.println(busy);
  Serial.print("Accuracy: ");
  Serial.print(100 - (100 / (test / mistakes)));
  Serial.println(" %");
  Serial.println(" --------------------- ");
  // Avoid Serial interference during test flushing
  Serial.flush();

  test = 0;
  mistakes = 0;
  busy = 0;
  fail = 0;
};`

gioblu, 2018-02-25:
Ciao @andrepalma I have tested using the Arduino IDE both:
- [SpeedTest](https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Local/SoftwareBitBang/SpeedTest)
- [NetworkAnalysis](https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Local/SoftwareBitBang/NetworkAnalysis)

Using pin 5 or D1 on both ESPs and it works fine:

Starting 1 second communication speed test...
Packet Overhead: 10B - Total: 580B
Absolute com speed: 1740.00B/s
Practical bandwidth: 1160.00B/s
Packets sent: 58.00
Mistakes (error found with CRC) 0.00
Fail (no answer from receiver) 5712
Busy (Channel is busy or affected by interference) 0
Accuracy: 100.00 %

For testing I have used 2 NodeMCU 1.0 with this board configuration:
- NodeMCU 1.0 (ESP-12E Module)
- Upload using serial
- CPU frequency 80MHz
- Flash size 4M (3M SPIFFS)
- Upload speed 115200

I suspect the problem could be:
- The use of a different CPU frequency (supported 80MHz only for now)
- Different compilation process in use by PlatformIO (for example code optimization flags that affect computation time)

WIll follow further tests using PlatformIO!

andrepalma, 2018-02-25:
Hi gioblu, Thank you so much for your support.
I will give a try with arduino IDE later today, as now I'm in another project for shutters :)

You don't like of platformio IDE?

About this wire communication, Have you try any time long distances? llike 3 or 4 meter long?

Thank you.
Regards

gioblu, 2018-02-25:
Ciao @andrepalma I am forced to test on Arduino IDE because PJON for now is mainly used with the Arduino toolchain, although recently we have added support to WINX86 VIsual Studio and simply make on Linux.

SoftwareBitBang data link has been extensively tested with wire length of 1-10-25-50 meters length.
It works fine using bare insulated wire or Ethernet cable. It works fine also if used with a star configuration. 3v3 voltage level maximum range has not been still verified.

andrepalma, 2018-02-26:
Hi gioblu, as i said during the afternoon I now accoumplish to test the communication testing for Speedtest so far.. During this week I will do more tests to assure the communication is good to my smarthome project.

Thank you so much for all your support and effort. 
Excellent work with this repo.
Best regards

closed 2018-02-26 by andrepalma

gioblu, 2018-02-26:
Ciao @andrepalma thank you for your compliments.
I have marked this issue as an inconsistency because clearly there may be some problem with PlatformIO, looking forward to test in more detail. Feel free to reopen if needed.