gioblu/PJON #185: ds18b20 arduino Pjon

Prutser81, 2018-03-05:
HI

 Found a sketch on the arduino forum works perfectly with pjon 2.0.
Have tried to change the sketch but without a good working result.
How do I change the sketch that it works well with  pjon 10.1 library.

#include  <PJON.h> //Libreria di GioBlu

// network(Arduino pin used, selected device id)
//PJON network(12, 99);

// Bus id definition
uint8_t bus_id[] = {0, 0, 0, 1};

// PJON object
PJON<SoftwareBitBang> bus(bus_id, 99);

void setup() {
  
  bus.strategy.set_pin(12);

  bus.set_receiver(receiver_function);
  bus.begin();
  
  Serial.begin(115200);
  

  //network.set_error(error_handler);
  //network.set_receiver(receiver_function);

  bus.send(1, "*", 1, 10000000); //CHECKING EVERY 10sec. if the id1 responds
  delay(200);
  bus.send(2, "*", 1, 10000000); //CHECKING EVERY 10sec. if the id2 responds
  delay(200);
  bus.send(3, "*", 1,  10000000); //CHECKING EVERY 10sec. if the id3 responds

 // network.send(1, "*", 1, 10000000); //CHECKING EVERY 10sec. if the id1 responds
 // delay(200);
 //network.send(2, "*", 1, 10000000); //CHECKING EVERY 10sec. if the id2 responds
 // delay(200);
 // network.send(3, "*", 1, 10000000); //CHECKING EVERY 10sec. if the id3 responds

};

void loop() {

  bus.receive(1000);
  bus.update();

  //network.receive(1000);
  //network.update();
} //fine loop

 
//static void receiver_function(uint8_t length, uint8_t *payload) {
static void receiver_function(uint16_t length, uint8_t *payload) {

  String data;

  // I create a string from the received data
  for (int i = 0; i <length; i++) {
  Serial.print((char)payload[i]);
    char tmp =  payload[i];
    data += tmp;
  }
  if (data.startsWith("T_1")) {
    Serial.print("sensor1");
  }
  else if (data.startsWith("T_2")) {
   Serial.print("sensor2");
  }
  else if (data.startsWith("T_3")) {
   Serial.print("sensor3");
  
  }

 
  //"nome_sensore/sp/dati"
 
  int pos_spazio = data.indexOf(' ');               //trovo la pos. dello spazio
  String tmp = data.substring (pos_spazio);     //Temperatura
  String nome_sen = data.substring (pos_spazio, 0); //Nome sensore
  Serial.println ("sensor Name: " + nome_sen);
  Serial.println ("Temperature: " + tmp);
}

//se perde la connessione con qualche device lo visualizzo
static void error_handler(uint8_t code, uint8_t data) {

  //if (code == CONNECTION_LOST) {
   // Serial.print("Connection lost with device id ");
    //Serial.println(data);
    
    }
//    }

-------------------------------------------------------------------------------------------------------------------
// Include the libraries we need
#include <OneWire.h>
#include <DallasTemperature.h>
#include <PJON.h> //Libreria di GioBlu

// network(Arduino pin used, selected device id)
//PJON network(12, 3);

// Bus id definition
uint8_t bus_id[] = {0, 0, 0, 1};

// PJON object
PJON<SoftwareBitBang> bus(bus_id, 3);

int Invio_Sec = 2000;                   //SET EVERY SECOND SEC. SEND THE DATA
//int PinTrasmissione = 99;               //RECEIVER ID
float floatValTemp=0;                   //INITIALIZE VARIAB. TEMPERATURE           
String NameSensor = "T_3 ";  //NAME SENSORE (attenzione allo spazio finale)
String stringVal = "";                  //VAR. SENSOR READING STRING

//SETTING SENSORE
//data pin sensore 
#define ONE_WIRE_BUS 4
OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);
// arrays to hold device address
DeviceAddress insideThermometer;
//FINE SETTING SENSORE

//INITIALIZE TIME
unsigned long time = millis();

 
void setup(void){

  bus.strategy.set_pin(12);
 
  //bus.set_receiver(receiver_function);
  bus.begin();
  
  // start serial port
  Serial.begin(9600);
  pinMode(13, OUTPUT);
  
  //PJON
 bus.set_error(error_handler);
 
 // network.set_error(error_handler);
 // network.set_receiver(receiver_function);
  
  //search for temperature sensor address 
  Serial.print("I look for sensors...");
  sensors.begin();
  Serial.print("Found/i ");
  Serial.print(sensors.getDeviceCount(), DEC);
  Serial.println(" devices.");

  if (!sensors.getAddress(insideThermometer, 0)) Serial.println("No devices found"); 
  Serial.print("Device address: ");
  printAddress(insideThermometer);
  Serial.println();
  // set 12bit sensor resolution
  sensors.setResolution(insideThermometer, 12);
  Serial.print("Device 0 resolution: ");
  Serial.print(sensors.getResolution(insideThermometer), DEC); 
  Serial.println();
  //Serial.flush();
}

void loop(void)
{ 
  
  
  //lettura sensore ogni X sec.
  if(millis() - time > Invio_Sec) {
    time = millis();
    //read the value of the sensor ds18b20
    stringVal=String(NameSensor + double(sensors.getTempCByIndex(0)));
    int lun=stringVal.length();
    char charVal[lun];
    stringVal.toCharArray(charVal,lun);

    bus.send(99, charVal, lun); 
    //network.send(99, charVal, lun);
    sensors.requestTemperatures(); // Send the command to get temperatures
     //debug
    Serial.print ("STRINGVAL:  ");
    Serial.println(stringVal);
    Serial.print("LUN:  ");
    Serial.println(lun);
    //lunhezza = lengte string
  }
  
  bus.update();
  bus.receive(1000);
  //network.update();
  //network.receive(1000);
  
}

// function to print a device address
void printAddress(DeviceAddress deviceAddress)
{
  for (uint8_t i = 0; i < 8; i++)
  {
    if (deviceAddress[i] < 16) Serial.print("0");
    Serial.print(deviceAddress[i], HEX);
  }
}
void error_handler(uint8_t code, uint8_t data) {
  if(code == PJON_CONNECTION_LOST) {
    Serial.print("Connection with device ID ");
    Serial.print(bus.packets[data].content[0], DEC);
    Serial.println(" is lost.");

//static void error_handler(uint8_t code, uint8_t data) {
  //if(code == CONNECTION_LOST) {  //se non c'è connessione lampeggia il led
   // Serial.print("Connection lost with device id ");
    //Serial.print(data);
    digitalWrite (13,HIGH);
    delay (2000);
  }
  digitalWrite (13,LOW);
}
static void receiver_function(uint16_t length, uint8_t *payload) {
    Serial.println ("Received data from Master 99");
    String data;
    
    Serial.print ("length: ");
    Serial.println (length);
    
    for (int i = 0; i < length; i++){
      char tmp=payload[i];
      data+=tmp;
    }
}

gioblu, 2018-03-07:
Ciao @Prutser81 the only difference has been introduced after 10.1 in unreleased state (actual master) and is the error handler that has now an additional parameter.

```cpp
// IN 2.0
static void error_handler(uint8_t code, uint8_t data) {

}
// AFTER 10.1
static void error_handler(uint8_t code, uint8_t data, void *custom_pointer) {

}
```
Adding the missing parameter both sketches are compiled correctly, although I now dont have the physical time to run a proper test run.

gioblu, 2018-03-08:
Ciao @Prutser81 closing for now, feel free to reopen if required. 
Happy tinkering :)

closed 2018-03-08 by gioblu

Prutser81, 2018-03-11:
Hi gioblu,

With pjon-10.1 the length is 587 it must be 9. Do you have any idea what the cause is. I add this line to the sketch ( Serial.print((char)payload[i]); ). I can read temperature but with many unreadable values.

With PJON-2.0					
serial monitor :

Temperature:  22.1
length: 9
sensore3
Sensor Name: T_3
Temperature:  22.1
Connection lost with device id 1
Connection lost with device id 2

With PJON-10.1
serial monitor :

Temperature: 
length: 587
sensor Name: I⸮
Temperature: 
Connection with device ID 1 is lost.
Connection with device ID 2 is lost.

gioblu, 2018-03-11:
Ciao @Prutser81 thank you for reporting.
what is the value of  `int lun=stringVal.length();` before passing it to PJON? Is that 9?

gioblu, 2018-03-11:
I have noticed another deprecated definition: change in the receiver callback definition:

```cpp
// IN 2.0
void receiver_function(uint8_t code, uint8_t data) {

}

// AFTER 10.1
void receiver_function(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info) {

}
```

gioblu, 2018-03-11:
Ciao @Prutser81 I have run some tests. Length value reported is always correct, that seems quite strange, also because the maximum accepted length by default or `PJON_PACKET_MAX_LENGTH` is set to 50 bytes.
So by default a packet of more than 500 characters should be dropped and not received.

Prutser81, 2018-03-11:
lun lenght is 9

I look for sensors...Found/i 1 devices.
Device address: 28C5BC2005000041
Device 0 resolution: 12
Received data from Master 99
length: 857
STRINGVAL:  T_3 21.88
LUN:  9
STRINGVAL:  T_3 21.88
LUN:  9
STRINGVAL:  T_3 21.88
LUN:  9

gioblu, 2018-03-11:
@Prutser81 have you changed or configured `PJON_PACKET_MAX_LENGTH` ? 
If not I do not see how a packet of 857 characters could be received:
https://github.com/gioblu/PJON/blob/master/src/PJON.h#L460

Which boards you are using for testing?

Prutser81, 2018-03-11:
I use the board arduino nano

gioblu, 2018-03-11:
@Prutser81 please can you post here the actual sketches? I will run some tests.

Prutser81, 2018-03-11:
receiver
``
#include <PJON.h> //Libreria di GioBlu

// network(Arduino pin used, selected device id)
//PJON network(12, 99);

// Bus id definition
//uint8_t bus_id[] = {0, 0, 0, 1};

// PJON object
//PJON<SoftwareBitBang> bus(bus_id, 99);
PJON<SoftwareBitBang> bus(99);

void setup() {
  
  bus.strategy.set_pin(12);
  bus.set_error(error_handler);
  bus.set_receiver(receiver_function);
  bus.begin();
  
  
  Serial.begin(115200);
  
  

  //network.set_error(error_handler);
  //network.set_receiver(receiver_function);

  bus.send_repeatedly(1, "*", 1, 10000000); //CHECKING EVERY 10sec. if the id1 responds
  delay(200);
  bus.send_repeatedly(2, "*", 1, 10000000); //CHECKING EVERY 10sec. if the id2 responds
  delay(200);
  bus.send_repeatedly(3, "*", 1, 10000000); //CHECKING EVERY 10sec. if the id3 responds

 // network.send(1, "*", 1, 10000000); //CHECKING EVERY 10sec. if the id1 responds
 // delay(200);
 //network.send(2, "*", 1, 10000000); //CHECKING EVERY 10sec. if the id2 responds
 // delay(200);
 // network.send(3, "*", 1, 10000000); //CHECKING EVERY 10sec. if the id3 responds

};

void loop() {

 bus.receive(1000);
 bus.update();
 
  //network.receive(1000);
  //network.update();
} //fine loop

static void receiver_function(uint16_t length, uint8_t *payload) { 
//static void receiver_function(uint8_t length, uint8_t *payload) {
//void receiver_function(uint16_t length, uint8_t *payload) {

  String data;
  Serial.print ("length: ");
  Serial.println (length);
  Serial.flush();
  //Serial.println (data);
  // I create a string from the received data
 // for(uint16_t i = 0; i < length; i++)
 // for (int i = 0; i < 9; i++)
  for (int i = 0; i <length; i++) {
  Serial.print((char)payload[i]);
    char tmp=payload[i];
    data+=tmp;
    
    
    
  }
  if (data.startsWith("T_1")) {
    Serial.print("sensor1");
  }
  else if (data.startsWith("T_2")) {
   Serial.print("sensor2");
  }
  else if (data.startsWith("T_3")) {
   Serial.print("sensor3");
  
  }

 
  //"nome_sensore/sp/dati"
 
  int pos_spazio = data.indexOf(' ');               //trovo la pos. dello spazio
  String tmp = data.substring (pos_spazio);     //Temperatura
  String name_sen = data.substring (pos_spazio, 0); //Nome sensore
  Serial.println ("sensor Name: " + name_sen);
  Serial.println ("Temperature: " + tmp);
  

}

//se perde la connessione con qualche device lo visualizzo
//static void error_handler(uint8_t code, uint8_t data) {
 void error_handler(uint8_t code, uint8_t data, void *custom_pointer) {
  
  //if (code == CONNECTION_LOST) {
   // Serial.print("Connection lost with device id ");
    //Serial.println(data);
    if(code == PJON_CONNECTION_LOST) {
   Serial.print("Connection with device ID ");
    Serial.print(bus.packets[data].content[0], DEC);
    Serial.println(" is lost.");
  }
    
    }
//    }``

Prutser81, 2018-03-11:
transmitter
// Include the libraries we need
#include <OneWire.h>
#include <DallasTemperature.h>
#include <PJON.h>

// network(Arduino pin used, selected device id)
//PJON network(12, 3);

// Bus id definition
//uint8_t bus_id[] = {0, 0, 0, 1};

// PJON object
//PJON<SoftwareBitBang> bus(bus_id, 3);
PJON<SoftwareBitBang> bus(3);

int Invio_Sec = 2000;                   //SET EVERY SECOND SEC. SEND THE DATA
//int PinTrasmissione = 99;               //RECEIVER ID
float floatValTemp=0;                   //INITIALIZE VARIAB. TEMPERATURE           
String NameSensor = "T_3 ";  //NAME SENSORE (attenzione allo spazio finale)
String stringVal = "";                  //VAR. SENSOR READING STRING
unsigned long time;

//SETTING SENSORE
//data pin sensore 
#define ONE_WIRE_BUS 4
OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);
// arrays to hold device address
DeviceAddress insideThermometer;
//FINE SETTING SENSORE

//INITIALIZE TIME
//unsigned long time = millis();

 
void setup(void){

  bus.strategy.set_pin(12);
 
  bus.set_receiver(receiver_function);
  bus.begin();
  
  // start serial port
  Serial.begin(9600);
  pinMode(13, OUTPUT);
  
  //PJON
 bus.set_error(error_handler);
 
 // network.set_error(error_handler);
 // network.set_receiver(receiver_function);
  
  //search for temperature sensor address 
  Serial.print("I look for sensors...");
  sensors.begin();
  Serial.print("Found/i ");
  Serial.print(sensors.getDeviceCount(), DEC);
  Serial.println(" devices.");

  if (!sensors.getAddress(insideThermometer, 0)) Serial.println("No devices found"); 
  Serial.print("Device address: ");
  printAddress(insideThermometer);
  Serial.println();
  // set 12bit sensor resolution
  sensors.setResolution(insideThermometer, 12);
  Serial.print("Device 0 resolution: ");
  Serial.print(sensors.getResolution(insideThermometer), DEC); 
  Serial.println();
  //Serial.flush();
}

void loop(void)
{ 
  
  
  //lettura sensore ogni X sec.
  if(millis() - time > Invio_Sec) {
    time = millis();
    //read the value of the sensor ds18b20
    stringVal=String(NameSensor + double(sensors.getTempCByIndex(0)));
    int lun=stringVal.length();
    char charVal[lun];
    stringVal.toCharArray(charVal,lun);

 unsigned long send_time = micros();
    
   bus.send(99, charVal, lun);
   bus.update();
   
    //network.send(99, charVal, lun);
    sensors.requestTemperatures(); // Send the command to get temperatures
     //debug
    Serial.print ("STRINGVAL:  ");
    Serial.println(stringVal);
    Serial.print("LUN:  ");
    Serial.println(lun);
    //lunhezza = lengte string
  }
  
  bus.update();
  bus.receive(1000);
  //network.update();
  //network.receive(1000);
  
}

// function to print a device address
void printAddress(DeviceAddress deviceAddress)
{
  for (uint8_t i = 0; i < 8; i++)
  {
    if (deviceAddress[i] < 16) Serial.print("0");
    Serial.print(deviceAddress[i], HEX);
  }
}
void error_handler(uint8_t code, uint8_t data) {
  if(code == PJON_CONNECTION_LOST) {
    Serial.print("Connection with device ID ");
    Serial.print(bus.packets[data].content[0], DEC);
    Serial.println(" is lost.");

//static void error_handler(uint8_t code, uint8_t data) {
  //if(code == CONNECTION_LOST) {  //se non c'è connessione lampeggia il led
   // Serial.print("Connection lost with device id ");
    //Serial.print(data);
    digitalWrite (13,HIGH);
    delay (2000);
  }
  digitalWrite (13,LOW);
}
static void receiver_function(uint16_t length, uint8_t *payload) {
    Serial.println ("Received data from Master 99");
    String data;
    
    Serial.print ("length: ");
    Serial.println (length);
    
    for (int i = 0; i < length; i++){
      char tmp=payload[i];
      data+=tmp;
    }
}

gioblu, 2018-03-11:
Ciao @Prutser81 here is the code cleaned up and with the calls updated to work with 10.1:
```cpp
#include <OneWire.h>
#include <DallasTemperature.h>
#include <PJON.h>

PJON bus(3);
int Invio_Sec = 2000; //SET EVERY SECOND SEC. SEND THE DATA
float floatValTemp=0; //INITIALIZE VARIAB. TEMPERATURE
String NameSensor = "T_3 "; //NAME SENSORE (attenzione allo spazio finale)
String stringVal = ""; //VAR. SENSOR READING STRING
unsigned long time;

//SETTING SENSORE
#define ONE_WIRE_BUS 4
OneWire oneWire(ONE_WIRE_BUS);
DallasTemperature sensors(&oneWire);
// arrays to hold device address
DeviceAddress insideThermometer;
//FINE SETTING SENSORE

void setup() {
  bus.strategy.set_pin(12);
  bus.set_receiver(receiver_function);
  bus.set_error(error_handler);
  bus.begin();

  // start serial port
  Serial.begin(9600);
  pinMode(13, OUTPUT);

  //search for temperature sensor address
  Serial.print("I look for sensors...");
  sensors.begin();
  Serial.print("Found/i ");
  Serial.print(sensors.getDeviceCount(), DEC);
  Serial.println(" devices.");

  if(!sensors.getAddress(insideThermometer, 0)) 
    Serial.println("No devices found");

  Serial.print("Device address: ");
  printAddress(insideThermometer);
  Serial.println();

  // set 12bit sensor resolution
  sensors.setResolution(insideThermometer, 12);
  Serial.print("Device 0 resolution: ");
  Serial.print(sensors.getResolution(insideThermometer), DEC);
  Serial.println();
}

void loop(void) {
  if(millis() - time > Invio_Sec) {
    time = millis();
    //read the value of the sensor ds18b20
    stringVal = String(NameSensor + double(sensors.getTempCByIndex(0)));
    uint8_t lun = stringVal.length();
    char charVal[lun];
    stringVal.toCharArray(charVal,lun);

    bus.send(99, charVal, lun);
    bus.update();
    sensors.requestTemperatures(); // Send the command to get temperatures

    // Debug
    Serial.print ("STRINGVAL:  ");
    Serial.println(stringVal);
    Serial.print("LUN:  ");
    Serial.println(lun);
  }

  bus.update();
  bus.receive(1000);

}

// function to print a device address
void printAddress(DeviceAddress deviceAddress) {
  for(uint8_t i = 0; i < 8; i++) {
    if(deviceAddress[i] < 16) Serial.print("0");
    Serial.print(deviceAddress[i], HEX);
  }
}

void error_handler(uint8_t code, uint8_t data, void *custom_pointer) {
  if(code == PJON_CONNECTION_LOST) {
    Serial.print("Connection with device ID ");
    Serial.print(bus.packets[data].content[0], DEC);
    Serial.println(" is lost.");
    digitalWrite (13,HIGH);
    delay (2000);
  }
  digitalWrite (13,LOW);
}

void receiver_function(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info) {
  Serial.println ("Received data from Master 99");
  String data;

  Serial.print ("length: ");
  Serial.println (length);

  for (int i = 0; i < length; i++) {
    char tmp=payload[i];
    data += tmp;
  }
}
```
Can you test it? Thank you again

Prutser81, 2018-03-11:
Get an error message
nvalid use of template-name 'PJON' without an argument list

gioblu, 2018-03-11:
@Prutser81 if you need chat direct support feel free to join the PJON gitter chat:
https://gitter.im/gioblu/PJON

Prutser81, 2018-03-11:
the length has been changed to 1

transmitter:
I look for sensors...Found/i 1 devices.
Device address: 28C5BC2005000041
Device 0 resolution: 12
Received data from Master 99
length: 1
STRINGVAL:  T_3 22.06
LUN:  9
STRINGVAL:  T_3 22.06
LUN:  9
STRINGVAL:  T_3 22.06
LUN:  9

receiver:

sensor Name: ⸮
Temperature: 
Connection with device ID 1 is lost.
Connection with device ID 2 is lost.
length: 587
sensor Name: ⸮⸮
Temperature: 
length: 587
sensor Name: ⸮
Temperature: 
length: 587
sensor Name: ⸮T
Temperature: 

The template parameter are not working. When I press the button insert code
Thank you in advance for the help.

Prutser81, 2018-03-11:
Only have a line change on the receiver side.
`void receiver_function(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info)` 
Thanks very much it works now. You're the best

gioblu, 2018-03-12:
Ciao @Prutser81 it is a pleasure to help.
It seems the compiler does not notice the missing parameter in the receiver handler and the compiled result is garbage. Thank you for pointing out, it will for sure happen again to other users using outdated scripts. Thank you again.