gioblu/PJON #193: PJON_ACK (6) returned with bus disconnected

ibantxo, 2018-03-30:
Hi everybody!

I have just tested this line:
response = pjon_bus.send_packet(10, "All is ok?!", 11);

And the value of response is 6, but the SerialPort is not connected at all.
I am using 
```
  PJON<ThroughSerial> pjon_bus; 
...
  pjon_bus.strategy.set_serial(&Serial1);
  // Avoid default sync ack
  pjon_bus.set_synchronous_acknowledge(false);
  // Set enable pins
  pjon_bus.strategy.set_enable_RS485_pin(SET_ENABLE_RS485_PIN);
  pjon_bus.set_receiver(pjon_receiver_function);  
  pjon_bus.set_id(pjon_bus_device_id);  
  pjon_bus.begin();
```

...where pjon_bus_device_id=16

Is that normal? Is it a bug?

Many Thanks

gioblu, 2018-03-30:
CIao @ibantxo, if setting `pjon_bus.set_synchronous_acknowledge(false);` PJON has no way to know if the packet has been correctly received or not, so if the packet is transmitted `PJON_ACK` is returned.

ibantxo, 2018-03-31:
umm... but code 6 means "correct reception occurred". Better "correct transmission occurred"?

gioblu, 2018-03-31:
Ciao @ibantxo, yes, can be effectively misleading, the missing differentiation is the result of the struggle to fit all features in small micocontrollers and so limit the amount program memory used by PJON. 

On the other side, 2 different values returned depending on the ack configuration could also generate some pain to users needing to check for PJON_ACK or PJON_SENT (or whatever name can be choosen for this potentially new return value) making the usage slightly more complicated at the same time stealing few bytes of precious memory. 

Could you describe me the situation where you need to differentiate the result between "Packet trasmitted, received required ack" and "Packet transmitted, not requiring ack"?

gioblu, 2018-03-31:
@ibantxo thank you for pointing out also the docs inconsistency in the transmission section. Will fix as soon as possible. 👍

ibantxo, 2018-04-03:
Well... I was trying to debug my multi-master project: 2 "speaking slaves":
each one sending a "heartbeat" at a freq=1/s. After a lot of hours the
error proportion was very high: about 5% of messages.
I was thinking about there was a problem in the physical (RS485) network...
Finally I upgraded to v11 and the number of errors went to 0.00% in the
same situation. :-)))) Something is very good inside that version!!!
Congratulations!!! I can go on my project.

Many thanks!!!

gioblu, 2018-04-03:
Ciao @ibantxo, thank you for reporting, fixed as you suggested.
Feel free to ask support if needed :)
Happy tinkering!

closed 2018-04-03 by gioblu