gioblu/PJON #194: SoftwareBitBang Common Ground

dominicklee, 2018-04-05:
Hi @gioblu, first of all, thank you for this wonderful library. It is very impressive and I wanted to use the "SoftwareBitBang" strategy on an Arduino wearable device powered by a LiPo battery. Basically I wanted the wearable device to allow the user to transmit information by touching a receiver device. The microcontroller on the receiver device can be powered by a USB supply, power supply, or LiPo battery. The wearable can only be powered by battery.

However, the wearable device and the receiver device will almost never be on the same ground line in my usage. The user will be wearing shoes and the wearable transmitter. This means there is little to no chance the ground reference would be the same. I have not seen any success in the SoftwareBitBang when both my Arduinos are not on the common ground. Even when both devices are each using their own LiPo battery at 3.7V, I could not get any data received.

Is there some circuit, chip, or module that I can use to make sure the signals are picked up properly? Is there some way to make a "virtual ground reference" for both devices so they have the same reference?

Thanks in advance!

gioblu, 2018-04-06:
Ciao @dominicklee, thank you for your compliments.
For now I do not know any solution for your request although I want to dedicate some time on it to see if something could be worked on, although I suspect it would be a dedicated data-link with probably a slower maximum bandwidth and a longer set of limitations if compared to SWBB.

dominicklee, 2018-07-11:
Ciao @gioblu, is there any updates or solution to this problem? Thanks.

dominicklee, 2018-10-07:
Ciao @gioblu, did you take some time to develop a dedicated data-link for human-body communication without the common ground? For example, from wearable device to wearable device.

dominicklee, 2019-01-18:
Hi @gioblu, I am working on a circuit to make the data-link for human body communication without the common ground. Can you please help me with the following questions so that I can come up with a solution:

1. What is the frequency range of SoftwareBitBang? I am seeing some kHz number in the scope but it varies alot.

2. Is there any way that I can use SoftwareBitBang with a fixed OOK frequency? For example, I want to use SoftwareBitBang on PJON with a 50 kHz carrier. If I put an LC circuit which is optimized for 50 kHz on the transmitter pin, what can I do to the code to make the square pulses close to 50 kHz?

3. If you feel that SoftwareBitBang is not the ideal, please suggest another strategy.

Thanks!

gioblu, 2019-01-19:
Ciao @dominicklee, happy new year and sorry for my late answer.
SoftwareBitBang's transmission method does not make use of any carrier, see here the PJDL specification:
https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang/specification/PJDL-specification-v3.0.md

I would be happy to support you in this, please consider:
- Human body has capacitive, resistive and interference component that variate in each person
- I have detected the skin of a small number of subjects to be extremely resistive, so much not to permit SWBB to go through the body or even few centimeters of skin (with separate common ground)

gioblu, 2019-01-19:
Ciao @dominicklee, here a practical test done on myself :), I generate a 500kHz square wave on an Arduino UNO on pin 13, powered by a laptop NOT connected to AC/DC power supply. I keep with the left hand my finger in contact with the oscilloscope and with the other hand I touch the conductive end of a wire connected to pin 13. As you can see a lot of 50Hz AC induced interference is present, but the signal is present too although its magnitude is extremely limited.

[image]

Here a picture where at the other end the 500KHz is effectively detected

[image]

dominicklee, 2019-01-20:
Hi @gioblu, happy new year. Thanks for your reply!  I understand the limitations of SWBB. We can make a receiver circuit for removing the 50Hz AC noise and amplifying the small voltage of the signal. I will be using an inductor and capacitor (LC) circuit to transmit a software-based carrier frequency. 

Thanks for sending me your oscilloscope tests. I would like to know: How did you generate the 500kHz square wave on the Arduino? What code did you use? Does it work with ESP8266? How did you choose 500kHz frequency? 

Also, which PJON strategy can send the data at 500kHz square waves at the moment? Or maybe even 250kHz? Do you have any example code?

gioblu, 2019-01-25:
Ciao @dominicklee I have just use `digitalWrite` and `delay` to generate the pulse. There is no strategy available for now that uses a On-off-keying or just a carrier in a similar way you describe your goal. I chose the 500Hz (sorry I have made a mistake about the frequency used above and will correct it) just because is 10 times faster than 50Hz and should not be influenced too much by 50Hz interference.

closed 2025-11-08 by gioblu