gioblu/PJON #195: Repeated packets received (v11)

ibantxo, 2018-04-09:
Hi,

My _slaves_ (id 33 and id 44) are sendig HEARTBEATS to a PJON gateway. In each heartbeat each slave sends its "millis()" time and the number of the calls to the "pjon.send" function. Slave 33 sends heartbeats each 500ms.

The gateway detects messages repeated.

Any suggestions to avoid the repeated info?

Many thanks in advance.

Iban

I am using sync ack:

> SLAVE's config:
//#define INCLUDE_ASYNC_ACK true
#define TS_RESPONSE_TIME_OUT 100000
#include <PJON.h>
   Serial.begin(57600); //57600 115200
  //PJON
  pjon_bus.strategy.set_serial(&Serial);
  // Avoid default sync ack
  pjon_bus.set_synchronous_acknowledge(true);
  pjon_bus.set_asynchronous_acknowledge(false);
  // Set enable pins
  pjon_bus.strategy.set_enable_RS485_pin(2);
  pjon_bus.set_receiver(pjon_receiver_function);  
  pjon_bus.set_crc_32(true);
  pjon_bus.set_error(error_handler);
  //pjon_bus.set_id(77);  
  pjon_bus.begin();
    
> GATEWAY's config:
#define PJON_MAX_PACKETS 20 //default 5
#define TS_MAX_ATTEMPTS 20 //default
#define PJON_MAX_RECENT_PACKET_IDS 100000
#define PJON_INCLUDE_TS
#include <PJON.h>

  //PJON
  pjon_bus.strategy.set_serial(&Serial1);
  // Avoid default sync ack
  pjon_bus.set_synchronous_acknowledge(true);
  pjon_bus.set_asynchronous_acknowledge(false);
  // Set enable pins
  pjon_bus.strategy.set_enable_RS485_pin(SET_ENABLE_RS485_PIN);
  pjon_bus.set_receiver(pjon_receiver_function);  
  
  pjon_bus.set_crc_32(true);
  pjon_bus.begin();

FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25509654 50894 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25510155 50895 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25510661 50896 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25511165 50897 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25511666 50898 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25512167 50899 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25512667 50900 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25513167 50901 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25513668 50902 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25514168 50903 | 10 0 0
FROM_GW_BETA FROM_PJON 44 HEARTBEAT 25516217 2551 | 0 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25514670 50904 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25514670 50904 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25514670 50904 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25515174 50905 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25515174 50905 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25515676 50906 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25515676 50906 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25516177 50907 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25516678 50908 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25517178 50909 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25517679 50910 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25518179 50911 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25518179 50911 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25518681 50912 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25519181 50913 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25519681 50914 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25520182 50915 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25520182 50915 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25520182 50915 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25520683 50916 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25520683 50916 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25520683 50916 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25521186 50917 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25520683 50916 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25520182 50915 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25521686 50918 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25521686 50918 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25520182 50915 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25522187 50919 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25522187 50919 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25520182 50915 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25522688 50920 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25522688 50920 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25522688 50920 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25523188 50921 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25523689 50922 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25523689 50922 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25524190 50923 | 10 0 0
FROM_GW_BETA FROM_PJON 44 HEARTBEAT 25526217 2552 | 0 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25524690 50924 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25525191 50925 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25525691 50926 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25526192 50927 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25526692 50928 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25527193 50929 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25527693 50930 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25528194 50931 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25528694 50932 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25529195 50933 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25529195 50933 | 10 0 0
FROM_GW_BETA FROM_PJON 33 HEARTBEAT 25529697 50934 | 10 0 0
`

gioblu, 2018-04-09:
Ciao @ibantxo depending on the data-link you are using that could be interference or a non-optimal timing setup. If in your application you need packet uniqueness and avoid possible repetitions, include the packet id feature that will guarantee uniqueness and avoid duplication. It may be evaluated if uniqueness is really required, if duplication can be handled by the application that would reduce the amount of overhead.

Here an example showing how to use the packet id feature:
https://github.com/gioblu/PJON/blob/master/examples/ARDUINO/Local/SoftwareBitBang/UsePacketId/Transmitter/Transmitter.ino

gioblu, 2018-04-09:
@ibantxo one of the things I would try to reduce the amount of repeats:
- `TS_RESPONSE_TIME_OUT` value could be too short on gateway, device 1 or 2
- in gateway call bus.receive() on both instances without any duration see if better
- in devices call bus.receive() without any duration see if better
- or the contrary in devices try to pass higher duration to bus.receive(), see if better
- Another problem could be that if the gateway is a real time operative system and is also printing the result, some of the repeats could be caused by the time lost by gateway printing the output or executing other programs

gioblu, 2018-05-08:
Ciao @ibantxo how its going? Have you solved this issue?

ibantxo, 2018-05-09:
Umm... I cannot include packet ide feature because memory usage... lol... 
Packet id feature gets the messages reach the destination, but:
1) sometimes too late
2) sometimes not in the correct order.

I have been inproving the network impedance load. It is something difficult to do because my TTL to RS485 transceivers have fixed resistors. They are designed for "peer-to-peer" (2 peers) network configuration and not multipoint.

I have to improve much more the network impedance adjustaments. 

I hope to tell you something soon! lol

Many Thanks!!!!

ibantxo, 2018-08-10:
I am looking forward to testing ThroughAsyncSerial strategy!!   lol
Many thanks to @sticilface, @gioblu and @everybody!!!

ibantxo, 2018-11-19:
After all the night testing, every "secondary" node has sent about 50000 messages to "main" node.
Two of them about 200 messages with a connection_lost error and about 50 messages with a packets_buffer_full error. The third one only 6 times connection_lost error.
I see a lot of repeated packets at receiver.

I am moving to T=1s for testing again.

ibantxo, 2018-11-19:
T=1s. Multi-master: 3 slaves speaking and one master listening (ack-ing). I am using this config at slaves and at master.
```
pjon_bus.set_synchronous_acknowledge(true);
pjon_bus.set_asynchronous_acknowledge(false);

```
About 50000 messages sent by each slave. 113, 92 and 42 connection_lost error. No other error type detected. Repeated packets received on "master" node.

I am moving to:
```
pjon_bus.set_synchronous_acknowledge(false);
pjon_bus.set_asynchronous_acknowledge(true);
```
for testing.

gioblu, 2018-11-19:
Ciao @ibantxo try setting a higher `TSA_RS485_DELAY` https://github.com/gioblu/PJON/blob/master/src/strategies/ThroughSerialAsync/Timing.h#L76

In some cases the packet is not received because the enable pin must be set a little earlier.

gioblu, 2018-11-22:
Ciao @ibantxo how the test went? Were you able to get it working without ack?

ibantxo, 2018-11-24:
Hi! I have changed the code at receiver to count the number of lost packets.
By now... 3 nodes sending 1 heartbeat each 500ms and about 1500 messages sent by each one... 0*3 errors. None error. I will do this test for some hours.

Umm... I thinking about my configuration then... "many speakers":
- some "nodes": they need to start speeking without a pooling wait-time (modbus is not possible)
- one master (it is a gateway), who can ask  to nodes in any time
- the needed bandwidth is small. Almost all time the bus is free.

One suggestion to inprove the "multimaster-ack" communication:
- gateway controlls everything: MASTER, but not doing a pooling.
- gateway sends a "who is next?" message periodically. Very fast and frequently if no response.
- if a slave wants to communicate to master, it does not send the whole message, only a small message to ask for permission to "get the bus" for a time
- master allows the use of the bus to that slave sending a "silence please, except device_id" broadcast message to all the slaves.
Maybe too much time for a realtime _conversation..._ lol...

ibantxo, 2018-11-25:
Witch ACK again (async)...
3 nodes sending 1 heartbeat each 500ms:
about 25000 messages sent by each node...
161, 110, 173 messages with problems at receiver ("master"): connection_lost error

gioblu, 2018-11-29:
Ciao @ibantxo thank you for your precious feedback and excuse e for the late answer.
About the scheme you propose:
>One suggestion to inprove the "multimaster-ack" communication:

> gateway controlls everything: MASTER, but not doing a pooling.
> gateway sends a "who is next?" message periodically. Very fast and frequently if no response.
> if a slave wants to communicate to [...]

Probably is more efficient for the master to roll-call all known nodes asking them if there is something new to transmit. Each slave could be configured to know that can transmit for a given amount of time, and transmit a "NO" if nothing has to be sent to spare bandwidth, all collisions should be practically excluded.

Another way to reduce the chances of a collision is to use a higher baudrate and set a longer `TSA_COLLISION_DELAY`.

About `TSA_TIME_IN` and the test of sending every 500ms, at which rate you receive the packets?

Thank you very much for your support.

closed 2021-03-22 by gioblu