gioblu/PJON #196: Programm a communication with only one links beetween two arduino Uno

Theodmir, 2018-04-11:
Hello,
I made a project where I need to do a communication with two Arduino Uno, this is simple, but now one of the arduino have a switch and i press it , a LED connected to the other Arduino Uno must be power up, if I don't press the switch the  LED don't bright that's all..

gioblu, 2018-04-11:
Ciao @Theodmir look the SoftwareBitBang examples, are really easy to setup and test.
https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Local/SoftwareBitBang

If I understood correct your requirements, you probably need only to copy the Blink example:
https://github.com/gioblu/PJON/blob/master/examples/ARDUINO/Local/SoftwareBitBang/BlinkTest/Receiver/Receiver.ino#L20
(here the receiver function example blinking if a certain character is received (in the example the character is "B")

The only modification you need to do is in the transmitter sketch:
https://github.com/gioblu/PJON/blob/master/examples/ARDUINO/Local/SoftwareBitBang/BlinkTest/Transmitter/Transmitter.ino#L9

Instead of sending a "B"every second, you may call send when the switch has been pressed on the transmiter side. Something like:
```cpp
if(digitalRead(switchPin)) bus.send(44, "B", 1);
// If the switch is pressed send a "B" to receiver
```

Good luck and happy tinkering.
Feel free to ask support here if needed.

Theodmir, 2018-04-12:
Ok thank i did it in another way but it's okay but now I need to have the transmission speed when i push the button and see if there is error in the packet

gioblu, 2018-04-12:
Ciao @Theodmir please paste or link here the code.

Theodmir, 2018-04-12:
First Device:
#include <PJON.h>
#include <PJONMaster.h> 
#include <PJONSlave.h>
    uint8_t bus_id[] = {0, 0, 0, 1}; 
    PJON<SoftwareBitBang> bus(1); 
    int pinLed;
    int pinBouton; 
    boolean etat;

void setup() {
  bus.set_communication_mode(PJON_HALF_DUPLEX);
  bus.strategy.set_pin(12);
  pinMode (pinBouton,INPUT);
  pinBouton=2;
  bus.begin();
  attachInterrupt (0, bouton1, CHANGE);
  bus.set_receiver(receiver_function1);
}

void bouton1()
{
  boolean etat=digitalRead(pinBouton);
      if (etat == HIGH)
  {
  bus.send(2, "A", 1); 
  }
  else
  {
  bus.send(2,"B", 1);
  }
}
void receiver_function1(
  uint8_t *payload, 
  uint16_t length, 
  const PJON_Packet_Info &packet_info
)
{
  if(payload[0] == 'A') 
  {
    digitalWrite(13, HIGH);
  }                        
  else if (payload[0]=='B')
  {
    digitalWrite(13, LOW);
  }                       
  else
  {
    digitalWrite(13, LOW);
  }                       
}
void loop() {
  bus.receive(1000);
  bus.update();
}

For second device:
#include <PJON.h>
#include <PJONMaster.h> 
#include <PJONSlave.h>
  
  uint8_t bus_id[] = {0, 0, 0, 1};
  PJON<SoftwareBitBang> bus(2);
  int pinLed;
  int pinBouton; 
  boolean etat;
    
void setup() {
  bus.set_communication_mode(PJON_HALF_DUPLEX);
  pinLed=13;
  pinBouton=2;
  pinMode(pinLed, OUTPUT);
  bus.strategy.set_pin(12);
  bus.begin();
  bus.set_receiver(receiver_function2); 
  attachInterrupt (0, bouton2, CHANGE);
}
void bouton2() 
{
  boolean etat=digitalRead(pinBouton);
      if (etat == HIGH)
  {
  bus.send(1, "A", 1); 
  }
  else
  {
  bus.send(1,"B", 1);
  }
}

void receiver_function2(
  uint8_t *payload, 
  uint16_t length, 
  const PJON_Packet_Info &packet_info
)
{
  if(payload[0] == 'A') 
  {
    digitalWrite(13, HIGH);
  }                        
  else if (payload[0]=='B')
  {
    digitalWrite(13, LOW);
  }                      
  else
  {
    digitalWrite(13, LOW);
  }                       
}

void loop() {
  boolean etat=digitalRead(pinBouton);
  while (etat==HIGH)
 {
  
 }
  bus.receive(1000); 
   bus.update();
}

gioblu, 2018-04-12:
Ciao @Theodmir I see some issues in the code:
- Do not include PJONMaster or PJONSlave if you are not making direct use of those classes
- Consider handling debounce to avoid sending many state changes
- Consider handling errors to better debug what is happening
- Consider to only set only a bool inside the ISR / the function called by interrupt to reduce its duration (send over PJON afterwards checking the bool state in loop)
- In the first device assign a value to `pinBouton` before calling `pinMode` or the pin mode will not be set for desired pin

closed 2021-04-05 by gioblu