gioblu/PJON #198: New to PJON have to understand 1 master -> many slaves Wiring

ovedmo, 2018-04-24:
Where can i see example of one master to many slaves Wiring diagrams ?
I need it to be as simple as possible, kind of serial one bus .
thank you

ovedmo, 2018-04-25:
Thank you for the replay.
as i see in Compatibility it support MK20DX256 96MHz (Teensy 3.1), what about teensy 3.2, 3.5, 3.6 it will not work?

gioblu, 2018-04-25:
Ciao @ovedmo I am not sure about that, it should be tested, I do not have those models to try here, but I have bought a couple each to make further test in the future, thank you for pointing out.

Here you can find the compatibility table:
https://github.com/gioblu/PJON/wiki/Interfacing

ovedmo, 2018-06-13:
"This example uses the SoftwareBitBang strategy to physically transmit data, it uses a single wire connection shared by all connected devices (like 1-Wire). See the strategy README for the wiring diagram: https://github.com/gioblu/PJON/tree/master/src/strategies/SoftwareBitBang"

**this is for one wire, what is the difference with 2 wires wiring??**

gioblu, 2018-06-14:
Ciao @ovedmo I am sorry for answering you with a big delay.
The SoftwareBitBang implementation permits the use of an input and an output pin because it may be required in some cases, if additional circuitry is used to implement the procedure. For example if using a transistor driven by the output pin, the power output can be amplified obtaining a higher range and minimizing the impact of interference on performance. On the receiver side, a transistor output can be driven by the bus state emitting just logic voltage level from vin or ground.

gioblu, 2018-06-14:
@ovedmo, I have added a remark in the README describing the 2 pins usecase, thank you for pointing out. I am closing here. Please try to keep the questions you wish to ask in separate issues, so other users will more clearly find and understand the content.

Thank you again!

closed 2018-06-14 by gioblu