gioblu/PJON #199: Compile for teensy fail

ovedmo, 2018-04-25:
Hi
**Get this Error :**

In file included from C:\Users\User\Documents\Arduino\libraries\PJON\src/strategies/EthernetTCP/EthernetLink.h:132:0,
                 from C:\Users\User\Documents\Arduino\libraries\PJON\src/strategies/EthernetTCP/EthernetTCP.h:24,
                 from C:\Users\User\Documents\Arduino\libraries\PJON\src/strategies/PJON_Strategies.h:65,
                 from C:\Users\User\Documents\Arduino\libraries\PJON\src/PJON.h:87,
                from C:\Users\User\Documents\Arduino\libraries\PJON\src/PJONMaster.h:50,
                 from C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_255467\Master.ino:1:
C:\Users\User\Documents\Arduino\libraries\PJON\src/interfaces/ARDUINO/TCPHelper_ARDUINO.h:10:30: fatal error: utility/socket.h: No such file or directory
compilation terminated.

**Any Help Please!?**

gioblu, 2018-04-25:
Ciao @ovedmo try defining a constat to exclude other strategies out of the one you are trying to use, see documentation/configuration: https://github.com/gioblu/PJON/blob/master/documentation/configuration.md

gioblu, 2018-05-14:
Ciao @ovedmo will close here for now, feel free to reopen if required.
Thank you for your support.

closed 2018-05-14 by gioblu