gioblu/PJON #201: Try to use ThroughSerial strategies

ovedmo, 2018-04-29:
Hi @gioblu 
with SoftwareBitBang i send all data through one line it work very good, Now what way can i connect many slaves and one master when want to try ThroughSerial strategie... (rs-485 ?)

gioblu, 2018-04-29:
ciao @ovedmo take a look at this example: https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Local/ThroughSerial/RS485-Blink

it simply demonstrates how to use TS along with RS485.

gioblu, 2018-04-30:
@ovedmo thank you for your testing feedback, always appreciated :)
Thank you again for your support!

gioblu, 2018-04-30:
Ciao @ovedmo take a look to this schematic:
[image]

Obviously the USB to RS485 converter could be an arduino. As you can see the enable pins are connected together. Use the setter below to tell to PJON which pin is used for the enable:
``cpp`
bus.strategy.set_enable_RS485_pin(11);
```

If separate enable setters are needed use:
```cpp
// Set RS485 reception enable pin
bus.strategy.set_RS485_rxe_pin(11);
// Set RS485 transmission enable pin
bus.strategy.set_RS485_txe_pin(12);
```

drawkula, 2018-04-30:
Are terminating resistors optional?
Probably that does not really matter on low length busses with only few clients.

gioblu, 2018-04-30:
What I am aware of is that termination are not needed If the latency is less than one bit duration, so required only in really extended setups with fast communication speed.

gioblu, 2018-05-12:
Ciao @ovedmo closing here for now, feel free to reopen if required.

closed 2018-05-12 by gioblu

ovedmo, 2019-07-24:
if i want to use bare max485 for the ThroughSerial how do i connect it...