gioblu/PJON #202: Using WI-FI what strategy should i use

ovedmo, 2018-05-08:
i have 1 master and 3 slaves with rf24 what can i use to make it work (0-: ?

gioblu, 2018-05-08:
CIao @ovedmo rf24 it is not compatible with PJON, or better, noone still have developed a strategy dedicated to this hardware. It is not a complicated task, specially looking at the ThroughLoRa strategy that is quite similar to what should be done for rf24. A discussion is needed to decide how to handle the identification, because the modules have their own identification method that is different from PJON. I will dig futher, thank you for reporting.

dariosalvi78, 2019-11-02:
I am also interested!
nrf24 has some interesting functionalities already baked in.
I wrote [an Arduino library](https://github.com/dariosalvi78/nRF24) some time ago, maybe it could help.