gioblu/PJON #203: Raspberry Pi serial com error

fabpolli, 2018-05-10:
Ciao Giovanni, I've tried PJON over raspberry pi B with example program, unfortunately I've encountered many problems.
If I send B from Arduino raspberry show BLINK in console many times (10/15) and after on Arduino terinal window appear PJON error 101.
If i send two or more chars raspberry doesn't show anytingh and Arduino show error 101 after few seconds.
Raspberry and Arduino are connected with level adapter and with RS485 adapter, I've tested with serial communication without problems, Arduino send and receive and also Raspberry pi. But with PJON seem there are problems with receiver function.
Any workaround for that problems?

gioblu, 2018-05-10:
Ciao @fabpolli, it seems a problem in receiving the acknowledge.
Have you tried receiver or transmitter example?

Try disabling the synchronous acknowledgment on transmitter side calling: 
```
  // Avoid sync ack
  bus.set_synchronous_acknowledge(false);
```
if works fine, it seems I forgot to note in the README that the transmitter device should have configured a longer than default response timeout to be able to receive a sync ack from RPI (who takes around 30/35 millis to answer):  https://github.com/gioblu/PJON/blob/master/examples/RPI/Local/ThroughSerial/BlinkTestTransmitter/Transmitter.cpp#L16

```
#define TS_RESPONSE_TIME_OUT 35000
/* Maximum accepted timeframe between transmission and synchronous
   acknowledgement. This timeframe is affected by latency and CRC computation.
   Could be necessary to higher this value if devices are separated by long
   physical distance and or if transmitting long packets. */
```

This value has not been set as default not to lower performance of setups where the response latency can be lower.

fabpolli, 2018-05-12:
Ciao @gioblu, 
 tests with 
`bus.set_synchronous_acknowledge(false);`
still not working correctly. Now if I send B this transmission will be received correctly but if I send more than one char the transmission is not received at all, no error on transmitter but no echo from receiver.
With this debugging code:
`
printf("payload len: %i\n", length);
printf("payload len: %i %s ->\t", length, myB);
fflush(strdout);
if(payload[0]=='B'){
  printf("BLINK\n");
  fflush(stdout);
}
`
with only B I will see
payload len: 1
payload: 1 B -> BLINK
with BL, no echo at all.
I've tried with synk ack setted to true with same results
Any other idea?

gioblu, 2018-05-12:
Ciao @fabpolli have you removed the serial console?
From the example README:

To correctly receive data on Raspberry Pi it may be necessary to disable the serial console, accessing as root to boot/cmdline.txt and removing console=ttyAMA0, 115200 or console=serial0, 115200 if present.

fabpolli, 2018-05-12:
Ciao @gioblu ,
yes I've followed all the readme, and I've checked it twice when I got trouble.
It'a very strange behaviour, just one char make communication works fine, never met this condition before on Arduino class devices, but this is my first time with PJON on Raspberry

gioblu, 2018-05-12:
Ciao @fabpolli thank you for reporting this, there may be something wrong I agree, I will dig further this weekend.

gioblu, 2018-05-12:
Could you post both codebase to let me recreate the setup?

gioblu, 2018-05-12:
Hei @fabpolli I was reading back the piece of code you posted before:
printf("payload len: %i\n", length); printf("payload len: %i %s ->\t", length, myB); fflush(**strdout**); 
if that is the actual code there is an "r" that should not be there, but I suspect you would have got a compiler error with that.

fabpolli, 2018-05-13:
@gioblu, yes sorry it's only a typo, program compiler without error or warning with -Wall parameter of gcc

gioblu, 2018-05-14:
Ciao @fabpolli I am sorry I may misinterpret your last response.
Should I consider the issue solved? Was the typo the problem?

fabpolli, 2018-05-14:
Ciao @gioblu , no no the typo is only in my comment on github, the problem persist on Raspberry/Arduino communication

gioblu, 2018-05-16:
CIao @fabpolli thank you I am testing, will give you feedback as soon as possible.

gioblu, 2018-05-17:
Ciao @fabpolli sorry for the delay, I have detected the problem you have reported.
If sending long packets using low baudrates, they can be lost using the default configuration because the time required for each byte to be received is higher than the byte timeout defined by default. As you well described this issue does not occur on most supported mcus, but may more probably occur where a real time OS is running along with the application.  
Thank you for reporting this issue. 

The issue is solved changing the default configuration of `TS_BYTE_TIME_OUT` to your needs with a longer duration expressed in microseconds if that is required.

Testing Arduino sending to RPI, setting a higher value (default 5000) on receiver side:
```cpp
#define TS_BYTE_TIME_OUT 15000
```
Also long packets are received as expected:
```cpp
bus.send(44, "BETTER FOR YOU TO WORK!", 23);
```

Next steps:
- Relate a commit to this issue changing where the default `TS_BYTE_TIME_OUT` value is set higher to support longer packets by default on RPI 
- Fix the examples with the use of makefiles.

Thank you again for your support.

fabpolli, 2018-05-17:
ok, thank You. I will try with more longer timing as You described.
I will try with
bus.set_synchronous_acknowledge(false);
and also with
bus.set_synchronous_acknowledge(true);
just to know if it work with all kind of configurations, after some test I'll report here the results

gioblu, 2018-05-17:
Ciao @fabpolli I will push soon the updated examples based on makefiles and higher timeout for you to test more easily. Thank you very much again.

fabpolli, 2018-05-21:
Ciao @gioblu , I've tested new version of PJON with this results:
Syncronous ack works fine if the trasmitter set the property to false, if is setted to true on trasmitter the message is received many times and after some seconds transmitter notify PJON error 101.
Async ack doesn't work at all. If I set the property to true the receiver doesn't show any message and transmitter notify PJON error 101, if setted to false the message will be received many times and after some seconds I will see PJON error 101 on transmitter, but I think that it caused by the properties's defaults value.
But now if I try to send a message from RPI to Arduino, message never reach the receiver, tested with specific ID and with PJON_BROADCAST with my test program and also with standard example, with or without sudo

gioblu, 2018-05-21:
Ciao @fabpolli I see 2 issues:
- For sync ack missing:
```
#define TS_RESPONSE_TIME_OUT 35000
/* Maximum accepted timeframe between transmission and synchronous
   acknowledgement. This timeframe is affected by latency and CRC computation.
   Could be necessary to higher this value if devices are separated by long
   physical distance and or if transmitting long packets. */
```
In the Arduino receiver and sender examples (to be compatible to RPI timing, and permit users to use default example with no modifications)
- For missing reception RPI->Arduino if reprogramming the Arduino compatible device, while RPI is transmitting you may end up freezing the Arduino compatible device because of this bug I have reported to the Arduino toolchain repository: https://github.com/arduino/Arduino/issues/6050
If this condition happens reset the Arduino device.

I will do further testing today, including async ack, and add the response timeout to Arduino examples.

gioblu, 2018-06-15:
Ciao @fabpolli take a look to this issue: https://github.com/gioblu/PJON/issues/210
Seems @hyndruide spotted the issue. It is a issue related to the timing of enable pins.
I have added a setter: `set_RS485_delay` to let users add a trailing delay before setting the enable pins low. This effectively fixes the issue you have rised, at least with my setup. Looking around in other libraries it seems a widely applied fix to this problem.

Could you give it a try?
I have set a default delay of 1 millisecond that fixes the issue, but that value can be set as required:
```cpp
// Set RS485 txe/rxe pins disable delay in milliseconds
bus.strategy.set_RS485_delay(1);
```

gioblu, 2018-06-22:
Ciao @fabpolli thank you for the fix. 
After extended testing it seems for now stable. Do you get the same results?