gioblu/PJON #204: Timing overflow

poursal, 2018-05-10:
First of all thank you for your great work.

I am experimenting with a SWBB bus using 5 transmitters and one receiver and after a while I was losing one of the transmitters (they send a heartbeat message). While reviewing the code I noticed that in timeouts i.e.:

      uint32_t time = PJON_MICROS();
      /* Transmitter emits a bit SWBB_BIT_WIDTH / 4 long and tries
         to get a response cyclically for SWBB_RESPONSE_TIMEOUT microseconds.
         Here Receiver synchronizes to the falling edge of the last incoming
         bit and transmits response variable */

      while( // If initially low Wait for the next high
        ((uint32_t)(PJON_MICROS() - time) < SWBB_BIT_WIDTH) &&
        !PJON_IO_READ(_input_pin)
      );

You do not take into account overflows (https://www.arduino.cc/reference/en/language/functions/time/micros/). I was wondering if this can cause a device to remain in a busy loop while waiting a timeout for 70 minutes (or 50 days if using millis)?

gioblu, 2018-05-10:
Ciao @poursal I am sorry I do not see how the overlow is not handed.
being typecasted as uint32_t the result is positive ie:
```
time = 4294967295  (max value uint32_t)
while 3 (overflowed new value) - 4294967295 or 4 < 40 (bit width) -> true
while 39 (overflowed last acceptable value) - 4294967295 or 40 < 40 (bit width) -> false 
```
so if overflows occurs the condition is met as it should for this reason I would say that overflow is handled.

Also as soon as the bus goes high, for example because of other devices transmission or interference the condition would not be met. So for me it seems strange to hear that a device could be stuck in that while for so long, also a wire connected to nothing would really soon pick up some interference and break the while not meeting the condition.

Which devices are you testing with?

gioblu, 2018-05-11:
Ciao @poursal thank you for your answer. If could be helpful, I can help you debugging the codebase if you can share it. Happy tinkering.

gioblu, 2018-05-12:
Ciao @poursal closing here for now, feel free to re-open if needed.

closed 2018-05-12 by gioblu