gioblu/PJON #205: PJON_Packet_Info does not name a type

fcastrovilli, 2018-05-15:
Hello there,
I'm trying to hook up two Arduino Micros through SoftwareBitBang.
Tried with the example sketch for Blink With Response everything does work correctly.

Implementing code lines from the example sketches into my main compiles for the transmitting board but doesn't compile for the receiver board.

The problem is in this line:
void receiver_function(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info) {

exit status 1
'PJON_Packet_Info' does not name a type

On the other board everything is fine, same function, both pasted from the example.
Don't really know whats the problem here

gioblu, 2018-05-15:
Ciao @fcastrovilli thank you for reporting please paste the code here.

fcastrovilli, 2018-05-15:
It's quite large thing I think and compiles by itself.
only thing I added are:

Global:
//PJON
#include <PJON.h>
PJON<SoftwareBitBang> bus(44);

in Setup():
  //PJON
  bus.strategy.set_pin(12);
  bus.begin();
  bus.set_receiver(receiver_function);

in Loop():
  bus.update();
  bus.receive(1000);

and the function:
void receiver_function(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info) {
  /* Make use of the payload before sending something, the buffer where payload points to is
     overwritten when a new message is dispatched */
  if (payload[0] == 'B') {
    Serial.println("BLINK");
    digitalWrite(13, HIGH);
    delay(30);
    digitalWrite(13, LOW);
    bus.reply("B", 1);
  }
}

The issue is in the function declaration, on the other board (different sketch oc) worked fine.
Boards are Pro Micro, not Arduino Micro.
Don't really know what to do about that

[PJON_not_working.zip](https://github.com/gioblu/PJON/files/2006211/PJON_not_working.zip)

fcastrovilli, 2018-05-15:
Tried, same results

gioblu, 2018-09-03:
Ciao @charliewilliams thank you for your support, what you mean pre-declaring the function? You mean to have the receiver function declared before `set_receiver` call?

If so, I obtain the same result you obtain, also to me is not clear why that is required in some cases, I can move in the examples all function definitions before the call to avoid the error.

charliewilliams, 2018-09-07:
I mean at the top of my file I need to do

`void receiveIR(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info);`

in order for the actual body of the function `void receiveIR(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info) { }` to compile later.

closed 2019-11-04 by gioblu