gioblu/PJON #207: Porting PJON to ESP32

mxbranson, 2018-05-23:
Really need this. Any pointers on where I should look to have a go myself? Cheers

gioblu, 2018-05-23:
Ciao @mxbranson, ESP32 would be for sure interesting to be supported.
Using an Arduino toolchain core would spare a lot of time defining al the standard system calls like micros, digitalWrite, ecc. Doing so the work would be minimal, it may work out of the box. I will buy a couple to make some tests and support you with the porting.

Jorgen-VikingGod, 2018-05-23:
arduino core functionality of ESP32 is really good supported: https://github.com/espressif/arduino-esp32 

Should work out of the box, without much tweaks...

I'm using some of [these](https://www.ebay.de/itm/ESP32-Development-Board-2-4GHz-Dual-WiFi-Bluetooth-Chipsntenna-Deko/263693864717)

gioblu, 2018-05-23:
Ciao @mxbranson thank you, for sure errors will help, will wait for your feedback.
Ciao @Jorgen-VikingGod thank you for your support. You mean you are using one of those along with PJON already? If so with which strategy?

Jorgen-VikingGod, 2018-05-23:
Ciao @gioblu I'm using these boards for many projects, but still had no time to try it with PJON.
Is still on my ToDo list. But I think it will work with less efforts.

gioblu, 2018-06-28:
Ciao @mxbranson thanks to the support of @Jorgen-VikingGod and some work tweaking timing now ESP32 is working fine with the SoftwareBitBang strategy. :)
Could you give it a try?

gioblu, 2018-10-25:
Closing here after SoftwareBitBang and ESPNOW strategies support for ESP32 has been added.
Feel free to open a dedicated issue for support to other unsupported strategies available.

closed 2018-10-25 by gioblu