gioblu/PJON #209: Random error in example slave of Dynamic Addressing

sabas1080, 2018-05-26:
Hi @gioblu 

I am testing in MKR1000 and SAMD21 Dev sparkfun and I get the following error
```
In file included from /var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_modified_sketch_977787/Slave.ino:1:0:
/Users/sabas/Documents/Arduino/libraries/PJON/src/PJONSlave.h: In member function 'void PJONSlave<Strategy>::generate_rid()':
/Users/sabas/Documents/Arduino/libraries/PJON/src/PJONSlave.h:220:32: error: no matching function for call to 'random()'
         (uint32_t)(PJON_RANDOM()) ^
                                ^
/Users/sabas/Documents/Arduino/libraries/PJON/src/PJONSlave.h:220:32: note: candidates are:
In file included from /Users/sabas/Library/Arduino15/packages/SparkFun/hardware/samd/1.4.0/cores/arduino/Arduino.h:77:0,
                 from /var/folders/ng/kwv364sj5jlg24q8f_vqh5y40000gn/T/arduino_build_690411/sketch/Slave.ino.cpp:1:
/Users/sabas/Library/Arduino15/packages/SparkFun/hardware/samd/1.4.0/cores/arduino/WMath.h:22:13: note: long int random(long int)
 extern long random( long ) ;
             ^
/Users/sabas/Library/Arduino15/packages/SparkFun/hardware/samd/1.4.0/cores/arduino/WMath.h:22:13: note:   candidate expects 1 argument, 0 provided
/Users/sabas/Library/Arduino15/packages/SparkFun/hardware/samd/1.4.0/cores/arduino/WMath.h:23:13: note: long int random(long int, long int)
 extern long random( long, long ) ;
             ^
/Users/sabas/Library/Arduino15/packages/SparkFun/hardware/samd/1.4.0/cores/arduino/WMath.h:23:13: note:   candidate expects 2 arguments, 0 provided
```
seems to be the random function requires a parameter necessarily, I have made the following an change

`(uint32_t)(PJON_RANDOM(254)) ^`

compile ok, but in master

`List of slaves known by master: ` is blank

in slave
`Received: M a s t e r   s a y s   h i !`

Thanks

sabas1080, 2018-05-26:
networkAnalysis Two SAMD21 DEv SparkFun, pin12, SoftwareBitBang
```
Packet Overhead: 18B - Total: 0B
Bandwidth: 0B/s
Data throughput: 0B/s
Packets sent: 0
Mistakes (error found with CRC): 0
Fail (no acknowledge from receiver): 48
Busy (Channel is busy or affected by interference): 0
Accuracy: nan %

```

But I have communication with example
https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Local/SoftwareBitBang/BlinkTest

gioblu, 2018-05-26:
Ciao @sabas1080, for some reason seems acknowledge procedure might have some issue, probably 
a timing issue that can be fixed with a minor tweak. If you can, give a try SpeedTest example (where who prints result is on receiver side), just adding in the transmitter setup `bus.set_synchronous_acknowledge(false);`

You should see positive results logged by the example. 
Please if you can share it here, it might help to understand why ack is not working.

If so, I would suggest you to try setting a longer than default `SWBB_RESPONSE_TIMEOUT` predefining the constant before the PJON.h inclusion (default 1500 micros) because for some reason one or both devices may need more time than other supported devices to answer back ACK, this may fix the acknowledge issue.

gioblu, 2018-10-28:
Ciao @sabas1080 thank you for reporting the compilation error, sorry for the late update, the suggested change has been added https://github.com/gioblu/PJON/commit/a2034a4f1c215d35672a884b6b0091bea0d31aa4

gioblu, 2019-04-16:
Ciao @sabas1080 I suspect this issue is related to https://github.com/gioblu/PJON/issues/261

The dynamic addressing v2.0 (which this issue refers to) has been discontinued and will not be developed further, see the new version, [dynamic addressing v3.0](https://github.com/gioblu/PJON/blob/master/specification/PJON-dynamic-addressing-specification-v3.0.md) and its related implementation.

I will close here, feel free to continue discussing about the SAMD21 in the issue linked above.

closed 2019-04-16 by gioblu