gioblu/PJON #214: bus.reply() does not work when using ThroughSerial

codebeat-nl, 2018-06-16:
Since yesterday I started to develop a PC Keyboard to Arduino project via ThroughSerial. What I want to achieve is an easy stand-alone prototype tool that can emulate a full-featured keyboard (as buttons) by sending virtual keycodes (byte) and keystate information such as "down", "up", "press" and "released" states (byte) to the Arduino. This can be handy to prototype, do a quick conditional test, navigate thru LCD menus etc, emulate hardware, any button(-state) or to provide a full-featured real-time virtual keyboard to the project in a breeze.

I also want to keep the default debug/monitor facility, or to exchange config information so it must be bi-directional. However there is no bi-directional ThroughSerial example like SoftwareBitBang SendAndReceive, only Transmitter or Receiver examples. 

I set up a test project and used the transmiter example in the WINX86 directory and modify it into functions that will be exposed via an Win32 DLL. The reason I do it this way is because want to use it in a Delphi project (and there is no pascal version) to create the actual program, visuals and controls.  The test setup is working perfectly fine however in one direction. I read some PJON docs and other bi-directional examples and modified it to a PJON_HALF_DUPLEX communication.  

However, when I do a bus reply() after a receive at the Arduino side, it will never trigger the receive handler at the Windows C++ side. After this the bus is locked and will never receive anything.  I tried everything, in different order, different options, etc, it will never be triggered.

Here is my arduino sketch:

```
#define TS_BYTE_TIME_OUT 15000
#define TS_RESPONSE_TIME_OUT 60000
/* Maximum accepted timeframe between transmission and synchronous
acknowledgement. This timeframe is affected by latency and CRC computation.
Could be necessary to higher this value if devices are separated by long
physical distance and or if transmitting long packets. */
#define TS_COLLISION_DELAY 3000
#define PJON_INCLUDE_TS true

#include <PJON.h>

#include <SPI.h>
#include <font.h>
#include <LM15SGFNZ07.h>

#define PIN_LCD_LED1    3
#define PIN_LCD_LED2    6
#define PIN_LCD_CS      9
#define PIN_LCD_RST     5
#define PIN_LCD_RS      4

#define COLOR_FG 0xFFF
#define COLOR_BG 0x22F

LM15SGFNZ07 lcd( PIN_LCD_CS , PIN_LCD_RST , PIN_LCD_RS );

// <Strategy name> bus(selected device id)
PJON<ThroughSerial> bus(44);

void receiveHandler(uint8_t *payload, uint16_t length, const PJON_Packet_Info &packet_info) {
  /* Make use of the payload before sending something, the buffer where payload points to is
     overwritten when a new message is dispatched */

  char sLength[3];
  char sData[PJON_PACKET_MAX_LENGTH]; // Needed to 'filter' garbage leftovers after payload's length, 
                                      // it is not null terminated or cleared.
                                 
   // Clear data
  memset( &sData[0], 0, sizeof( sData )); 
   // Copy data
  strncpy( &sData[0], &payload[0], length );
  
  String(length).toCharArray(sLength, 3);
  
  lcd.clear( COLOR_BG );
  lcd.drawString(sData, 1, 1, COLOR_FG, COLOR_BG);
  lcd.drawString(sLength, 1, 20, COLOR_FG, COLOR_BG);
  
  if( payload[0] == 'B' || strcmp((char*)&sData[0], (char*)"Hello World" ) == 0 ) 
  {
    // Can't do this:
    //if(!bus.update()) // If all packets are delivered, send another
    // bus.reply(&sData[0], length);
    
    digitalWrite(LED_BUILTIN, HIGH);
    delay(5);
    digitalWrite(LED_BUILTIN, LOW);
    delay(5);

[... 434 lines cut ...]
   cbSerialPower.Checked:= pjonSerialStart( 1, 115200, 45, 44 );
   if( cbSerialPower.Checked ) then
    begin
      pjonSerialSend( PChar( Edit1.Text+#0 ), length( Edit1.Text ) );
    end;

    Label2.Caption:=IntToStr( pjonSerialLastErr() );
  end;

 if( NOT cbSerialPower.Checked ) then
  pjonSerialStop();
end;

procedure TForm1.Timer1Timer(Sender: TObject);
begin
 pjonSerialReceiveUpdate();
end;

end.

```

[image]

Can it be solved? Many thanks for any reply.

gioblu, 2018-06-17:
Ciao @codebeat-nl your project seems really interesting! 
You are experiencing a bug, if using an Arduino compatible device with a hardware serial port, when a collision occurs the Arduino compatible device ends up frozen. I am precisely now working to fix this issue. 

If you want to go further without waiting for the fix:
1. Avoid the use of synchronous acknowledgment calling `bus.set_synchronous_acknowledge(false)`
2. Avoid structuring the communication order any possible collision or the chance that 2 packets are sent at the same time, that means to use a master slave approach where the master provokes a slave response: master request -> (small delay) -> slave response

Following point 1 and 2 should be a safe workaround to avoid this bug and also a stable procedure.
I am sorry, I hope to get a fix for this as soon as possible.

Feel free to request for support if required :)

gioblu, 2018-06-17:
@codebeat-nl it would really be nice to add an example for users to get the result you showed. It may enable users to build a clean and professional custom solution in minutes. I am sincerely not that much into Delphi :).

gioblu, 2018-06-18:
Ciao @codebeat-nl did the solutions 1 or 2 temporarily solved your issue?
Thank you again :)

codebeat-nl, 2018-06-22:
Ciao @gioblu , thanks for the replies. Nearly close to the weekend I just want to let you know I am working on it, the solutions you provide doesn't help because I tried that before. With `bus.set_synchronous_acknowledge(false) `it lost it's security to be reliable (unreliable like a normal serial connection).

However, make some progress at the keyboard 'interface' design, it is working great and missing bidirectional doesn't have to be a problem (when doing it the 'smart' way) to be a keyboard or mouse emulator. 

To be a simple solution to everybody, out-of-the-box solution, bidirectional must be possible because it also provides debug possibilities. So this is still a wanted/required feature.

Just want to let you know, please don't close the question/bug. This weekend I will try to show my findings.

YSmetana, 2018-08-22:
Hello! :) Gioblu, thanks for your work!

I have similar problem with HC-12 (433 MHz UART) and AltSoftSerial.

    #define BYTES 9
    #define TS_BYTE_TIME_OUT      80000
    #define TS_RESPONSE_TIME_OUT TS_BYTE_TIME_OUT * BYTES * 2.5 // PJON timeout, if operating at less than 9600Bd should be longer
    #define PJON_PACKET_MAX_LENGTH 32
    #define PJON_INCLUDE_TS
    #include <PJON.h>
    ...

    // Remote node. Receives and replies.
    void receiver_function(...) {
          if((char)payload[0] == 'B') {
       //
      }
        // If all packets are delivered, send another
        if(!bus.update())  {
          delay(500);
          bus.reply("B", 1);
        }
    };

    ...
    bus.receive(50000);

So, I checked PJON send and "raw" receive. And I can see PJON packed on 200+ meters (was not able to move further away). 

For some reason, after reply, my remote side says "Connection with [main] node lost". So it seems it can't receive ACK for its reply that had been sent.

Then, I moved back to PJON-PJON pinger (receive - send back). But with serial console this time. And it turned out that remote side receive packet, but can not send it back (on a long distances).

I am sending packet (every 4 s) from node 1 to node 2, waiting a bit (0.5 s) and sending a packet back from node 2 to node 1. And it looks like node 2 can't get acknowledgement of sent packet.

On short distances (up to 150 m) it works fine.

"Raw" transmission pinger (1 byte, no PJON) works for 750+ meters on HC-12 in 2400 bps.

gioblu, 2018-08-30:
Ciao @codebeat-nl I think I may have found the problem thanks to the report of @Girgitt, 
Could you give it a try and see if the latest referenced commit solves your issue on Windows?
Thank you again :)