gioblu/PJON #216: SoftwareBitBang strategy problem

ovedmo, 2018-06-18:
Hi @gioblu 
i have SoftwareBitBang strategy one wire and a problem (-:

when slave send_packet he get PJON_ACK at most but when master send he get PJON_FAIL 95% of the time, any suggestion how to debug and find why?

gioblu, 2018-06-18:
CIao @ovedmo please attach here both codebases I will support you to solve the issue.
If you need direct support you can also find me on gitter in the PJON room.

ovedmo, 2018-06-19:
Hi @gioblu 
I will attach 2 files one for master and one for slave....

[slave.txt](https://github.com/gioblu/PJON/files/2113928/slave.txt)
[master.txt](https://github.com/gioblu/PJON/files/2113929/master.txt)

**thank you for your help**

gioblu, 2018-06-22:
As described in the dupe I was able to replicate the issue. Working to fix the timing issue.
Thank you for your report @ovedmo

ovedmo, 2018-07-01:
hI @gioblu 
Any update?
thank you

gioblu, 2018-07-02:
Ciao @ovedmo I was able to reproduce the problem you have reported. Thank you very much again for your support. Polling duration calling just `receive()` (so one reception attempt per loop) is not enough to have optimal performance. I have now corrected the example calling `rececive(1000);` (so attempt to receive data for 1 millisecond) and that fixes the issue, I can experiment here 100% reliability after the tweak. Probably the recent additions have added some overhead to the loop and have degraded the example's performance.

gioblu, 2018-07-02:
@ovedmo consider that while developing your application, and so adding its implementation, you will add also a certain overhead to the loop time (other tasks duration). This problem may occur again if added tasks are extremely long and blocking. If so try to minimize the duration of other tasks and or dedicate more time to reception.

closed 2018-10-28 by gioblu