gioblu/PJON #218: PJON unreliable between ESP and Arduino Nano

skorokithakis, 2018-06-23:
Hello,
I just found PJON and it looks perfect for what I want to do, communication between an ESP8266 and an Arduino Nano. However, trying the "LED blink" example, I get a "blink" on the Arduino around once every second, and around ten seconds later the devices disconnect with "Connection lost".

This is a very simple circuit of one wire, so I don't know if PJON is this unreliable or if I'm doing something wrong. Is there something I can fix?

gioblu, 2018-06-23:
Ciao @skorokithakis it should not happen. Are you testing using master?

gioblu, 2018-06-23:
Be sure you have pulled from master https://github.com/gioblu/PJON/commit/a5563b58748885d638058f15b9640044d6c6d894

Feel free to contact me on gitter if some support is required.
Thank you again.

skorokithakis, 2018-06-23:
I'm not sure, I'm using PlatformIO, so I think yes. I will try again and tell you more on Gitter, thanks!

gioblu, 2018-06-23:
Ciao @skorokithakis will be a pleasure to help, sorry for the late answer.

gioblu, 2018-06-23:
Ciao @skorokithakis as we discussed on gitter, I close here being this issue produced by using v11.0.
The issue has already been fixed with this commit: https://github.com/gioblu/PJON/commit/a5563b58748885d638058f15b9640044d6c6d894

This will be released soon in v11.1
Thank you very much for your support.

closed 2018-06-23 by gioblu