gioblu/PJON #220: Spreading Factor settings with PJON

jwktje, 2018-07-03:
I'm using arduino-LoRa with PJON. It works great when using Spreading Factor 7, but when I change it to anything else it stops working. I'm communicating between two arduino's using the RFM95W.

Is there anything else I need to configure for other spreading factors to work? I want to increase the range I get and I wanted to experiment with a higher SF but can't get it to work.

Any help at all is really appreciated!

Matheus-Garbelini, 2018-07-04:
@jwktje By increasing Spreading factor, each symbol time increases by a little when transmiting. 
Try increasing  TL_RESPONSE_TIME_OUT in Timing.h
Your message may be exceeding the 100ms default timing.

jwktje, 2018-07-04:
Hey Matheus,

Thanks for your input!
I tried increasing the timeout by 10 times by defining it before including the library like so
`#define TL_RESPONSE_TIME_OUT 1000000UL`

I printed the value later to check it was defined as such and that worked. But even going from Spreading factor 7 to 8, the sketch doesn't work. 

On spreading factor 7 the two nodes communicate fine, but on 8 or higher, it doesn't.

I'm using the PJON dynamic addressing which works great on SF 7. I feel like there must be more that I need to config for other Spreading factors to work. Any idea?

gioblu, 2018-07-05:
@jwktje have you tried tweaking the amount of time dedicated to reception or `receive(reception_time)`?

jwktje, 2018-07-05:
Thanks for responding! No I haven’t yet. How can I tweak that? I’ve only
tweaked the mentioned parameters in the code in my previous code.

Matheus-Garbelini, 2018-07-05:
@jwktje ah so you're using auto addressing. I remember tweaking 
PJON_ID_REQUEST_INTERVAL in PJONDefines.h

It's defaulted to 100ms which is too low when using other sf. Try increasing it by 1 sec.

Also, did you test pjon without auto addressing with a higher sf?

jwktje, 2018-07-06:
Increasing that interval I got SF8 to work in a stable way too. But anything higher is still having the same issue.

The TX/RX windows are important with LoRa I think. I did a little testing with TTN and this was the case there. So I wouldn't be surprised if there are more parameters to tweak to get SF10+ to work well..
If you have any pointers on what I could try, or the time to test, that would be awesome. 

Your point about testing higher SF's without the auto addressing sounds like good advice to debug this issue. I will try that and report back once I have the time to get some testing in.

jwktje, 2018-07-06:
Did some testing without dynamic addressing based on the example sketches. A weird problem appeared. When using setSignalBandwidth and setSpreadingFactor, the Transmitter example never seems to start sending. When disabling setFrequency functionality returns.

I don't know if this gives you any insight to a possible cause? I will return to my original test-case and I will stop using setSignalBandwidth for now and see if that changes anything.

gioblu, 2018-07-10:
Ciao @jwktje I am sorry for answering you back so late. A lot of real life issues piled up.
Thank you for your support to the project. I also would go to the bottom of this because I agree would be nice to higher up the SF. I am now not able to test practically but I see some timing issues could affect the functionality.

https://github.com/gioblu/PJON/blob/master/src/PJONDefines.h#L169

Not only `PJON_ID_REQUEST_INTERVAL` should be higher, also the `PJON_ADDRESSING_TIMEOUT` may be required to be higher and also `PJON_LIST_IDS_TIME` may be required to be higher.
 
I am sorry if you have experienced this issues, although I am happy to hear that at least in some configuration those classes work out of the box over the air, when initially was all designed for wired networks.

I am considering to iterate the dynamic addressing specification to make it effectively useful for setups like the one I think you would achieve. For now the dynamic addressing procedure let devices self assign a random slave id, although a user inserted, known id could be useful to reach a certain device that moves in a network of PJON-LoRa routers.

Thank you again.

jwktje, 2018-07-10:
Yes that would sound great. To still dynamically have nodes join but have a user defined node-id. Maybe from a custom board that has a DIP to address or something. This would be perfect for my use-case.

I will test with the mentioned intervals and see if increasing these makes SF12 work. If I succeed, then maybe these could all be changed in the library to be overridden before including the library with an ifndef. 

Again, compliments on PJON. Loving it. We can get this to work easily. Just some tweaking to do! 👍

gioblu, 2018-07-10:
This is what was rolling in my mind about this:
- We have a network layer that handles 1 trillion nodes
- We have a sort of dynamic addressing with a sort of narrow applicability envelope
- We have a network layer able to go over internet, LoRa, wires, other radio modules, RS485 and so on

Why not to make the actual RID or randomly generated 4 byte slave identifier, a 5 bytes "device name" that can be obtained by the PJON website as it is done for the bus id? 

That with some coverege wouldnt that be cool?
Couldnt internet initially be able to close the gaps?
A sort of new free internet made and maintained by makers?

It could even give some sense to exist to the LoRa pager project :)
If I have to be sincere, this always was the goal of the project, although it was never too much advertising it because i did not want to be ostracized or considered crazy.

jwktje, 2018-07-10:
Hey @gioblu 

Just tested by upping all mentioned intervals times 8. This made it work on SF10, which is an improvement. But the joining process was a lot more flaky than on SF7. It continuously took about 4+ tries to get an ID and sometimes it would just hang and do nothing.

So I feel like someone with more Lora expertise than me would be able to apply some logic to this problem, to get specific correct timing/intervals for the different SF's. Do you have experience in this regard?

But the good news is, it did work to a certain extent. Not stable yet but for a quick test this is good news!

About your other point; I think I kinda understand what you are getting at. So a public universal network that can traverse all sorts of infrastructure? That would be awesome for a lot of use-cases I imagine, especially when using a universal protocol like this one. 

For my specific use case, I have an idea for an application that would use Lora to get one central node (that would function as a bridge) to send downlinks to a bunch of nodes in a ~5km radius. I want to control hardware through this from a central point without needing GPRS or WiFi. For this use-case it would probably make sense to have one universal Arduino Sketch for all nodes (excluding the bridge) without a hardcoded ID. But be able to set the address on the hardware by solderpads or a DIP. So that I know that a certain node will always be on a certain ID without having differences in the code.

That's at least what I'm thinking at the moment. So I wouldn't probably need dynamic addressing for this? I'm using it in my prototype as I don't have any addressing hardware solution in my prototypes so dynamic ID's made sense.

Matheus-Garbelini, 2018-07-11:
@jwktje Actually there's a mode called Channel Activity Detection (CAD) that these Lora transceivers can operate and allows to detect the preamble of an incoming message before sending anything.
There's planning to include this in the strategy, but it still needs to be tested, this should not only increase reception efficiency but avoid a lot more broadcast collisions.