gioblu/PJON #221: Nodemcu GPIO Pins SWBB

isolution-de, 2018-07-06:
I'm having issues using other GPIO pins than the default D1...D4. Is it possible to use pins D8, RX, TX, A0, S1, S2 or S3 for SWBB on Nodemcu? Has anyone experience with that?

gioblu, 2018-07-06:
Ciao @isolution-de sorry for the late answer.
I have worked a timing setup that seems working nominally only on pins D1 to D4. If you need to use another pin it may be possible tweaking the timing setup here:  https://github.com/gioblu/PJON/blob/master/src/strategies/SoftwareBitBang/Timing.h#L143

I will try in the next iteration to work out a timing setup that could operate nominally also on other pins, although that it is not necessarily possible.

isolution-de, 2018-07-08:
Ciao Giovanni,
thank you very much, you're help is appreciated very much. I was able to change my sketch and free up port D3 to use PJON, but I think for future users it would be nice to have more options regarding the IO port.

closed 2018-07-08 by isolution-de

gioblu, 2018-07-09:
Ciao @isolution-de thank you for your feedback! 
I agree with you the pin support for NodeMCU must be wider. 
I will work on that as soon as I will be able to close the other high priority issues that has been rised.

Thank you again  :)