gioblu/PJON #222: Local bus going through walls / outdoors

drtrigon, 2018-07-06:
Now as I have my minimal bus up and running (linux/raspi - LUDP - Uno+ETHshield - SWBB - devices) I would like to extend it to outdoor sensors. What setup would you suggest to go through walls (without drilling holes) or windows? I can imagine these setups:
A) SWBB - AS+LED - window - AS+LED - SWBB
B) SWBB - LoRa/radio - window/wall - LoRa/radio - SWBB
C) ?
for A+B I would like to use 2 tunnelers. The LoRa radio setup would be preferred as that would allow also to add devices to the LoRa bus (between the 2 tunnelers) and I do not need to setup LED line-of-sight communication. If LoRa is the best choice, I would like to use a Dragino LG01-S (Uno/Yun combo) for the SWBB - LoRa part, so also direct LUDP - LoRa tunneler is a possibility...

What would be your advice?

gioblu, 2018-07-06:
ciao @drtrigon for sure LoRa is a choice, but if the requirement is a point to point connection, maybe HC-12 it can be cheaper and simpler. It can be used with SoftwareSerial and it has a range of around 150 meters without line of sight. If "paranoid" and you dont want to broadcast your data over a relatively wide range, you may also use AS along with a couple of laser diodes, although it is for sure a solution that would require some more tinkering and collimation, with line of sight it could work for hundreds of meters.

gioblu, 2018-07-06:
@drtrigon I do not have personal experience with Yun and Dragino. Who has more experience with LoRa is @Matheus-Garbelini who is the creator of the ThroughLoRa strategy. For sure I would choose from the list of supported devices if possible.

gioblu, 2018-07-06:
@drtrigon another thing, if you use a shared medium, be sure to use PJON in shared mode, in order to avoid potential collisions with others' devices nearby.

drtrigon, 2018-07-06:
@gioblu: Thanks for your fast reply! I don't know the HC-12 radios and a quick view into google showed me no local distributor so I'll keep looking, but for now it does not seem like a quick alternative. Laser diodes are something I did consider too at some point. Do you have experience? How well does the beam need to be aligned to the receiving diode?
Another question is do you have an idea/experience how good the performance is when putting 3 tunnels in a row (LUDP-SWWP, SWBB-?, ?-SWBB) ? I mean do you expect it to work at all? ;)
I also wondered if it is possible to setup a "3-way tunneler", e.g. LUDP - SWBB - LoRa on a single Arduino Uno?

drtrigon, 2018-07-06:
@Matheus-Garbelini: Thanks for your post and verifying the specs with the manual! That sounds great! I actually already wrote the SWBB - LoRa tunneler and the plan was to use it twice [1]. The thing is - I was not able to progress to that point, as the SWBB bus communication between my Uno+ETH (LUDP-SWBB tunneler [2]) and the Dragino LG01 SWBB part did not work properly. It did on a test setup with Nano (TS-SWBB tunneler [3]) so I currently have no idea (may be the bidirectional level shifter [4] causes issues) ... in any case I have to fix that first ... so at this point I first wanted to verify that my concept can work theoretically at least.

[1] https://github.com/drtrigon/sketchbook/blob/master/OWPJON/ARDUINO/Local/SoftwareBitBang/Tunneler/BlinkingSwitch_SWBB-TL_DRAFT/BlinkingSwitch_SWBB-TL_DRAFT.ino
[2] https://github.com/drtrigon/sketchbook/blob/master/OWPJON/ARDUINO/Local/SoftwareBitBang/Tunneler/BlinkingSwitch/BlinkingSwitch.ino
[3] https://github.com/drtrigon/sketchbook/blob/master/OWPJON/ARDUINO/Local/SoftwareBitBang/Tunneler/BlinkingSwitch_SWBB-TS/BlinkingSwitch_SWBB-TS.ino
[4] https://cdn-shop.adafruit.com/datasheets/txb0108.pdf

gioblu, 2018-07-06:
I have experimented a lot with light pulses communication and laser communication using PJDLS is quite effective. Depending on the beam diameter at the receiver side a certain deviation in the path can be present without issues. So using a lens that has an extremely wide beam at a short distance (let's say a meter) the deviation can be big (a lot of degrees of error) although the maximum range will be obviously limited. In the opposite situation, with a small beam, you will have a longer range but a smaller acceptable deviation. The acceptable deviation is obviously slightly less than the beam radius at receiver side. The beam diameter obviously grows with the distance between transmitter and receiver. Knowing the beam diameter then with simple geometry is possible to determine at a given distance which is the maximum acceptable deviation for a system of this sort.

drtrigon, 2018-07-08:
Good news! I got most of it working! For the 2nd TL-SWBB tunnel, the time I had was to short - will follow. LoRa works! (I have that stuff - for laserdiodes I would have to wait several weeks.)
What is the recommended way for SWBB bus 3.3-5v level conversion? The bidirect. shifter seems not to behave well. Currently I use a voltage divider 68 ohms and 3.6v zener diode. That works on a 5v bus with 2 devices, but I expect it to cause issues on a bigger bus with longer cables?
Now I have to test a bit more and consider all your hints...

drtrigon, 2018-07-08:
> @drtrigon another thing, if you use a shared medium, be sure to use PJON in shared mode, in order 
> to avoid potential collisions with others' devices nearby.

@gioblu: How can I enable the "PJON shared mode" ? https://github.com/gioblu/PJON/wiki/Configuration does not seem to work for LoRa tunnel with PJONAny?

gioblu, 2018-07-08:
Ciao @drtrigon my stupidity to leave outdated pages in the wiki, that will be deleted now, this is the configuration docs (that are also linked in the wiki menu): https://github.com/gioblu/PJON/blob/master/documentation/configuration.md
there you find the info about shared mode and here:
https://www.pjon.org/get-bus-id.php
you can request a unique bus id.

Sorry for the inconvenience :)

gioblu, 2018-07-08:
@drtrigon forgot to say previously that, considering the stability of PJDL using also unconventional conductors for communication, if conductive parts of the house/fixtures that are not grounded are present they should be considered as a potential path for a PJDL bus if the necessary caution is taken to protect the IO pin.

gioblu, 2018-07-09:
Ciao @drtrigon about the level shifter, what about this design:
```cpp
          Schottky diode
      |---[___|_]---|
IO ---|             |---  BUS
      |---\/\/\/\---|
          resistor 
```
so 3v3 from 3v3 device can pass through the path with less resistance (the diode) and conserve its voltage - the diode voltage drop across its ends (using a schottky the drop can be negligible). In the opposite case the 5v would pass through the resistor and its voltage can be brought down to 3v3 selecting the required resistance value.

gioblu, 2018-07-09:
@drtrigon here a simulation with circuitjs:
http://www.falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+12.050203812241895+88+5+43%0Av+112+384+112+256+0+0+40+5+0+0+0.5%0Ad+496+80+112+80+1+0.875%0Aw+112+80+112+144+0%0Aw+496+80+496+144+0%0Ap+496+144+592+144+1+0%0Aw+256+384+496+384+0%0Ap+112+144+0+144+1+0%0Ar+496+144+496+384+0+1000000%0Ar+112+144+496+144+0+510000%0Av+176+384+176+256+0+0+40+3.3+0+0+0.5%0Aw+112+384+176+384+0%0As+256+384+176+384+0+1+false%0Aw+176+256+256+256+0%0Aw+496+144+256+320+0%0As+256+320+256+256+0+0+false%0Ar+64+224+64+352+0+1000000%0Aw+64+352+112+384+0%0As+112+144+112+256+0+1+false%0As+112+144+64+224+0+0+false%0A

Closing the active switches (clicking on them) and activating the others you can simulate the opposite case. It seems it is required to have:
- 2 pull down resistors of 1M
- 1 voltage reducer resistor of 510k
- 1 diode with a forward voltage of 0.875mv

drtrigon, 2018-07-13:
@gioblu: Thanks for all your answers! I try to answer properly:

Thanks for the info in https://github.com/gioblu/PJON/blob/master/documentation/configuration.md about shared mode - will consider that. However I am a bit confused by "Medium" column in the table; why is it radio for OverSampling but serial port for ThroughLoRa? That's radio too (actually the medium is "air" ;). Then I am also missing the more exotic ones you mentioned "unconventional conductors (human body, fruits and vegetables ecc.)". Because this is one of the main reasons why I love PJON meanwhile. Additionally I would strongly suggest to mention that security measures to protect human body might be needed for several of them (e.g. laserdiodes, etc.).

I used this circuit (somehow inspired by V-USB but is not the same):
```
3v3 IO  ---o---\/\/\/\---  5v IO
           |   resistor (68...4k7)
          ---              
          | |  3v6 Zener (protect 3v3)
          ---              
           |
          GND
```
It works but has at least these disadvantages:
  - knee of zener known not to be very sharp, so 3v6 are not precise
  - a device on the 5v bus needs to be able to drive (5v - 3.6v) / 68 ~ 20mA for a high
  - (cut-off frequency might be low due to big capacity)

Because of the 2nd item I plan to setup a "3 way" tunneler (LUDP, SWBB, SWBB) that gives me 2 SWBB pins (on same bus). That way I could put all 5v divices on 1 pin and the 3.3v on a separate, isolating the 2 parts from each other.
Your first circuit looks interesting, but the later post confuses me. So I also setup a simulation http://www.partsim.com/simulator/#148244 and do not understand how it should work. However I did the same with my proposal http://www.partsim.com/simulator/#148247 and there I learned that using a smaller current (bigger resistor, e.g. 4k7) is actually an advantage as it reduces the cut-off voltage. So actually my 2nd concern about drawing to much current is solved.

drtrigon, 2018-07-13:
I should may be also mention that I got the system working: LUDP - SWBB - LoRa - SWBB where on the two SWBB buses my devices are connected. I also setup a device that can connect to DS2438 and DS18B20 1wire devices in order to be able to use the old sensors with my new network also. Very nice!

Some issues still remain, e.g. sometimes the first message does somehow get lost or the answer is related to an request done earlier.

gioblu, 2018-07-13:
Ciao @drtrigon thank you for reporting the inconsistencies in configuration docs, will follow commits to solve those issues. I agree with you must follow also a remark about safety.

The working principle of the suggested circuit is to use a Schottky diode to reduce the amount of voltage loss (if compared to a normal diode) when electrical impulses pass through it making a high pulse from 3v3 side high enough (3/3.1v) for the 5v side to detect it as a 1 also if there is a considerable voltage drop in between. On the other side a single resistor reduces the voltage of the 5v pulses to 3v3 and the Schottky blocks the flow in that direction.

My compliments for having already a network of that complexity in operation :)

gioblu, 2018-07-13:
@drtrigon see this simulation (press play and run it):
http://www.falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+12.050203812241895+88+5+43%0Ad+496+80+112+80+1+0.875%0Aw+112+80+112+144+0%0Aw+496+80+496+144+0%0Ap+496+144+592+144+1+0%0Aw+256+384+496+384+0%0Ap+112+144+0+144+1+0%0Ar+496+144+496+384+0+1000000%0Ar+112+144+496+144+0+510000%0Av+176+384+176+256+0+0+40+3.3+0+0+0.5%0Aw+112+384+176+384+0%0Ar+64+224+64+352+0+1000000%0Aw+64+352+112+384+0%0Aw+112+144+64+224+0%0Aw+176+384+256+384+0%0Aw+176+256+496+144+0%0A
Where 3v3 passes through the Schottky and on "5v" side is 3.09v

Here the contrary:
http://www.falstad.com/circuit/circuitjs.html?cct=$+1+0.000005+12.050203812241895+88+5+43%0Av+112+384+112+256+0+0+40+5+0+0+0.5%0Ad+496+80+112+80+1+0.875%0Aw+112+80+112+144+0%0Aw+496+80+496+144+0%0Ap+496+144+592+144+1+0%0Aw+256+384+496+384+0%0Ap+112+144+0+144+1+0%0Ar+496+144+496+384+0+1000000%0Ar+112+144+496+144+0+510000%0Aw+112+384+176+384+0%0Aw+112+144+112+256+0%0Aw+176+384+256+384+0%0A
Where the resistor reduces the 5v impulse to 3v3.

Obviously depending on the drain of the IO pin in real life the optimal resistors value may vary.

drtrigon, 2018-07-14:
Hmmm, thanks for the simulations I had a look also at mines and replaced the Schottky with an other type, now I can see how this is supposed to work. I have the impression that the resistor needs to have quite the right value as the level (3v3) depends on the divider 1M / (1M + 510k).

My PJON network is very nice - I enjoy playing with it. But it will take some more time to get it working properly as I use solar cells for power supply and thus the used Yun is a bad choice (very power hungry - need to be replaced by an Uno).
I do have minor issues that sometimes the network is not very responsive (through 3 tunnels) what would be the correct way to debug that? Is there a value similar to `TS_RESPONSE_TIME_OUT` to set? Where would I have to set them in the two end-point devices or in all tunnels in between also?
Does it make sense to setup and run https://github.com/gioblu/PJON/tree/master/examples/ARDUINO/Local/SoftwareBitBang/NetworkAnalysis ?

drtrigon, 2018-07-14:
Ohh and thanks a lot for the wiki updates and fixes. Just tried to use the code from https://github.com/gioblu/PJON/wiki/Error-handling and found that the constant names are not correct anymore:
* `CONNECTION_LOST` is now `PJON_CONNECTION_LOST`
* `PACKETS_BUFFER_FULL` is now `PJON_PACKETS_BUFFER_FULL`
* `CONTENT_TOO_LONG` is now `PJON_CONTENT_TOO_LONG`

drtrigon, 2018-07-16:
As mentioned earlier sometimes the first message does somehow get lost or the answer is related to an request done earlier. Other issues like delay resp. not replying on first message also exist mainly on the LoRa tunnel part. The SWBB-TL tunnel reports this:

```
Packet buffer is full, has now a length of 3
Possible wrong bus configuration!
higher MAX_PACKETS in PJON.h if necessary.
Packet buffer is full, has now a length of 3
Possible wrong bus configuration!
higher MAX_PACKETS in PJON.h if necessary.
Connection with device ID 0 is lost.
Connection with device ID 1 is lost.
Connection with device ID 0 is lost.
```

(the devices on the bus have ids 42, 44, 45 and 46 - 0 and 1 do not exist)

gioblu, 2018-07-16:
@drtrigon about possible serial output related to stuff happening before the reset, it may happen in general if serial data was incoming before reset. Probably some earlier data still in one of the buffers. I would say that is an issue of the Arduino IDE.

drtrigon, 2018-07-17:
@gioblu: Let me first write down the hardware setup:

`linux (0) --[ethernet]-- LUDP/SWBB tunnel (1) --[SWBB]-- SWBB/LoRa tunnel (2) --[LoRa]-- SWBB/LoRa tunnel (3) --[SWBB]-- ...`

so that debug output was coming from (2). As I derived all the tunnels from `BlinkingRGBSwitch` they contain `PJON_MAX_PACKETS 3` which I removed now (should increase to default of 5). My "brute-force" solution would be to add a watchdog reset whenever `PJON_PACKETS_BUFFER_FULL` occurs but that does not feel correct... ;)

drtrigon, 2018-07-17:
@gioblu: Am I right in assuming that `Connection with device ID 0 is lost.` refers to the first entry in
```
PJONInteractiveRouter<PJONVirtualBusRouter<PJONSwitch>> router(2, (PJONAny*[2])
{
  &bus1, &bus2
});
```
so this is `bus1` and in that case `StrategyLink<SoftwareBitBang> link1;` so the SWBB part?

Is there any statement possible whether `Connection ... lost.` is more related to hardware (e.g. level shifter, pull-up/downs, etc.) or software issues (like buffer size, etc.)?

fredilarsen, 2018-07-17:
Comments to a couple of questions (quoted):
`Another question is do you have an idea/experience how good the performance is when putting 3 tunnels in a row (LUDP-SWWP, SWBB-?, ?-SWBB) ? I mean do you expect it to work at all? ;)
I also wondered if it is possible to setup a "3-way tunneler", e.g. LUDP - SWBB - LoRa on a single Arduino Uno?`

1. The switches (at least in local mode) start in "hub mode", sending duplicates of each incoming packet to all attached buses except the source bus. When a packet (often a reply) from the target device has been received, the switch will skip duplicating packets because it knows where the device is. This reduces bus traffic and increases speed. So multiple switches in a row is expected to work, slowly at first, but faster as the switches learn which devices are where. Obviously the latency will increase by each switch layer, but speed should not be severely degraded unless sender is waiting for replies.
2. A 3-way (or more) switch is designed to be possible, but not tested. Obviously the maximum packet size and send queue length determines if this is possible with the memory of an Uno. Each PJON object uses (1+PJON_MAX_PACKETS)*PJON_PACKET_MAX_LENGTH bytes of RAM for the buffers. Aside from RAM limitations, the listening time on the SWBB bus will be reduced when polling several buses, so speed will be reduced.

Regarding the use of LoRa and different spreading factors for multiple LoRa switch pairs, given the long range of LoRa, would it not be possible to avoid this by having _one_ SWBB-LoRa switch indoor, and a common LoRa bus for all outdoor SWBB-LoRa switches? This would reduce the number of devices and wireless complexity. @Matheus-Garbelini and @gioblu, what do you think about this?

drtrigon, 2018-07-17:
@fredilarsen: Thanks for your answers!

The "Connection with device ID" text comes from the error handler I added to the tunnel code given in https://github.com/drtrigon/sketchbook/blob/master/OWPJON/ARDUINO/Local/SoftwareBitBang/Tunneler/BlinkingSwitch_SWBB-TL/BlinkingSwitch_SWBB-TL.ino (sorry for not beeing clear about that!).

Yes, I am running the scheme you noted `Device1(Linux, UDP) <-> LUDP-SWBB Switch <-> 2 SWBB-LORA Switches <-> Device2(Arduino, SWBB)`. On both SWBB parts is one Nano using this sketch https://github.com/drtrigon/sketchbook/blob/master/OWPJON/ARDUINO/Local/SoftwareBitBang/DeviceGeneric/DeviceGeneric.ino (with id 44 on the first bus connected to the LUDP-SWBB tunnel and the other has id 42 and is connected to the TL-SWBB tunnel). The one you mention on the LUDP-SWBB Switch (id 44) works very well - though I did not look at the error the LUDP-SWBB tunnel/switch outputs yet.

I wrote - thought not tested yet - also a "draft" for the 3-way tunnel/switch in https://github.com/drtrigon/sketchbook/blob/master/OWPJON/ARDUINO/Local/SoftwareBitBang/Tunneler/BlinkingSwitch_3way/BlinkingSwitch_3way.ino - thanks for the info and insights! Also nice would be to have e.g. a Yun or Dragino used as 3-way switch; LUDP-SWBB-LoRa that way I could reduce the number of switches by 1.

> Regarding the use of LoRa and different spreading factors for multiple LoRa switch pairs, given the long range of LoRa, would it not be possible to avoid this by having one SWBB-LoRa switch indoor, and a common LoRa bus for all outdoor SWBB-LoRa switches?

I don't think I understand that properly but this is exactly the concept I had in mind. I do also plan - and think that should work - to add LoRa devices to this setup. So in total 2 LoRa-SWBB switches to have a remote/outdoor SWBB network and any number of LoRa devices additionally when needed (either sensors or additional LoRa-? switches).

fredilarsen, 2018-07-17:
Basically, to extend to a 3-way switch:
```
StrategyLink<SoftwareBitBang> link1;
StrategyLink<ThroughLora> link2;
StrategyLink<LocalUDP> link3;

PJONAny bus1(&link1);
PJONAny bus2(&link2);
PJONAny bus3(&link3);

PJONInteractiveRouter<PJONVirtualBusRouter<PJONSwitch>> router(3, (PJONAny*[3])
{
  &bus1, &bus2, &bus3
});
```

fredilarsen, 2018-07-17:
I noticed that your handler for PJON_CONNECTION_LOST interprets the data parameter as the target device id, but it is actually the queue position of the packet that failed to be sent. This explains why the id is shown incorrectly. The id is the first byte of that packet (bus.packets[data].content[0]).

drtrigon, 2018-07-17:
@fredilarsen: It seems I used code from an old example/docu - I updated it, but that does actually not compile as there is no `bus`, we have `bus1` and `bus2` instead. So which one to use? Does it matter?

fredilarsen, 2018-07-17:
Ah, laziness always comes back to haunt :-) At the time of implementation of the routing classes I saved a little bit of complexity and time by reusing the standard PJON error callback function type instead of introducing a new one for PJONInteractiveRouter. But as you can see it lacks a parameter pointing to the sending bus, which is needed to do non-trivial error handling. So this is a shortcoming that must be solved.

The best solution would be to introduce a new PJONInteractiveRouter error callback type. That would lead to a minor break of backward code compatibility. But this is relatively new so probably not many will have started using it.
Alternatively, there is a function get_current_bus() that returns the index (array position) of the bus currently executing a callback. We can introduce a get_bus(index) to get hold of the responsible bus.

Until one of these improvements is made available, you can get hold of the correct bus by calling router.get_current_bus() and using bus1 if it returns 0, bus2 if it returns 1 and so on.

gioblu, 2018-07-17:
Ciao @drtrigon @fredilarsen I agree I think an older callback coming from an outdated wiki page I have already deleted. @fredilarsen I see as cleaner to identify the subject using a function instead of defining a new type for the handler. What do you think?

fredilarsen, 2018-07-17:
Well, actually I think a callback function with a reference to the actual bus object, plus perhaps the PacketInfo object, would be the most intuitive solution. All information will be available as parameters to the callback -- no need for investigating which functions to call and how to use them in combination. Perhaps no need for the callback function to know about the router object at all.
The callback function change only relates to the PJONInteractiveRouter class, so it will be few places to change. Mostly the switch examples.

But of course, adding the get_bus function breaks nothing and adds possibilities, so that should be added anyhow. I will send a pull request for that and a few other changes I have to the routing classes, then we can evaluate whether to change the callback function as well.

drtrigon, 2018-07-17:
@fredilarsen: I have the impression that the system works well in "hub mode" (you mentioned earlier) as after reset everything work fine and is fast. With time it runs it starts degrading until some of the tunnels start resetting (using watchdog).

I am at a point where I strongly need support how to debug and improve the system, as I do not understand PJON well enough to get it. I played with `PJON_MAX_PACKETS` but do not even know whether it has to be set to the same value in all devices on the system. The progress I made is by using watchdog for reset and cleaning the bus by receiving and dropping old messages before sending. So I need hints in what direction I have to investigate.

fredilarsen, 2018-07-17:
We can now get the correct target id like this:
```
void error_handler(uint8_t code, uint16_t data, void *custom_ptr)
{
  if(code == PJON_CONNECTION_LOST) {
    SERIAL.print(F("Connection with device ID "));
    SERIAL.print(router.get_bus(router.get_callback_bus()).packets[data].content[0]);
    SERIAL.println(F(" is lost."));
  }
```
Not straight-forward, but a solution.

I sent a pull request for the PJONSimpleSwitch::get_bus(ix) function and some other changes to @gioblu. Should appear in master after review.

fredilarsen, 2018-07-18:
@drtrigon How long does it take before the slowdown is noticeable?
And how many packets per second is passing through the switch in each direction in normal operation?

Setting PJON_MAX_PACKETS to 0 (no packet queue) means that a receiver device that has long intervals spent on other tasks will slow down the sender device as well because it will retry sending until receiving an ACK or getting a timeout. With a packet queue size larger than one, a packet that is not ACKed will be left in the queue and the sender will try to deliver another packet. A packet queue size of 1 makes little sense in most cases I guess. Note that the packet queue is unordered, so it is random which packet is sent first. Using a packet queue gives better performance when communicating with multiple devices, at least if they do not spend most of their time listening for packets (especially for strategies without a hardware buffer). It also makes a sender less vulnerable to slowdowns if one receiver stops responding.

drtrigon, 2018-07-18:
I have to do a measurement in order to give you numbers. It is hard to say currently as I did manual tests only. I need to write a script that communicates say one time per minute and log the results.

From what you explain I do get the impression that my linux code does the wrong thing. As I try to send several times if there is no answer within a few seconds. But actually PJON should take care of this. So I have to go back and rewrite that code I think. Can you explain to me what a proper code should do if there is no reply to a send? How to handle that?

drtrigon, 2018-07-18:
I changed the code to just connect to bus send a message wait for PJON_ACK and return the value. If I run this say 20 times I do get wrong results, correct would be e.g.:
```
owp:dg:v1
b�@
m�A
```
but after some time I get
```
b�@
b�@
m�A
```
this is device id 44 so through 1 tunnel only (the LUDP-SWBB tunnel)... What am I doing wrong?

fredilarsen, 2018-07-18:
Seems you are using a master-slave approach where the master is contacting one slave at a time, waiting for reply. It seems reasonable, sending maximum 3 requests and waiting up to 3 seconds for a reply after each request.

But the UDP port is a shared resource on the computer, so you cannot run multiple of these Linux processes in parallel, meaning that if one slave is down the process will use a lot of time before timing out and exiting and slow down the master running the process and waiting for the reply from stdout.

As an alternative, you could use async logic. Have a packet queue and post packets to multiple devices without explicitly waiting for replies, then have an action when/if a reply actually is received, and potentially a check somewhere to find slaves that have not given a reply within a given time. This will scale better, and allow short downtime for devices without slowing down the master.

A third, hybrid, approach: In my ModuleInterface repo I am contacting each device and waiting for reply for a specific amount of time, where the time is depending on if the device is currently registered as active or inactive. So a device that has replied previously will get a long wait time, while an inactive device will have a very short wait time to avoid slowing the master down. If it replies after the wait time has expired, the response is still accepted and will change it to active state. The only reason why the master is actually waiting for a reply in this context is to minimize the amount of collisions and retransmissions on the bus. It works without a send queue.

fredilarsen, 2018-07-18:
So the first response is lost and the second is repeated? Is this consistent?
I assume the SWBB-LUDP switch is not restarted between each try.
Is the behavior different if it is restarted than if it is not?

drtrigon, 2018-07-18:
> On second thought, should we have a StdIO strategy so that this Linux process could also be a ready-made LUDP-StdIO switch?

Have a look at `owpshell` e.g. in https://github.com/drtrigon/sketchbook/tree/master/OWPJON.

gioblu, 2018-07-18:
In my opinion, a general solution to avoid the proliferation of one middleware per user should be provided :)

fredilarsen, 2018-07-18:
@gioblu @drtrigon Then we have a plan:
1. A StdIO strategy the reads from stdin and writes to stdout. It can write standard PJON packets for talking to other PJON processes in standard PJON packet format, or be set to work in a mode easy to implement in any programming language, for example just start+end separators around each packet, or SFSP or COBS. We can have an abstract base class, with two or more derived classes with different serialization like the above. So StdIOBase (abstract)+StdIONativePJON+StdIOSFSP to begin with.
2. A Linux example setup with 3 programs:

A. A LUDP-StdIO switch program for Linux (and other multi-process OSes), just a modified copy of the existing switches.
B. A master program _not_ using PJON but simply reading and writing to stdin/out with simple means. We could have a collection of serialize/deserialize functions in different languages like C++, Java, Python, Perl, etc to make it easy for the users.
C. A a slave sketch for Arduino, using LUDP.

The master and switch programs can be started and connected on the Linux command line, probably by something like `mkfifo fifo; prog1 < fifo | prog2 > fifo`.
Of course the switch program can be started as a child process by a more complex master program instead of connecting them on the command line.

Sounds good?

drtrigon, 2018-07-19:
As I have mainly issues with the LoRa part and due to other observations I increased `setTxPower(13)` from 13 to default of 17. Did not to much testing yet, but less `BUSY` on the bus and fast response are a good sign - report back in a few hours.

fredilarsen, 2018-07-19:
I think the non-PJON master should be assigned an id, and that sender and receiver ids must be part of the non-PJON messages, otherwise the master would not be able to send to individual devices.

So the master will in a way use PJON principles, relating to ids (and potentially bus ids), but we are trying to avoid duplicating the code base in multiple languages and still support cross-process and cross-language connectivity.

The example from the command line would be running permanently until one of the processes exit. Note that I have not yet tested that command line bidirectional piping so there is some uncertainty, but processes can be connected in multiple ways. For example, I could create a StdIO subclass that starts a specified subprocess and keeps it connected via anonymous pipes. A switch program could then start one or more masters as subprocesses and transfer packets between them and other buses like LUDP.

(The command line pipe character seems to have provided bidirectional piping on Unix (BSD, Solaris, ...) but has degraded to unidirectional piping on Linux. Otherwise command line would be simpler.)

fredilarsen, 2018-07-19:
@drtrigon One off-topic question: You looked at my ModuleInterface repo but stopped. It was created among other things to collect sensor values, so I believed it to be close to what you need. 
May I ask for feedback on what I could change to make it more attractive?

drtrigon, 2018-07-19:
@fredilarsen: I was asking myself the same question yesterday evening. I think the main reason at the moment is that my "productive" hardware is a raspi 1 (!) using owfs and collectd. Thus I was hoping for a quick substitute to owshell (http://www.owfs.org/uploads/owdir.1.html.bak) that I can use to step-by-step replace owfs stuff in collectd. The "productive" system is already loaded quite heavily and I want avoid to change too many things. So it is not directly related to ModuleInterface but schedule.
For the next generation raspi server using the most recent one I will for sure test ModuleInterface but that is a question of free time.
But for you I could start a test setup on a raspi 2 I have around - so I had a look again at https://github.com/fredilarsen/ModuleInterface/issues/5 and https://github.com/fredilarsen/ModuleInterface/tree/master/examples/WebPage and have hardware issues (again);
* I do not own a Mega (not fully true, I have the chip but in another form factor) ...
* LightSensor should be replaced by on-board LED for test, e.g. confer https://github.com/drtrigon/sketchbook/blob/master/Uno_Builtin_LED_Interface/Uno_Builtin_LED_Interface.ino (inspired by AnalogSampling) - all needed is a wire from an analog input to pin 13
* may be suggest to use a VM like VirtualBox for testing the LAMP/WAMP setup, if testing on desktop machine - avoid to change a running system
Otherwise the added infos are very detailed and I would like to test it! I just cannot give promises on my schedule.

fredilarsen, 2018-07-19:
Thanks, some good points there.

drtrigon, 2018-07-20:
I was able to improve a bit here and there (adding pull downs, increase LoRa rx/tx power, changing the send and receive algorithm). Basically what I see is device in front of lora behaves very well. Device behind 2 lora switches is slow but behaves well as long as there is not much traffic. As soon as I start stressing the devices and the bus a bit I start seeing `PJON_BUSY` (sometimes also `PJON_CONNECTION_LOST`) and from that point on the system degrades (messages out of sync, messages getting lost, garbage being returned - all kinds of miss-behaviour). So it basically boils down to the question how to react properly to  `PJON_BUSY` in order to recover the bus?

gioblu, 2018-07-20:
CIao @drtrigon I am not sure the [arduino-LoRa](https://github.com/sandeepmistry/arduino-LoRa/) takes care of avoinding collisions. I think just spits out what has been asked to transmit. What do you mean with "start stressing the devices and the bus a bit" ? Have you taken a look to the optinal configuration documented in the ThroughLoRa README? 

- Consider that when using LoRa care should be taken not to monopolize the exiguous amount of bandwidth available
- Consider to tweak the reception time in the switches to leave enough time for polling, specially if using high spreding factor

@Matheus-Garbelini may help here, I am still a noob in using TL.

drtrigon, 2018-07-20:
> What do you mean with "start stressing the devices and the bus a bit" ?

What I mean is continuous communication - what works is say sending and receiving 10 messages, wait for some time and do that again. But sending and receiving more messages, respectively repeating this continuously results in the behaviour described. Code used is https://github.com/drtrigon/sketchbook/blob/master/OWPJON/LINUX/Local/LocalUDP/RemoteWorker/DeviceGeneric/DeviceGeneric.cpp.

> Have you taken a look to the optional configuration documented in the ThroughLoRa README?
> * Consider that when using LoRa care should be taken not to monopolize the exiguous amount of bandwidth available
> * Try to tweak the reception time in the switches to leave enough time for polling, specially if using high spreding factor

Can you be more specific/concrete, please? What values or constants do I have to tweak?

The code I use is this https://github.com/drtrigon/sketchbook/blob/master/OWPJON/ARDUINO/Local/SoftwareBitBang/Tunneler/BlinkingSwitch_SWBB-TL/BlinkingSwitch_SWBB-TL.ino using these settings:
* PJON_MAX_PACKETS = 3
* bandwidth = 250E3
* TxPower = 17
* SpreadingFactor = 7
* the 2 switches are separated about 1m from each other with about 20cm brick or concrete wall in between

Another question; is CRC enabled by default? Because I do never see `PJON_NAK`...

> It seems strange also for corrupted packets to be able to pass through the 2 crcs, are you sure that what lays on top of PJON is behaving properly?

I am quite sure, as the code is very similar for all devices I use. It is very simple as well; waits for a specific byte and then returns e.g. a string given as `#define` - of course strange things could be going on, but that is not very probable. Code is https://github.com/drtrigon/sketchbook/blob/master/OWPJON/ARDUINO/Local/SoftwareBitBang/OWP_DG_1w-adaptor/OWP_DG_1w-adaptor.ino.

gioblu, 2018-07-20:
Ciao @drtrigon look at the routing docs:
https://github.com/gioblu/PJON/blob/master/documentation/routing.md#routing

You can optionally pass the polling time of each bus:
```cpp
// Optionally polling time can be configured:
PJONBus<SoftwareBitBang> bus2(
  PJON_NOT_ASSIGNED, // Switch device id
  1000               // Polling time in microseconds
);
```
Setting a higher polling time for LoRa buses may help.

About "continuous" sending, depends on what you mean with that, if you mean literally non interrupted sending, you may be probably jammed by other systems using that band, or even other LoRa users, nearby. It is also mandatory not to use LoRa modules for continuous sending, not to completely monopolize the band.

Will now read the linked codebases in detail.

gioblu, 2018-07-20:
@drtrigon sorry I forgot about your crc question, yes the crc is always included:
- one is used to check the integrity of the initial packet info: https://github.com/gioblu/PJON/blob/master/specification/PJON-protocol-specification-v3.0.md#initial-meta-data-crc8
- another is applied at the end taking care to check all the packet: https://github.com/gioblu/PJON/blob/master/specification/PJON-protocol-specification-v3.0.md#end-crc8crc32

Depending on the length of the packet crc8 or crc32 is applied. They should be both safe if used in their operational envelope (as PJON does). If you want to force the use of CRC32 you can call `  bus.set_crc_32(true);`

gioblu, 2018-07-20:
About the codebase: 
- This may not be the best wait to have a timeout: https://github.com/drtrigon/sketchbook/blob/master/OWPJON/LINUX/Local/LocalUDP/RemoteWorker/DeviceGeneric/DeviceGeneric.cpp#L141 specially on a real time OS
- Here longer polling time may help: https://github.com/drtrigon/sketchbook/blob/master/OWPJON/LINUX/Local/LocalUDP/RemoteWorker/DeviceGeneric/DeviceGeneric.cpp#L143
- Also here: https://github.com/drtrigon/sketchbook/blob/master/OWPJON/ARDUINO/Local/SoftwareBitBang/OWP_DG_1w-adaptor/OWP_DG_1w-adaptor.ino#L262

drtrigon, 2018-07-20:
My switch (derived from https://github.com/gioblu/PJON/blob/master/examples/ARDUINO/Local/SoftwareBitBang/Tunneler/BlinkingRGBSwitch/BlinkingRGBSwitch.ino) uses `PJONAny` instead of `PJONBus` might this be relevant for the issues I see?

gioblu, 2018-07-20:
@drtrigon I think @fredilarsen may answer better to this but in the `PJONAny` instantiation: 
```cpp
PJONAny bus2(&link2, PJON_NOT_ASSIGNED, 10);
// Set polling time of 10 microseconds
```
here good hints about LoRa:
https://arxiv.org/pdf/1607.08011.pdf

> A. Real Time Monitoring
> Agriculture, leak detection or environment control are applications
> with a reduced number of periodic/aperiodic messages
> and relaxed delay constraints. In contrast, the communication
> range must be long enough to cope with dispersed location
> of end-devices. LoRaWAN has been designed to handle the
> traffic generated by this type of applications and meets their
> requirements as long as the deployment of the gateways is
> enough to cover all end-devices.
> On the other hand, industrial automation, critical infrastructure
> monitoring and actuation require some sort of real time
> operation. Real time is understood in general by low latency,
> and bounded jitter and depends on the specific application. LoRaWAN
> technology cannot claim to be a candidate solution for
> industrial automation, considering for example that industrial
> control loops may require response times around 1 ms to 100
> ms and that, even for small packets of 10 Bytes, the time on
> air with SF=7 is around 40 ms. As presented in the previous
> section, due to the MAC nature of LoRaWAN, deterministic
> operation cannot be guaranteed despite of application specific
> periodicity as ALOHA access is subject to contention which
> impacts network jitter. Despite that, small LoRaWAN networks
> can deliver proper service to applications that require, for
> instance, sampling data every second. To do that, two main
> design considerations should be taken into account:
> • The spreading factor should be as small as possible to
> limit both the time on air and the subsequent off-period.
> In other words, the gateway must be close enough to the
> end-devices.
> • The number of channels must be carefully designed
> and must be enough to i) minimize the probability
> of collisions (tightly coupled with the number of enddevices)
> and ii) offer quick alternative channels for nodes
> to retransmit collided packets thereby diminishing the
> impact of the duty-cycle.
> Despite the two aforementioned aspects, latency will not be
> deterministic

This seems promising.

I know my suggestion of laser communication may have sounded a little bit silly, but considering the LoRa limitations well described in the document linked and the undoubtedly present privacy issues, laser connectivity, if line of sight is available, may be in some cases a faster, more reliable and more private/secure way to link together branches of a local network.

drtrigon, 2018-07-20:
No laser comm does not sound silly - I just have concerns related to the mechanical setup and safety since I know my skills... Producing accidents is one of my biggest skills. ;)
I understand that LoRa is slow and that I can influence how slow with spreading, bandwidth, etc. But that is no concern to me I am willing to wait for the answer. What I do not understand is why PJON degrades in the way described and stops being reliable in a variety of ways (wrong answer, no answer, etc.). I will try increasing the values you mentioned earlier - is there anything else I can do?

gioblu, 2018-07-20:
Ciao @drtrigon I would be happy to inspect the degrading issues you describe, and thank you very much for dedicating an effort reporting those. Can you try to describe me better what happens? Do you have any log/print? In which section/medium happens? Have you saved some garbage occurrence?

gioblu, 2018-07-20:
@drtrigon we recently pushed some fixes, and @fredilarsen pushed some fixes in particular to the routing classes, are you using master?

drtrigon, 2018-07-21:
I was using v11 now I downloaded master and updated all 3 switches (devices not updated yet as they are SWBB) to this most recent version.
May be we should build unittest cases for this in order to have useful data. So I use this very simple script https://gist.github.com/drtrigon/e7d8c0601539f92f9ace369780735f2a#file-test-sh and do get output like this https://gist.github.com/drtrigon/e7d8c0601539f92f9ace369780735f2a#file-log ("nan" means no data where "nan," means received nan as data). As this includes unpacks that just return "nan" in case unexpected stuff happens it might not be verbose enough, so I updated the unpack to output also raw hex data and did some more tests. Very perliminary; I do have the impression is harder to provoke disconnects and it seems to recover faster.
Is it possible that lora sees reflections, e.g. from walls etc.?

gioblu, 2018-07-21:
Ciao @drtrigon about reflections, yes that can happen. I remember testing LoRa, HC-12 and STX/RX882, being able to obtain a higher range with no line of sight exploiting reflections on buildings, so for sure they occur.

Also @fredilarsen have experimented degradation, and we have isolated the problem in the switch. The problems identified are 2:
1. Freeze in the long run after some days
2. Degradation until halt

Problem 2 seems resolved by the latest changes added by @fredilarsen. He is still testing for the problem 1 although it may have been solved by his fixes or by one fix made by me to `send_packet_blocking`.

fredilarsen, 2018-07-22:
Point 2: I did improve one thing: A missing or silent device would cause delivery to time out, and the switch would slow to a crawl. Now it does not wait if device location is not known, it just sends the packet and continues.

I had a case here where contact with a device was lost, and this effectively destroyed switch performance and I therefore practically lost contact with the devices behind the switch as well. Robustness towards this scenario has been added.

Point 1: I have experienced the LUDP-SWBB switch (Arduino Nano + red Ethernet shield) to lock up after 4 days in my production setup, and have not found the reason yet. I have a test setup at 10x the speed hoping to reproduce it faster, but it has run for 2 days without a problem now. Ongoing task.

drtrigon, 2018-07-22:
So from what I do see after about 1 day is I can tell that it did improve (thanks a lot!!) but still has issues. I don't know whether it is what you mean by 1 and 2, I do see this:
* for the device behind lora (only rarely the one in front) I still see degradation sometimes (even with connection lost) but most of the time fast(er) recovery than with v11
* for the device behind lora I see long delays after some time; so when stopping communication with a device, wait say 15mins and start comm again, the first few send/receives take very long and often even fail - then it usually recovers with say 10 to 20 messages

About the LUDP-SWBB switch I can only say that it is very reliable with v11; had to restart it 1 or 2 times during about a week of operation (or longer), but only after testing and debugging not during "productive" usual operation - so not really to blame the switch. Cannot say to much about the current master version yet.

Please tell me if I can provide you with specific info or run some test code.

@gioblu and may be @fredilarsen: One thing I was wondering about the timing (`TL_COLLISION_DELAY`, polling time, etc.) is that this should actually be configured by PJON automatically given the constraints of lora, see http://www.rfwireless-world.com/calculators/LoRa-Data-Rate-Calculator.html. For my case using SF=7, CR=1 (4/5), BW=250 we get 10937.5 bps (=1.3671875 kB/s) and that basically should also give the values for timing, correct? So either calculate timing from these settings, or constrain these settings in order to fullfill timing needs. (for example using SF=6, CR=1, BW=500 we would get 4.6875 kB/s which would be perfectly fine with SWBB mode 2 using 2.42kB/s)

[25 more replies]

closed 2021-03-25 by gioblu