gioblu/PJON #223: Raspberry Pi compile LocalUDP strategy must not need wiringPi

drtrigon, 2018-07-07:
I wanted to report that I had issues compiling my LUDP code [1] on raspi 1:
* in order to use `-std=c++11` I had to get gcc 4.8, luckily that was as easy as `sudo apt-get install g++-4.8` on my system (wheezy) - run it by using `gcc-4.8` or `g++-4.8`
* the code MUST NOT use `#define RPI true` because that ALWAYS needs wiringPi no matter whether ThroughSerial is used at all
* for the same reason - ThroughSerial not used - `-lwiringPi` is not needed during compilation
* however we do need to link against standard library so `-lstdc++` is needed

Summarizing: I could use exactly the same code as on linux (ubuntu14.04) but had to modify the build command to become (see Makefile [2]):
`gcc-4.8 -DLINUX -I. -I ../../../../../../src/ -std=c++11 DeviceGeneric.cpp -o owpshell -lstdc++`

[1] https://github.com/drtrigon/sketchbook/tree/master/OWPJON/LINUX/Local/LocalUDP/RemoteWorker/DeviceGeneric
[2] https://github.com/drtrigon/sketchbook/blob/master/OWPJON/LINUX/Local/LocalUDP/RemoteWorker/DeviceGeneric/Makefile

gioblu, 2018-07-08:
Ciao @drtrigon thank you for your report. Sadly I do not see a way to conditionally define the wiringPi inclusion without violating the rules I have defined for the system calls definition (I need it to for the serial system calls). 

I will add a dedicated example in the RPI dir including a README with the information you have provided and the required code.

drtrigon, 2018-07-08:
I see no need for separating linux and raspi code. The flag `RPI` should may be renamed to `SERIAL_SUPPORT` or anything like this. And the examples merged. Actually I used the raspi TS example with wiringPi under linux (ubuntu 14.04), changed it to LUDP under linux and then took it to raspi as mentioned. Only the build command/makefile needs modification. I consider this a plus! ;)

gioblu, 2018-07-08:
Ciao @drtrigon, thank you for your support :) I totally agree with you about the examples unification; but about the the `RPI` flag, being not only used to define Serial, but also all the other system calls I am not sure about the name change you propose. 

`RPI` definition can be avoided on RPI if `LINUX` is detected and the linux system calls are defined instead (but only for project that are not making using of analog or digital RPI io pins, that would not work). for this reason I am not sure if the right way to fix this is to call the `RPI` constant `SERIAL_SUPPORT`. 

For example whoever needs io pin handling or analog readings on RPI would need to define `SERIAL_SUPPORT` and in that case `RPI` is probably clearer.

gioblu, 2018-10-25:
Ciao @drtrigon thanks to the support of @xlfe we have now a dedicated set of system calls for serial communication so I can now update the TS LINUX examples and delete the useless `RPI` constant definition and wiringPi inclusion.

gioblu, 2018-10-25:
@drtrigon I also have removed `-lwiringPi` from the makefiles (https://github.com/gioblu/PJON/commit/0148622abd3336ed35bfdecab412c33f52c75f5a), all should be fine now.

drtrigon, 2018-10-25:
Your welcome! And thanks for fixing! (Though I have to update quite some code now... ;)

I will test it as soon as I can - do you have an idea by when you will release the next version? I might want to wait until then.

gioblu, 2018-10-25:
Ciao @drtrigon a double check fro you would obviously be appreciated. I do not have any other change in mind before the release of v11.2. I have to just run again all tests in all possible configurations (which takes few days). So master should reflect v11.2 if I do not find any other inconsistency while testing.

drtrigon, 2018-10-25:
You're right! I will test master before release, but I am under some time constraints thus it might take some time. :)

gioblu, 2018-10-25:
Ciao @drtrigon thank you very much for your support and passion :)

fredilarsen, 2018-12-14:
@drtrigon If you are using LocalUDP you may want to have a look at the new DualUDP strategy which is now in master.

I switched to using this in my own setup, and it is equally simple to use but has some advantages like causing no disturbance in other devices than the receiver device, and less WIFi traffic on any access points if present, and it can reach outside the LAN to specific IPs.

closed 2019-07-12 by gioblu