gioblu/PJON #225: LocalUDP strategy - Mac cannot receive packets

xlfe, 2018-07-12:
I have been testing LocalUDP between a Mac and a RaspberryPi and have encountered some unexpected behaviour.

The network setup looks like   Mac <--> Wifi <--> RaspberryPi - both on the same private subnet.

Using the examples/LINUX/Local/LocalUDP/PingPong/ 

CASE A - If I run the Receiver on the Mac and the transmitter on the RPI, I never get any PINGS or PONGS.

CASE B - If I swap it around, and I run the Transmitter on the Mac and the Receiver on the RPI, the RPI shows PINGS but the Transmitter never shows any PONGS.

If I run Wireshark on the Mac I can see the PONGS from the RPI Receiver on my Mac (as well as the PINGS from the Transmitter running on my Mac) - but it looks like PJON is not picking them up for some reason - the transmitter never shows any PONGS -

PONG/s: 0
PONG/s: 0
PONG/s: 0
PONG/s: 0
PONG/s: 0

gioblu, 2018-07-13:
Ciao @xlfe thank you very much for reporting. I have never tried PJON on mac, I am still working on making it reliable on lower level systems :).

 Nice to know it compiles fine already. 
Could you give a try to LocalUDP or GlobalUDP examples?

I will try to replicate your setup.
Thank you again.

gioblu, 2018-07-15:
Ciao @xlfe could you please describe the building process you have used? I would like to replicate the same setup. Thank you again!

xlfe, 2018-07-15:
I have been doing more debugging but haven't figured it out exactly.

My build process is just typing make in the transmitter/receiver directory
of the LINUX LocalUDP example.

I've started testing GlobalUDP too.

xlfe, 2018-07-16:
RaspberryPI

```bash
pi@raspberrypi:~ $ git clone https://github.com/gioblu/PJON.git
Cloning into 'PJON'...
remote: Counting objects: 11962, done.
remote: Compressing objects: 100% (69/69), done.
remote: Total 11962 (delta 38), reused 67 (delta 25), pack-reused 11864
Receiving objects: 100% (11962/11962), 4.13 MiB | 626.00 KiB/s, done.
Resolving deltas: 100% (6627/6627), done.
pi@raspberrypi:~ $ cd PJON/examples/LINUX/Local/LocalUDP/PingPong/Transmitter/
pi@raspberrypi:~/PJON/examples/LINUX/Local/LocalUDP/PingPong/Transmitter $ make
g++ -DLINUX -I. -I../../../../../../src -std=c++11 Transmitter.cpp -o Transmitter
pi@raspberrypi:~/PJON/examples/LINUX/Local/LocalUDP/PingPong/Transmitter $ ./Transmitter
PONG/s: 0
PONG/s: 0
PONG/s: 0
PONG/s: 0
PONG/s: 0
PONG/s: 0
PONG/s: 0
PONG/s: 0
PONG/s: 0
^C
pi@raspberrypi:~/PJON/examples/LINUX/Local/LocalUDP/PingPong/Transmitter $ cd ..
pi@raspberrypi:~/PJON/examples/LINUX/Local/LocalUDP/PingPong $ cd Receiver/
pi@raspberrypi:~/PJON/examples/LINUX/Local/LocalUDP/PingPong/Receiver $ make
g++ -DLINUX -I. -I../../../../../../src -std=c++11 Receiver.cpp -o Receiver
pi@raspberrypi:~/PJON/examples/LINUX/Local/LocalUDP/PingPong/Receiver $ ./Receiver
PING/s: 0
PING/s: 1
PING/s: 0
PING/s: 1
PING/s: 0
PING/s: 1
PING/s: 0
PING/s: 1
PING/s: 0
PING/s: 1
PING/s: 0
PING/s: 1
PING/s: 0
PING/s: 0
^C
pi@raspberrypi:~/PJON/examples/LINUX/Local/LocalUDP/PingPong/Receiver $
```

MacOS X
```bash
xlfe@mbp ~ (master)$ git clone https://github.com/gioblu/PJON.git
Cloning into 'PJON'...
remote: Counting objects: 11962, done.
remote: Compressing objects: 100% (69/69), done.
remote: Total 11962 (delta 38), reused 67 (delta 25), pack-reused 11864
Receiving objects: 100% (11962/11962), 4.13 MiB | 423.00 KiB/s, done.
Resolving deltas: 100% (6627/6627), done.
xlfe@mbp ~ (master)$ cd PJON/examples/LINUX/Local/LocalUDP/PingPong/Receiver/
xlfe@mbp ~/PJON/examples/LINUX/Local/LocalUDP/PingPong/Receiver (master)$ make
g++ -DLINUX -I. -I../../../../../../src -std=c++11 Receiver.cpp -o Receiver
xlfe@mbp ~/PJON/examples/LINUX/Local/LocalUDP/PingPong/Receiver (master)$ ./Receiver
PING/s: 0
PING/s: 0
PING/s: 0
PING/s: 0
PING/s: 0
PING/s: 0
PING/s: 0
PING/s: 0
PING/s: 0
PING/s: 0
PING/s: 0
^C
xlfe@mbp ~/PJON/examples/LINUX/Local/LocalUDP/PingPong/Receiver (master)$ cd ../Transmitter/
xlfe@mbp ~/PJON/examples/LINUX/Local/LocalUDP/PingPong/Transmitter (master)$ make
g++ -DLINUX -I. -I../../../../../../src -std=c++11 Transmitter.cpp -o Transmitter
xlfe@mbp ~/PJON/examples/LINUX/Local/LocalUDP/PingPong/Transmitter (master)$ ./Transmitter
PONG/s: 0
PONG/s: 0
PONG/s: 0
PONG/s: 0
PONG/s: 0
PONG/s: 0
PONG/s: 0
PONG/s: 0
PONG/s: 0
PONG/s: 0
^C
xlfe@mbp ~/PJON/examples/LINUX/Local/LocalUDP/PingPong/Transmitter (master)$
```

fredilarsen, 2018-07-16:
Great to see that it builds on Mac as well.

Here are a few points that could be tested:
1. Using wired Ethernet instead of WiFi -- is it the same behavior when connected to the same switch/router?
2. Do you have another Mac or PC or Arduino with Ethernet shield available to test without the RPI? If all is perfect in both directions between two computers, this confirms Mac support, which is nice, and points a finger on the RPI.

xlfe, 2018-07-17:
1. I can't *easily* test wired support sorry
2. No I don't have another machine to test with sorry

But.... when I run wireshark on the Mac I can see the broadcast packets coming in from the Pi - its just that PJON never sees them to respond to... So they are definitely "arriving at the Mac"

fredilarsen, 2018-07-19:
Unfortunately I am in the same boat lacking hardware that you are using. I do not have a Mac or a RPI (I never had a use for the graphics part, so I have been using other devices). I can try to simulate the Mac with a FreeBSD VMWare VM, and will have to get a RPI for test purposes.

xlfe, 2018-07-19:
Okay - sorry for wasting everyone's time. 

My firewall was silently dropping the packets with two weird features

1: The packets made it to Wireshark but not to PJON
2: Disabling the firewall only resolved the issue if I restarted the PJON ./Receiver process

So, I did test turning off the firewall before posting the issue, but I obviously did not restart the ./Receiver after turning off the firewall (I just had it running, no packets, firewall off, still no packets and therefore assumed it was not the firewall)!

I think you can close this issue...., and know that your code works on Mac OS X!

closed 2018-07-19 by xlfe