gioblu/PJON #226: Unable to compile PJONslave under ESP8266

sticilface, 2018-07-12:
It seems to bork due to use of `random.  Occurs under the ArduinoIDE and PIO. 

sketch
```cpp
/* ESP8266 - Node MCU side
   Devices can be wired directly being ESP8266 5v tolerant */

#include <PJONSlave.h>

// <Strategy name> bus(selected device id)
PJONSlave<SoftwareBitBang> bus(PJON_NOT_ASSIGNED);

char content[] = "01234567890123456789";

void setup() {
  bus.strategy.set_pin(5); // ESP8266 GPIO 5 or pin D1
  bus.set_error(error_handler);
  bus.begin();

  Serial.begin(115200);
}

void error_handler(uint8_t code, uint16_t data, void *custom_pointer) {
  if(code == PJON_CONNECTION_LOST) {
    Serial.print("Connection lost with device id ");
    Serial.println(bus.packets[data].content[0], DEC);
  }
};

void loop() {
  if(!bus.update())
    bus.send(44, content, 20);
};
```

error under PIO
```
Processing esp8266 (platform: https://github.com/platformio/platform-espressif8266.git#feature/stage; board: esp12e; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: Espressif 8266 (Stage) > Espressif ESP8266 ESP-12E
SYSTEM: ESP8266 80MHz 80KB RAM (4MB Flash)
Converting esp8266.ino
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PJON> 11.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <Ethernet(esp8266)> 1.0.4
|   |   |-- <SPI> 1.0
Compiling .pioenvs/esp8266/src/esp8266.ino.cpp.o
In file included from .piolibdeps/PJON_ID245/src/interfaces/ARDUINO/UDPHelper_ARDUINO.h:5:0,
from .piolibdeps/PJON_ID245/src/strategies/LocalUDP/LocalUDP.h:24,
from .piolibdeps/PJON_ID245/src/strategies/PJON_Strategies.h:66,
from .piolibdeps/PJON_ID245/src/PJON.h:87,
from .piolibdeps/PJON_ID245/src/PJONSlave.h:50,
from /Users/amelvin/Dropbox/projects/Campervan/Builds/esp8266/esp8266/esp8266.ino:5:
/Users/amelvin/.platformio/packages/framework-arduinoespressif8266/libraries/ESP8266WiFi/src/WiFiUdp.h:28:0: warning: "UDP_TX_PACKET_MAX_SIZE" redefined [enabled by default]
#define UDP_TX_PACKET_MAX_SIZE 8192
^
In file included from /Users/amelvin/.platformio/packages/framework-arduinoespressif8266/libraries/Ethernet/src/Dhcp.h:7:0,
from /Users/amelvin/.platformio/packages/framework-arduinoespressif8266/libraries/Ethernet/src/Ethernet.h:9,
from .piolibdeps/PJON_ID245/src/strategies/EthernetTCP/EthernetLink.h:130,
from .piolibdeps/PJON_ID245/src/strategies/EthernetTCP/EthernetTCP.h:24,
from .piolibdeps/PJON_ID245/src/strategies/PJON_Strategies.h:65,
from .piolibdeps/PJON_ID245/src/PJON.h:87,
from .piolibdeps/PJON_ID245/src/PJONSlave.h:50,
from /Users/amelvin/Dropbox/projects/Campervan/Builds/esp8266/esp8266/esp8266.ino:5:
[... 130 lines cut ...]
In file included from /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_modified_sketch_587401/ESPTransmitter.ino:5:0:
/Users/amelvin/Documents/Arduino/libraries/PJON/src/PJONSlave.h: In member function 'void PJONSlave<Strategy>::generate_rid()':
/Users/amelvin/Documents/Arduino/libraries/PJON/src/PJONSlave.h:222:32: error: no matching function for call to 'random()'
         (uint32_t)(PJON_RANDOM()) ^
                                ^
/Users/amelvin/Documents/Arduino/libraries/PJON/src/PJONSlave.h:222:32: note: candidates are:
In file included from /var/folders/nv/8j3v5v0s18v7p4249kv8qbz80000gq/T/arduino_build_313433/sketch/ESPTransmitter.ino.cpp:1:0:
/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:287:6: note: long int random(long int)
 long random(long);
      ^
/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:287:6: note:   candidate expects 1 argument, 0 provided
/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:288:6: note: long int random(long int, long int)
 long random(long, long);
      ^
/Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:288:6: note:   candidate expects 2 arguments, 0 provided
Multiple libraries were found for "Ethernet.h"
 Used: /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/Ethernet
 Not used: /Applications/Arduino.app/Contents/Java/libraries/Ethernet
Using library PJON at version 11.0 in folder: /Users/amelvin/Documents/Arduino/libraries/PJON 
Using library Ethernet at version 1.0.4 in folder: /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/Ethernet 
Using library ESP8266WiFi at version 1.0 in folder: /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi 
Using library SPI at version 1.0 in folder: /Applications/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/SPI 
exit status 1
Error compiling for board Generic ESP8266 Module.
```

gioblu, 2018-07-13:
Ciao @sticilface nice to hear from you again :). Thank you very much for the report, will follow a fix to this compilation error.

sticilface, 2018-07-13:
I’ve started work on my project!! No doubt more questions will follow:)

sticilface, 2018-07-13:
And thanks ;)

gioblu, 2018-07-15:
Ciao @sticilface I have referenced few commits fixing the issue and the examples that now can be compiled by ESP8266 rightaway. Thank you again for the report!

closed 2018-07-15 by gioblu