gioblu/PJON #228: Analog Sampling: a high value of interference parameter

andrei-volkau, 2018-07-17:
Hi,
I am currently trying to get PJON (Analog Sampling) running on Arduino Uno 328p. I am using standard sketches: Analog Sampling-->Network Analysis --> Receiver/Transmitter.

I am using visible light / IR LEDs as wireless bidirectional transceivers. Both of the LEDS are connected in the following way
a) I connected LED + to A0
b) I connected LED - to GND
c) I connected a 1-5 MΩ pull down resistor between A0 and GND (I tried different resistor values)

I see the following info via terminal:
_Packet Overhead: 7B - Total: 0B
Bandwidth: 0B/s
Data throughput: 0B/s
Packets sent: 0
Mistakes (error found with CRC): 0
Fail (no acknowledge from receiver): 4
Busy (Channel is busy or affected by interference): 27
Accuracy: nan %_

It means that packets are not sent.

In some cases, one packet is sent.
_Packet Overhead: 7B - Total: 7B
Bandwidth: 27B/s
Data throughput: 20B/s
Packets sent: 1
Mistakes (error found with CRC): 0
Fail (no acknowledge from receiver): 2
Busy (Channel is busy or affected by interference): 32
Accuracy: 100.00 %_

 I think that the problem is related to "Busy (Channel is busy or affected by interference)" parameter. The parameters accepts values between 27 and 153. **How can I decrease interference?** I tried to use a current limiting resistor. The resistor did not helped. I work in completely dark room.

I am attaching a signal from transceiver
[image]

I am using PJON v 11.0.0. I tried the current master also.
Kind regards

gioblu, 2018-07-17:
Ciao @andrei-volkau yes, depending on the type of LED the channel may be detected often busy. If you comment the `can_start` function leaving only `return true` you will probably get a more stable flux of packets. I am still not sure about the `can_start` function, but soon a patch will come to fix this issue. After the disable of `can_start` if the reliability is still not optimal, try changing the pull down resistor value.

gioblu, 2018-07-17:
@andrei-volkau if you need more range use AnalogSampling with 2 pins (input and output) so you can drive the emitter separately with an amplification stage (for example just dive the base of a transistor able to output 100ma to LED).

gioblu, 2018-07-17:
@andrei-volkau another way to reduce the amount of busy without changing the codebase is to reduce the value of the pull down resistor under 1 MΩ although that reduces the maximum range.

andrei-volkau, 2018-07-17:
@gioblu 
**Item 1.**
I commented the `can_start` functionleaving only `return true`.  I mean the function from `src/strategies/AnalogSampling/AnalogSampling.h` Unforunately it did not helped. 

I am receiving the following messages. 
```
Packet Overhead: 7B - Total: 0B
Bandwidth: 0B/s
Data throughput: 0B/s
Packets sent: 0
Mistakes (error found with CRC): 0
Fail (no acknowledge from receiver): 4
Busy (Channel is busy or affected by interference): 0
Accuracy: nan %
```
I am receiving the above messages in case of using  1-5 MΩ pull down resistors. I am receiving the above messages in case of using pull down resistor under 1 MΩ also (I used  13kΩ pull down resistor).

**Item 2.**
I reduced the value of the pull down resistor under 1 MΩ without changing the codebase. 13kΩ pull down resistor works well for me. So the value of `Busy (Channel is busy or affected by interference):` parameter is equal to 0 in most cases. The more resistance value, the higher value of `Busy (Channel is busy or affected by interference):` parameter. 

Unfortunately there is no any sent packets. I am able to see that sometimes 1-2 packets are sent after a lot of attempts.

andrei-volkau, 2018-07-18:
I tried 2 IR LEDs and one visible light LED. Please, see the following images. I tested them according to the proposed method.

[image]
[image]
[image]
[image]

IR diode (probably VISHEY TSAL6100) seems to be the best option.

Unfortunately any diode do not allow packets to be sent. I use 13kΩ pull down resistor. So `Busy (Channel is busy or affected by interference):`parameter is equal to 0 in case of using 13kΩ pull down resistor.

> Packet Overhead: 7B - Total: 0B
Bandwidth: 0B/s
Data throughput: 0B/s
Packets sent: 0
Mistakes (error found with CRC): 0
Fail (no acknowledge from receiver): 11
Busy (Channel is busy or affected by interference): 0
Accuracy: nan %

gioblu, 2018-07-18:
@andrei-volkau are you using Arduino UNO on both sides?

andrei-volkau, 2018-07-18:
Hi @gioblu 

Sorry, for the silly mistake. I added Arduino UNO as a receiver. So the transmission was started. The transmission works in case of a low value of `Busy (Channel is busy or affected by interference)` parameter 

> Packet Overhead: 7B - Total: 70B
Bandwidth: 270B/s
Data throughput: 200B/s
Packets sent: 10
Mistakes (error found with CRC): 0
Fail (no acknowledge from receiver): 2
Busy (Channel is busy or affected by interference): 0
Accuracy: 100.00 %

[image]

gioblu, 2018-07-18:
Ciao @andrei-volkau I am happy to hear that now it works fine, thank you for your feedback, can I use some of those pictures in the wiki? I was considering if to add the checklist I have wrote and those images, if you are ok with it, in the [AnalogSampling](https://github.com/gioblu/PJON/tree/master/src/strategies/AnalogSampling) README obviously including an acknowledgment. :)

gioblu, 2018-07-19:
@andrei-volkau thank you for your support to the project :), I hope PJDLS will be useful for you. I will close here, feel free to reopen if required.

closed 2018-07-19 by gioblu