gioblu/PJON #232: Add ThroughSerial support for LINUX?

xlfe, 2018-07-21:
It would be great to be able to use ThroughSerial support for LINUX

I have a prototype which is working well (including using Python through PJON-cython) using the same serial interface from WiringPi (WiringPiSerial)

I have got it working on my Mac talking bi-directionally to an Arduino over USB serial - (but I had to set self.bus.set_synchronous_acknowledge(false) on both sides of the connection).

Would you be open to a pull request?
https://github.com/gioblu/PJON/compare/master...xlfe:LINUX_Serial

gioblu, 2018-07-22:
Ciao @xlfe thank you very much for your support.
I have studied your contribution, one thing I see is that wiringpi (at least the serial part) should work on a linux machine and in some way your contribution proves it, Have you tried to download and test wiringpi on linux?

My only concern with your contribution is the code duplication and the future need to keep track of wiringpi development, If wiringpi can be downloaded and included as an external dependency as we do on RPI, I think may spare us some work in the future.

What do you think?
Thank you again and sorry for the delayed answer.

xlfe, 2018-07-25:
Hi @gioblu no problems at all - sorry for the delay in getting back to you, I will try to refactor slightly and then send you an update in the next few days

xlfe, 2018-07-31:
Hi @gioblu - I wanted to come back and ask a few questions - I'm not sure it makes sense to use WiringPi for the generic Linux/Posix/mac serial interface?

I understand you have used wiringpi for the hardware pin support on Raspberry Pi - which makes sense - but rather than try and use it for Linux serial support too, would you be open to finding a cross platform serial library to use instead? 

Ie we should be able to use native functions on Arduino (and other micros) and then one common library supporting linux (and rpi), mac os x and windows

Two options are 
https://github.com/wjwwood/serial
https://sigrok.org/wiki/Libserialport

If you're amenable I will put together a pull request that doesn't use wiring pi but does cover all use cases? 

That, combined with the interesting work @sticilface is doing under #237 will be a great improvement to serial support across all platforms

gioblu, 2018-08-01:
Ciao @xlfe I agree with you that wiringPi used on LINUX is just a quick and dirty hack.

The wjwwood/serial project is really cool I did not know about it, it can be for sure a source of inspiration.

A couple of considerations:
- I see the project maintenance is poor and the latest release is of 22 Apr 2015 
- I also a fair amount of open issues for linux and windows https://github.com/wjwwood/serial/issues

For this reason I am not sure if this swap of the serial system calls could be effectively proficient. I think the safest way to go should be to select the most updated and reliable dedicated implementation for each architecture. What do you think?

xlfe, 2018-08-06:
@gioblu Hmm - I didn't see those open issues but yes there looks to be a few...

What about libserialport instead? - I think that is used by the Arduino project for their cross platform serial support - https://github.com/arduino/Arduino/issues/5096#issuecomment-230305207

gioblu, 2018-08-31:
Ciao @xlfe I have taken a look back to serial (https://github.com/wjwwood/serial) and I see that the last commit is of feb and many issues are piling up, on the other side I see your link to the other library (https://github.com/arduino/listSerialPortsC) although it seems probably not as mature as wiringpi is. Probably for now, until a good cross-platform library is found your solution can let users use serial on LINUX, for this reason I asked you for a pull request. 

I will also try as soon as possible Libserialport as you suggested.

xlfe, 2018-09-25:
Now supported thanks!

closed 2018-09-25 by xlfe