gioblu/PJON #233: Connecting 2 LEDs per one robot (question)

andrei-volkau, 2018-07-21:
Hi

I want to create a swarm of robots using "AnalogSampling".
Please look at the follow image.
[image]

I have the following question. How can I connect 2 IR LEDs to Robot №2. 
Requirements:
1. Robot №2 should be able to communicate with Robot №1
2. Robot №2 should be able to communicate with Robot №3 

Best regards,
Andrei

gioblu, 2018-07-21:
AH @andrei-volkau look, a hint `strategy.threshold` contains the analog value that determines the difference between 0 and 1, AnalogSampling uses an adaptive system for threshold to be able to work with signals magnitude between few millivolts and 1.1v (if analog reference 1.1 is used). The threshold is effectively the distance between the transmitter and the receiver, it has a high value if the 2 transceiver are near or in contact and a value near 0 at the maximum range.

gioblu, 2018-07-21:
@andrei-volkau it may have sense to leverage of the Switch or SimpleSwitch classes contributed by @fredilarsen to route packets between robots, so you dont have to reimplement that part of functionality.

andrei-volkau, 2018-07-25:
Hi @gioblu ,
**Item 1.**

> _Gioblu: "AnalogSampling samples using polling, so you must dedicate the same time to both PJON instances' receive call, but it should be fine"_

Sorry, Do you mean passing the maximum reception time in microseconds?
`bus.receive(1000);`

**Item 2.**
I executed standard sketches (Switch/SimpleSwitch). I adopted the code for AnalogSampling strategy. The sketches works fine.

 It easy to specify `receiver_function` in the case of using `InteractiveRouter`.
`router.set_receiver(receiver_function);`
How to do the same thing in case of using `Switch/SimpleSwitch`?

_Reason: it is great to call `receiver_function` as soon as new packet is received._ 

**Item 3.**
My goal is to connect 2 IR LEDs per one Arduino Uno. Each of the LEDs should be able to receive/transmit packets.

How can I do that? Please correct me if I am wrong, but It looks like that I need to follow the following steps.
1) I have to create 2 PJON instances. 
```
PJON<AnalogSampling> bus1(44);
PJON<AnalogSampling> bus2(44);
```
2) I have to specify pins for each of the PJON instances
`bus1.strategy.set_pin(A0);`
`bus2.strategy.set_pin(A1);`
3) I can specify the same `receiver_function` for both of the PJON instances
```
bus1.set_receiver(receiver_function);
bus2.set_receiver(receiver_function);
```
4) I can send packets via any bus that I want
`bus1.send(45, "B", 1);`
`bus2.send(45, "B", 1);`

**Item 4.**
This chart is in regards to determining the distance between the transmitter and the receiver .

[image]
I am using 680kΩ pull down resistor. 
I am using 100Ω current limiting resistor.
I am using IR LED

Thank you in advance for you response.

gioblu, 2018-07-25:
Ciao @andrei-volkau 

**Item 1.**

https://github.com/gioblu/PJON/blob/master/documentation/routing.md
second parameter of `PJONBus` is the polling time for that bus. 

if using PJONAny:
```cpp
PJONAny bus2(&link2, PJON_NOT_ASSIGNED, 10);
// Set polling time of 10 microseconds
```
**Item 2.**
For the switch / simpleswitch see the source here:
https://github.com/gioblu/PJON/blob/master/src/PJONSimpleSwitch.h#L105

**Item 4.**
Would be interesting to see which graph comes out from `distance = sqrt(max_threshold / threshold);`
 where `max_threshold` in this chart is 350.

try using a more resistive pull down, if works you are extending range having a higher voltage output.

fredilarsen, 2018-07-25:
Hi @andrei-volkau 
Regarding callback functions, these are not supported by the routing classes except when PJONInteractiveRouter is "added on top". I named it "Interactive" because it calls user functions when something happens,
So PJONInteractiveRouter does not change the underlying routing, it just adds the callback functionality.

Look at the ARDUINO\Local\SoftwareBitBang\Tunneler\BlinkingRGBSwitch example to see how it is used along with the other classes.
If you do not want `PJONInteractiveRouter<PJONVirtualBusRouter<PJONSwitch>>` you can just do `PJONInteractiveRouter<PJONSwitch>` instead to add the callback functionality directly to PJONSwitch.

gioblu, 2018-07-26:
@andrei-volkau good, not sure why is still not perfectly linear but it will probably more than enough to let them rove in formation.

gioblu, 2018-07-26:
@andrei-volkau if you need more range than the one you can achieve with this configuration, you can use a couple of pins (separate transmitter and receiver channels) and add a transistor driven emitter with a higher intensity output. This should let you achieve higher range.

gioblu, 2018-09-28:
Ciao @andrei-volkau closing here, feel free to reopen if required!

closed 2018-09-28 by gioblu