gioblu/PJON #237: ThroughSerial is currently a blocking function.  Why not let serial interrupts do the work?

sticilface, 2018-07-28:
Extensive testing on ESP32, ESP8266 when connected to a AVR leonardo acting as a router between SWBB and ThroughSerial demonstrate very poor and intermittent performance.   This has been quite troublesome to track down and probably is the result of more than one factor.   Despite getting no transmission using ThroughSerial and the  speed test sketch at 115200 baud i could happily print text between the two devices.  

I then refactored the ThroughSerial to work async and I get a large speed increase.    Leonado to Leonado went from 19 packets per second to 48,  leonardo to ESP32  went from 59 to 0!  This was unexpected.  If i then included `bus.receive()` in the speed test sending loop as per below:
```cpp
void loop() {  
   if (!bus.update()) {
     bus.send(44, content, 20);
   }
bus.receive();
}
```
packets went from 0 to 265 per second, so a huge increase.  

Additionally sending packets at a given rate also works without including `bus.receive();` on the sending side, so the 0 fail is due to the `!bus.update()`. 

```cpp
uint32_t timer = 0; 
void loop() { 
  if (millis() - timer > 20)
  {
    bus.send_packet_blocking(44, content, 20);
  }
};
```
so it is the returned ACK value that is stopping the speed test from firing!  I currently have no idea why `bus.receive()` does this.  my thoughts were that the ACK is getting lost in the Rx buffer if there are other bytes before it, but this is not the case.  as consuming the buffer before sending makes no difference.  and this is the receive_response() debug output...  4480us for the ACK to arrive.  one byte! 
```
[receive_response, line 139] Pre ACK Rx size: 0, time = 4
[receive_response, line 139] Pre ACK Rx size: 0, time = 192
[receive_response, line 139] Pre ACK Rx size: 0, time = 249
[receive_response, line 139] Pre ACK Rx size: 1, time = 4480
```
including the `bus.receive()` shortens the time for the ACK
```
[receive_response, line 139] Pre ACK Rx size: 0, time = 4
[receive_response, line 139] Pre ACK Rx size: 1, time = 2427
[receive_response, line 139] Pre ACK Rx size: 0, time = 3
[receive_response, line 139] Pre ACK Rx size: 1, time = 2432
```
Here is a gist with the speed test results: https://gist.github.com/sticilface/0a68f3c4a35ad2b5928f1d41569e7a06#file-speedtests-md

# Changes

I have made changes to ThroughSerial and created a new strategy called ThroughAsyncSerial. 
It has the following changes... 

- `receive_byte` has been removed.  byte retrieval is performed by lower level routines. 
- `ThroughAsyncSerial` creates a second `uint8_t` buffer with size `PJON_PACKET_MAX_LENGTH`
- `receive_string` is called by `bus` continually and it checks serial availability, consumes irrelevant/missed bytes, copies data after start byte to its own buffer then when the end is reached returns the whole packet to PJON.  
- during packet reception TAS_FAIL is returned to PJON even if partial packets are actually retrieved. 
- a 100us timeout has been added to `receive_string` as i found a little delay is good.  calling `serial.available()` to frequently actually slows down reception. this can be changed at runtime with `set_timeout(uint32_t)` and `get_timeout()`. 
- no changes to the sending of packets.  receive_byte function has been placed in `receive_response` as this is the only place it is used. It is still blocking. sending packets is still blocking. 

- Router now works flawlessly and can forward packets at a rate of 30/s with no errors.  SWBB can spend all time on it, as ThroughSerial no longer blocks sketch execution. 

# still to do

- work out why `bus.receive()` improves speed test by 2x!  - no idea.... 
- remove constant timeout https://github.com/sticilface/PJON/blob/AsyncSerial/src/strategies/ThroughAysncSerial/ThroughAysncSerial.h#L152
- move to async send if possible. 
- check if works on other platforms (windows, mac,  rPi).  I do not have these set up 
- Take on suggested Improvements + remove debugging. 
- Potentially replace original ThroughSerial. 

changes here.. 

https://github.com/sticilface/PJON/tree/AsyncSerial

gioblu, 2018-07-31:
Ciao @sticilface this is an extremely valuable piece of work and I am sure will solve many issues to many users. My compliments for your contribution and thank you for your support to the project.

About the issue you firstly expose I do see an explanation: 
The send function does not clear the ACK byte in the buffer and being always present data transmission is never initiated because of its presence (because of `can_start`). For sure calling `receive` solves the issue.

About async send, requiring a breaking change in PJON.h I would work on it in the next iteration as soon as we will release v11.1 (before august 8). 

As I wrote you directly on gitter, we can decide if we want to include in 11.1 the strategy developed by you via pull request or wait for v11.1 to then work on async. 

I have thought a little about it and I see that the complete async solution as the most effective and probably the more reliable, so probably could have sense to wait for v11.1 and then jump in :)

About the surrogate issue see gitter,
thank you very much again.

sticilface, 2018-07-31:
Just saw this now!  sorry.  

yes i agree.  I feel that async serial will provide a significant speed improvement for most embedded use cases.  It would be good to get it working with full async.  

Regarding when to include it..  
I have pushed it to my repo... would be good to get some other feedback and wider testing maybe before including it. so we could maybe encourage people to at least test it if they have working set ups with ThroughSerial included.  

Regarding inclusion.  I'm easy either way!  it has certainly (combined with surrogate) solved my initial issue of poor performance when routing!  I'd be interested to know if others feel the same? I will make a pull request but only for other people to try it and for comments, with the aim of including it in the future. 

You are welcome.  It is a pleasure to work together and solve problems.  I do love this approach to problem solving and open source software in general!

ibantxo, 2018-11-27:
Umm... I have been reading the [ThroughSerialAsync]( PJON/src/strategies/ThroughSerialAsync/ThroughSerialAsync.h) code and I cannot find the [SerialEvent](https://www.arduino.cc/en/Tutorial/SerialEvent) function.

Is that OK? Many thanks

ibantxo, 2018-11-28:
The condition "can_start" to know if channel is free is (both):
1) no data at serial port available
2) PJON_MICROS() - _last_reception_time >= TSA_TIME_IN

_last_reception_time = 0 each time receive_string is called and state=TSA_WAITING

I think that channel is free if no changes at bus: 
`attachInterrupt(digitalPinToInterrupt(interruptPin), fast_and_easy_function, CHANGE);`

void fast_and_easy_function(void){
_last_bus_activity=PJON_MICROS();
}

or something like that?

gioblu, 2020-03-04:
Solved with the addition of ThrouSerialAsync.

closed 2020-03-04 by gioblu